MichaIng/DietPi #426: Dropbear logging - potential bug?

Stickerpants, 2016-07-13:
I was looking for the logfile for dropbear the other night, and I ran across something that might be a bug. Here's some relevant config info first:

* Default dropbear, only thing changed is the config file for it
* Using logging solution \#2 (Write to ramdisk, persist to SD card periodically)

Based on what I can see in `/etc/dropbear/log`, there's a `run` script that want to execute the line `exec chpst -udropbearlog svlogd -tt ./main`.

Judging by how `dropbearlog` user isn't set up by default, and both `chpst` and `svlogd` don't exist, I assume that this `run` script isn't in use.

So ultimately, I'd like to get dropbear logging working. I haven't ruled out the possibility that this is just in my configuration - if there's anything I can install to get it to work that would be super cool.

If this is a legitimate bug or lack of functionality, I would be more than happy to tackle this in a pull request, if you are willing to provide me with a little nudge in the right direction :)

Stickerpants, 2016-07-14:
I figured it out, in the end it was an issue on my end. If `rsyslog` is installed, then dropbear properly logs to `/var/log/auth.log`. My mistake!

closed 2016-07-14 by Stickerpants

Fourdee, 2016-07-15:
@Stickerpants 
Yep, rsyslog is not installed by default on DietPi systems to save resources. 

You can either manually install it (as you did), or, select full logging mode which will also install it: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68