MichaIng/DietPi #427: v123 | PiHole web requests update

Fourdee, 2016-07-16:
>  ttps://github.com/Fourdee/DietPi/commit/6a6159b665bc9ed92f9509e604410de34ae875dd#commitcomment-18268826

Fourdee, 2016-07-16:
> https://github.com/pi-hole/AdminLTE/blob/master/footer.php#L8-L9

Fourdee, 2016-07-16:
Ok, we'll hard code the version numbers in. Else, we have to install git.

Notes for self:
```
#From their installation

#PiHole ver:
root@DietPi:/etc/.pihole# cd /etc/.pihole/ && git describe --tags --abbrev=0
v2.8.1

#Web ver: 
root@DietPi:/etc/.pihole# cd /var/www/html/admin/ && git describe --tags --abbrev=0
v1.3
```
Replace in https://github.com/Fourdee/AdminLTE/commit/5d348d5c11709e4213d1747c751c3a789ee6841f

k-plan, 2016-07-16:
@Fourdee 

- short test on Odroid C1+
- update to last `gitbranch=testing`
- `dietpi-uninstall` PiHole => `dietpi-software` PiHole

Results: - looks good so far 😄 

[image]

Fourdee, 2016-07-16:
Whitelist and Blacklist additions are not working from the web, looking into it.

Fourdee, 2016-07-16:
notes:
```
root@DietPi:~# git_version='04_07_2016';git_owner='Fourdee';/usr/local/bin/gravity.sh;wget https://raw.githubusercontent.com/"$git_owner"/pi-hole/"$git_version"/pihole -O /usr/local/bin/pihole
:::
::: You are root.
::: Neutrino emissions detected...
:::
::: No custom adlist file detected, reading from default file... done!
:::
::: Existing pihole directory found
:::
::: Getting raw.githubusercontent.com list... List updated, transport successful!
::: Getting adblock.gjtech.net list... No changes detected, transport skipped!
::: Getting mirror1.malwaredomains.com list... No changes detected, transport skipped!
::: Getting sysctl.org list... No changes detected, transport skipped!
::: Getting zeustracker.abuse.ch list... No changes detected, transport skipped!
::: Getting s3.amazonaws.com list... No changes detected, transport skipped!
::: Getting s3.amazonaws.com list... No changes detected, transport skipped!
::: Getting hosts-file.net list... No changes detected, transport skipped!
::: Getting raw.githubusercontent.com list... List updated, transport successful!
:::
::: Aggregating list of domains... done!
::: Formatting list of domains to remove comments.... done!
::: 127903 domains being pulled in by gravity...
::: Removing duplicate domains.... done!
::: 103689 unique domains trapped in the event horizon.
::: Formatting domains into a HOSTS file...
:::
::: Adding 9 adlist sources to the whitelist... done!
::: Running whitelist script to update HOSTS file.... 7 domains whitelisted!
::: Running blacklist script to update HOSTS file.... 0 domains blacklisted!
:::
::: Cleaning up un-needed files... done!
:::
::: Refresh lists in dnsmasq...sed: can't read /etc/dnsmasq.d/01-pihole.conf: No such file or directory
 done!
--2016-07-16 16:02:43--  https://raw.githubusercontent.com/Fourdee/pi-hole/04_07_2016/pihole
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... failed: Name or service not known.
wget: unable to resolve host address ‘raw.githubusercontent.com’
```

gravity is restarting dnsmasq during installation and creates a temp downtime.

Fourdee, 2016-07-16:
@k-plan 
Whitelist/blacklist now functions through web interface.

Fourdee, 2016-07-16:
Completed, please reopen if required.

closed 2016-07-16 by Fourdee