MichaIng/DietPi #429: Pine A64 | Fails to boot a second time on fresh installation

Fourdee, 2016-07-17:
I'am unable to replicate this on any of my Pine A64 devices.

> http://pastebin.ca/3660158

Fourdee, 2016-07-17:
http://pastebin.ca/3660158
```
** Unable to read file boot.scr **
Image lacks image_size field, assuming 16MiB
[mmc]: MMC Device 2 not found
```

Fourdee, 2016-07-17:
@rhkean 
Possible uEnv.txt entry? any ideas?
```
image_size
```

kb3vgw, 2016-07-17:
the is RNeese on the latest img did you updated the bootloader when you updated the kernel ?

Fourdee, 2016-07-18:
@kb3vgw 

Yep, uboot is updated as well as kernel. We use Longsleeps update script.

Can you tell me the revision of your board and mgr date, eg:
- A64-DB-Rev B
- 2016-02-24

kb3vgw, 2016-07-18:
A64-DB-rev-d
2016-03-21

Fourdee, 2016-07-18:
@kb3vgw 
All mine are Rev B.

Ok, I know you've done alot of tests on your own already, but I'd like to get a list of results that we can work from.

Can you please run the following tests. One at a time. Then let me know the results for each:

#### First
- Download and extract the DietPi image again: http://dietpi.com/downloads/images/DietPi_PineA64-arm64-(Jessie).7z

#### Test 1 (MD5)
- Please check the MD5 (0bb911e301e9ed65af42e570d2bc00a8) against your downloaded and extracted image ```DietPi_v123_PineA64-arm64-(Jessie).img```. 

##### Test 2 (PSU)
Write a fresh v123 image to SD: http://dietpi.com/downloads/images/DietPi_PineA64-arm64-(Jessie).7z
- Use a power supply from one of your Raspberry Pi and not the PSU that came with the PineA64

#### Test 3 (Partition expansion disabled)
Write a fresh v123 image to SD: http://dietpi.com/downloads/images/DietPi_PineA64-arm64-(Jessie).7z
Extract these files to ```/boot/dietpi/``` before powering on the device
[boot.zip](https://github.com/Fourdee/DietPi/files/369301/boot.zip)

kb3vgw, 2016-07-18:
ok that works but we still have the issue of the serial terminal failing after the mmc 2 point. but it does boot and run

kb3vgw, 2016-07-18:
when you boot the 123 its saying its a 122 img and then goes into update mode btw.

k-plan, 2016-07-18:
@kb3vgw 

> when you boot the 123 its saying its a 122 img and then goes into update mode btw.

I had  the equal question. :smile: 

Answer: https://github.com/Fourdee/DietPi/issues/389#issuecomment-228564487

Fourdee, 2016-07-19:
> ok that works but we still have the issue of the serial terminal failing after the mmc 2 point. but it does boot and run

Which test worked for you?

kb3vgw, 2016-07-19:
Adding the boot file

Fourdee, 2016-07-19:
@kb3vgw 

> serial terminal failing after the mmc 2 point.

DietPi disables serial console by default to save resources. You can enable it again by running:
```
/DietPi/dietpi/func/dietpi-set_hardware serialconsole 1
```
Its also in ```dietpi-config``` > advanced options

> Adding the boot file

Thanks :+1: , looks like it might be the partition resize during 1st (which was disabled on that boot file). I also disabled a few other options. 

#### Test 4 
i'd like to run the same test again with only partition resize disabled to verify its the issue:
- Write a fresh v123 image to SD
- Extract this file ([boot.zip](https://github.com/Fourdee/DietPi/files/371726/boot.zip)
) to /boot/dietpi/ before powering on the device

If the device runs successfully, after DietPi has completed 1st run setup, please run the following and paste results please:
```
df -h
blkid
```

Fourdee, 2016-07-22:
For reference, email of @kb3vgw results:
```
/dev/mmcblk0p2  860M  824M     0 100% /
devtmpfs        458M     0  458M   0% /dev
tmpfs           491M     0  491M   0% /dev/shm
tmpfs           491M  6.6M  484M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           491M     0  491M   0% /sys/fs/cgroup
tmpfs            10M  1.2M  8.9M  12% /DietPi
tmpfs            20M   12K   20M   1% /var/log
tmpfs           491M     0  491M   0% /tmp
/dev/mmcblk0p1   50M   14M   37M  27% /boot
```

Fourdee, 2016-07-22:
@kb3vgw 
I still believe this issue is caused by a unstable PSU and/or SD card that corrupts the filesystem on resize.
We use the same resize script for all DietPi boards that targets the 2nd partition ```/dev/mmcblk0p2```. So if the resize was "wrong", all boards would be effected. I've not received any reports of a failed partition resize from all our users.

In a previous email, you mentioned that other users on IRC were experiencing the same issue as you. I've not received any reports from other users yet. Please ask the users to contact me via this git ticket, or email.

Fourdee, 2016-08-11:
No response from user. Marking as closed. Please reopen if required.

closed 2016-08-11 by Fourdee