MichaIng/DietPi #430: Supervisors folder created in /var/log doesn't survive a reboot

Rkieltyka, 2016-07-18:
An app that I'm running on DietPi uses supervisor that requires `/var/log/supervisor` directory to run. This directory is created initially when I install my app and everything works fine. But when I reboot my Raspberry PI that directory disappears and my app doesn't start.

There is a fix : 
in` /etc/init.d/supervisor`
I added
```
# Create LOGDIR for tmpfs                                                                                                   
[ -d /var/log/supervisor ] || mkdir /var/log/supervisor
```

Which solves the issue. Please investigate this bug and reply what your thoughts on it are.

Fourdee, 2016-07-18:
@Rkieltyka 

Hi, I am unable to replicate this issue on Odroid C2:
```
apt-get install supervisor -y
reboot
```

```
root@DietPi:~# ls -lha /var/log/supervisor/
total 4.0K
drwxrwxr-x 2 root root  60 Jul 18 12:50 .
drwxrwxr-x 9 root root 380 Jul 18 12:50 ..
-rwxrwxr-x 1 root root 380 Jul 18 12:50 supervisord.log
```

I'll need some more information on your system. Please can you send us a bug report? http://dietpi.com/phpbb/viewtopic.php?f=8&t=499

Rkieltyka, 2016-07-18:
As requested : 47cb7af2-e9b2-4828-a087-9a6ab7226534-0

Fourdee, 2016-07-18:
@Rkieltyka 
Thanks.

Everything looks fine.

By any chance did you have a power cut, or any loss of power on the device, that would not allow a full shutdown or reboot? 
This would affect dietpi-ramlog and could prevent it from saving ```/var/log/*``` to disk. So any changes to ```/var/log/*``` would not be restored after boot.

Rkieltyka, 2016-07-21:
I did not have any power cut or loss of power , when I reboot I do it from command line.

Fourdee, 2016-07-23:
@Rkieltyka 

 > I did not have any power cut or loss of power , when I reboot I do it from command line.

Thanks.

As the original issue is now resolved with your modified service, and, i'am unable to replicate the issue, i'll mark this as closed. 
Hopefully this is a "one off" occurrence, but should a similar issue appear, please reopen the ticket and let us know.

closed 2016-07-23 by Fourdee