MichaIng/DietPi #433: wordpress 404 Not Found nginx

arv187, 2016-07-20:
When I do click in a vídeo obtain  this error (Permalink-postname):
404 Not Found
nginx

Only if I set permalinks in simple work well.

http://tenpi2.ddns.net/phpinfo.php
/etc/nginx/nginx.conf http://pastebin.com/pkahCuMZ
/etc/nginx/wordpress.conf http://pastebin.com/G4hMhcEC

In a support from a plug-in theme they said me this:

> May be related with page codes are fetching or passing.

Zone22, 2016-07-22:
First off always update to the latest from V119 to V123.

Fourdee, 2016-07-22:
@arv187 

Seems to work for me on C1 + DietPi v123:
[image]

I'll test on RPi 2, just to make sure.

arv187, 2016-07-22:
The version of system is old, yesterday I update the dietpi version and update firmware, uninstall wordpress and ftp software with dietpi-uninstall command and after install it again.
Linux DietPi 4.4.15-v7+ #897 SMP Tue Jul 12 18:42:55 BST 2016 armv7l GNU/Linux
I have not change manually nothing, I think.

Fourdee, 2016-07-22:
@arv187 

> Seems that i have addiotionally this /etc/php5/fpm/conf.d/20-gd.ini & /etc/php5/fpm/conf.d/20-mcrypt.ini

Yep, they are fine, just additional php modules.

> uninstall wordpress and ftp software with dietpi-uninstall command and after install it again.

Try selecting all ```WEBSERVER_*``` entries aswell as ```WORDPRESS``` in ```dietpi-uninstall```. Then reinstall with dietpi-software.

Failing that, try checking wordpress is installed, and its permissions:
```
ls -lha /var/www/wordpress
```

And, try checking nginx log file for any clues after receiving the 404:
```
cat /var/log/nginx/error.log
```

As a last resort, reinstall with a fresh image write.

arv187, 2016-07-22:
I think that uninstall module remove a important piece to uninstall mysql database, I had errors.
I cant access to phpmyadmin, when i write my credentials:

```
 #2002 Cannot log in to the MySQL server
Connection for controluser as defined in your configuration failed.
```
I cant continue with wordpress instal if i not know the right database access data.

Fourdee, 2016-07-23:
> #2002 Cannot log in to the MySQL server
Connection for controluser as defined in your configuration failed.

Can you upload the mysql conf file? I think its in ```/etc/mysql/*.conf```

> I cant continue with wordpress instal if i not know the right database access data.

Default username and password for all sql DB's in dietpi is.
user = root
pw = dietpi
I think the mysql conf may be causing this issue.

> Apt-get has failed:    

Possibly holding broken packages or a dependency issue. Try running ```apt-get install -f```

arv187, 2016-07-23:
Yes I try with default login data but and with my new root pass, same error.

My mysql.conf [http://pastebin.com/A8kz0JSE](url)

I go to restore a backup, I will inform you later. :+1:

arv187, 2016-07-24:
In my first attempt (after my first post) I could update update software and firmware, but 404 error still there.
Now I havent network, no connect via wifi (I need upgrade drivers because new kernel) and not connect via ethernet:

1Restoring backup image ok
2updating firmware ok
3reboot ok
4network Not ok

1Restoring backup image ok
2updating software ok
3updating firmware ok
4reboot ok
5network Not ok

I tried to unplugin wifi dongle too, but ethernet still not connecting.

Connecting to tv boot right, or that seems, but not network. There are any file to edit in sd card to create a log file?

arv187, 2016-07-24:
Ok, fixed installation issues at the moment, continuing with 404 error:

```
root@DietPi:~# ls -lha /var/www/wordpress
total 196K
drwxrwxr-x  5 www-data www-data 4.0K Jul 24 18:16 .
drwxr-xr-x  4 www-data www-data 4.0K Jun 19 19:43 ..
-rwxr--r--  1 root     root       39 Jun 20 02:15 .htaccess
-rwxrwxr-x  1 www-data www-data  418 Sep 25  2013 index.php
-rwxrwxr-x  1 www-data www-data  20K Mar  5 20:14 license.txt
-rwxrwxr-x  1 www-data www-data 7.2K Jul 24 18:16 readme.html
-rwxrwxr-x  1 www-data www-data 5.0K Jan 28 03:35 wp-activate.php
drwxrwxr-x 10 www-data www-data 4.0K Jun 19 23:11 wp-admin
-rwxrwxr-x  1 www-data www-data  364 Dec 19  2015 wp-blog-header.php
-rwxrwxr-x  1 www-data www-data 1.5K Jan 30 21:56 wp-comments-post.php
-rw-rw-rw-  1 www-data www-data 3.3K Jun 20 02:15 wp-config.php
-rwxrwxr-x  1 www-data www-data 2.8K Dec 16  2015 wp-config-sample.php
drwxrwxr-x  7 www-data www-data 4.0K Jun 20 02:16 wp-content
-rwxrwxr-x  1 www-data www-data 3.3K May 24  2015 wp-cron.php
drwxrwxr-x 16 www-data www-data  12K May  6 18:33 wp-includes
-rwxrwxr-x  1 www-data www-data 2.4K Oct 24  2013 wp-links-opml.php
-rwxrwxr-x  1 www-data www-data 3.3K Nov  5  2015 wp-load.php
-rwxrwxr-x  1 www-data www-data  34K Mar  6 03:06 wp-login.php
-rwxrwxr-x  1 www-data www-data 7.8K Oct  6  2015 wp-mail.php
-rwxrwxr-x  1 www-data www-data  13K Feb 17 22:58 wp-settings.php
-rwxrwxr-x  1 www-data www-data  28K Jan 28 03:51 wp-signup.php
-rwxrwxr-x  1 www-data www-data 4.0K Nov 30  2014 wp-trackback.php
-rwxrwxr-x  1 www-data www-data 3.0K Oct  2  2015 xmlrpc.php
```
cat /var/log/nginx/error.log
The file is blank.

Fourdee, 2016-07-25:
@arv187 

>cat /var/log/nginx/error.log
The file is blank.

Are you using Ramlog mode 1 (default)?: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68

> + DietPi will automatically clear log files in /var/log every hour (to free memory used by DietPi-Ramlog). 

Try full logging mode to debug log files.

arv187, 2016-08-01:
Now i have select Full log, but error.log still in blank.
Im trying do this, but routes are diferents. [https://www.digitalocean.com/community/questions/enabling-nginx-mod_rewrite](url)

arv187, 2016-08-05:
any idea where i must write that location?

Fourdee, 2016-08-05:
@arv187 

> any idea where i must write that location?

No idea. I'am not Nginx savvy.

To enable logging in Nginx edit ```/etc/nginx/nginx.conf``` as per https://www.nginx.com/resources/admin-guide/logging-and-monitoring/

Warning logging:
```
error_log /var/log/nginx/error.log warn;
```

Access logging, comment the following:
```
#access_log off;
```

```
root@DietPi:~# cat /var/log/nginx/access.log
192.168.0.14 - - [05/Aug/2016:10:27:33 -0400] "GET /phpinfo HTTP/1.1" 404 564 "-" "Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/52.0.2743.116 Safari/537.36"
192.168.0.14 - - [05/Aug/2016:10:27:33 -0400] "GET /favicon.ico HTTP/1.1" 404 564 "http://192.168.0.112/phpinfo" "Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/52.0.2743.116 Safari/537.36"
192.168.0.14 - - [05/Aug/2016:10:27:37 -0400] "GET /index HTTP/1.1" 404 564 "-" "Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/52.0.2743.116 Safari/537.36"
192.168.0.14 - - [05/Aug/2016:10:27:41 -0400] "GET /phpinfo.php HTTP/1.1" 200 74094 "-" "Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/52.0.2743.116 Safari/537.36"
```

### Heres the default DietPi nginx.conf, with access logging and warning logging enabled:
http://pastebin.com/kX47URhi

Fourdee, 2016-08-11:
Marking as closed. Please reopen if required.

closed 2016-08-11 by Fourdee

rayoul, 2017-07-19:
Hi all,
I know that's an old topic but I'm still stuck with that issue.
I've found a way to resolve this permalinks with nginx and dietpi.
You have to edit /etc/nginx/sites-availabe/default
 on the line 
location / {
    try_files $uri $uri/ =404;
}
change to 
location / {
    try_files $uri $uri/ /index.php?$args;
}
but that s not all, on the line
root /var/www
add /worpress:
root /var/www/wordpress

reload nginx
/etc/init.d/nginx reload

That works for me