MichaIng/DietPi #435: DietPi-Software | FireFox Sync 1.5 Server

Zone22, 2016-07-22:
Although this is limited to Firefox it would be nice to find something that worked for Chrome.

 FireFox Sync allows users devices to access Firefox browsing history, bookmarks, and recent tabs.
https://github.com/mozilla-services/syncserver
https://docs.services.mozilla.com/howtos/run-sync-1.5.html#howto-run-sync15

joaofl, 2016-07-26:
I had it once, but I no longer found means of configuring my Firefox with a custom server address. How can it be done?

Fourdee, 2017-05-19:
+1 > http://dietpi.com/phpbb/viewtopic.php?f=12&t=788&p=7246#p7244

molnarti, 2017-05-19:
@joaofl see here https://mozilla-services.readthedocs.io/en/latest/howtos/run-fxa.html

Jlkam, 2017-10-30:
+1 hope we see this soon!

Fourdee, 2017-12-30:
v

closed 2017-12-30 by Fourdee