MichaIng/DietPi #439: DietPi-Config | RPi3  LCD Panel addon: waveshare32 don't work

k-plan, 2016-07-23:
@Fourdee 

- fresh install with auto-update to:

 ```
───────────────────────────────────────
 DietPi     | 18:56 | Sat 23/07/16
 ───────────────────────────────────────
 V124       | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
```

- after activate LCD Panel in `dietpi-config` and reboot

white backlighted  display become black, but no output (hdmi works)

```
root@DietPi:~# dmesg
 . . .
[    3.719136] w1-gpio onewire@0: gpio pin 4, external pullup pin 1, parasitic power 0
[    3.719204] w1_add_master_device: set_pullup requires write_byte or touch_bit, disabling
[    3.874069] random: nonblocking pool is initialized
[    4.175502] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[    4.199541] bcm2835-wdt 3f100000.watchdog: Broadcom BCM2835 watchdog timer
[    4.222061] spi-bcm2835 3f204000.spi: chipselect 1 already in use
[    4.222084] spi_master spi0: spi_device register error /soc/spi@7e204000/waveshare32b_ts@1
[    4.222103] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare32b_ts@1
[    4.222125] spi_master spi0: /soc/spi@7e204000/waveshare35a-ts@1 has no valid 'spi-max-frequency' property (-22)
[    4.222139] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare35a-ts@1
[    4.308417] ads7846 spi0.1: touchscreen, irq 183
[    4.309265] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input2
[    4.346626] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    4.350397] fb_ili9340: module is from the staging directory, the quality is unknown, you have been warned.
[    4.351187] pinctrl-bcm2835 3f200000.gpio: pin gpio17 already requested by spi0.1; cannot claim for spi0.0
[    4.351207] pinctrl-bcm2835 3f200000.gpio: pin-17 (spi0.0) status -22
[    4.351224] pinctrl-bcm2835 3f200000.gpio: could not request pin 17 (gpio17) from group gpio17  on device pinctrl-bcm2835
[    4.351237] fb_ili9340 spi0.0: Error applying setting, reverse things back
[    4.351306] fbtft_of_value: buswidth = 8
[    4.351318] fbtft_of_value: debug = 0
[    4.351328] fbtft_of_value: rotate = 270
[    4.351338] fbtft_of_value: fps = 25
[    4.736480] graphics fb1: fb_ili9340 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=25, spi0.0 at 16 MHz
[    5.761614] w1_master_driver w1_bus_master1: w1_search: max_slave_count 64 reached, will continue next search.
[   53.143310] w1_master_driver w1_bus_master1: Family 0 for 00.800000000000.8c is not registered.
[  112.451047] w1_master_driver w1_bus_master1: Family 0 for 00.400000000000.46 is not registered.
[  159.800975] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.
```

```
root@DietPi:~# lsmod
Module                  Size  Used by
fb_ili9340              3204  0
fbtft                  26084  1 fb_ili9340
evdev                  11396  0
syscopyarea             2945  1 fbtft
joydev                  9024  0
sysfillrect             3443  1 fbtft
sysimgblt               2069  1 fbtft
ads7846                11390  0
fb_sys_fops             1309  1 fbtft
hwmon                   2840  1 ads7846
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
spi_bcm2835             7286  0
w1_gpio                 3657  0
wire                   25219  1 w1_gpio
cn                      4374  1 wire
r8712u                161764  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
fuse                   83461  1
ipv6                  347530  20
```

- after : 

`root@DietPi:~# wget http://www.waveshare.com/w/upload/2/24/LCD-show-160520.tar.gz` 
`tar xvf LCD-show-160520.tar.gz`
` cd LCD-show/ `
`./LCD32-show`

- followed by a auto-reboot

it fire up with DietPi start-up screen.

 `dmesg` output and `lsmod` is exactly the same.

Fourdee, 2016-07-23:
@k-plan 
I did a fresh install of v124 and duplicated your steps, its running fine here.

Did you make any changes to ```config.txt``` before booting up?

Bizarrely, i get the same error messages, although its displaying term fine:
```
root@DietPi:~# dmesg | grep fb1
[   14.318312] graphics fb1: fb_ili9340 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=25, spi0.0 at 16 MHz
root@DietPi:~# dmesg | grep spi
[   13.680055] spi-bcm2835 3f204000.spi: chipselect 1 already in use
[   13.680096] spi_master spi0: spi_device register error /soc/spi@7e204000/waveshare32b_ts@1
[   13.680117] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare32b_ts@1
[   13.680144] spi_master spi0: /soc/spi@7e204000/waveshare35a-ts@1 has no valid 'spi-max-frequency' property (-22)
[   13.680161] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare35a-ts@1
[   13.894871] ads7846 spi0.1: touchscreen, irq 183
[   13.895508] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input0
[   13.901690] pinctrl-bcm2835 3f200000.gpio: pin gpio17 already requested by spi0.1; cannot claim for spi0.0
[   13.901715] pinctrl-bcm2835 3f200000.gpio: pin-17 (spi0.0) status -22
[   13.901745] fb_ili9340 spi0.0: Error applying setting, reverse things back
[   14.318312] graphics fb1: fb_ili9340 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=25, spi0.0 at 16 MHz
```

[image]

And yes, my office is ontop of a freezer lol :+1:

k-plan, 2016-07-23:
@Fourdee 

> Did you make any changes to config.txt before booting up?

no, no burn `DietPi_v120_RPi-armv6-(Jessie).img` to sd-card, put it direct in RPi3 and fire it up ... (tested twice!)

But I will download image again and burn it to sd-card and will repeat the test.

Fourdee, 2016-07-23:
> But I will download image again and burn it to sd-card and will repeat the test.

Thanks @k-plan :+1:  . Please let me know the results.

k-plan, 2016-07-23:
tested again (twice)

- download image and let 7zip test archive (say-i.o.)
- unpack
- md5 sum: DietPi_v120_RPi-armv6-(Jessie).img) = e55125f885392621901e9a5280456dad
- burn to sd card (null changes) and unmount
- put in RPI3, start up, auto update from v.119 => v.125 , auto-reboot
- finish first install dialog without changing anything => auto apt-get update 0> auto-reboot
- `dietpi-config` only activated LCD Panel => exit with auto-reboot

Result: - the same as before

```
root@DietPi:~# cat /var/log/dpkg.log
2016-07-23 20:27:25 startup archives unpack
2016-07-23 20:27:25 install libxau6:armhf <none> 1:1.0.8-1
2016-07-23 20:27:25 status half-installed libxau6:armhf 1:1.0.8-1
2016-07-23 20:27:25 status unpacked libxau6:armhf 1:1.0.8-1
2016-07-23 20:27:25 status unpacked libxau6:armhf 1:1.0.8-1
2016-07-23 20:27:25 install libxdmcp6:armhf <none> 1:1.1.1-1
2016-07-23 20:27:25 status half-installed libxdmcp6:armhf 1:1.1.1-1
2016-07-23 20:27:25 status unpacked libxdmcp6:armhf 1:1.1.1-1
2016-07-23 20:27:25 status unpacked libxdmcp6:armhf 1:1.1.1-1
2016-07-23 20:27:26 install libxcb1:armhf <none> 1.10-3
2016-07-23 20:27:26 status half-installed libxcb1:armhf 1.10-3
2016-07-23 20:27:26 status unpacked libxcb1:armhf 1.10-3
2016-07-23 20:27:26 status unpacked libxcb1:armhf 1.10-3
2016-07-23 20:27:26 install libx11-data:all <none> 2:1.6.2-3
2016-07-23 20:27:26 status half-installed libx11-data:all 2:1.6.2-3
2016-07-23 20:27:27 status unpacked libx11-data:all 2:1.6.2-3
2016-07-23 20:27:27 status unpacked libx11-data:all 2:1.6.2-3
2016-07-23 20:27:27 install libx11-6:armhf <none> 2:1.6.2-3
2016-07-23 20:27:27 status half-installed libx11-6:armhf 2:1.6.2-3
2016-07-23 20:27:27 status unpacked libx11-6:armhf 2:1.6.2-3
2016-07-23 20:27:27 status unpacked libx11-6:armhf 2:1.6.2-3
2016-07-23 20:27:27 install libxext6:armhf <none> 2:1.3.3-1
2016-07-23 20:27:27 status half-installed libxext6:armhf 2:1.3.3-1
2016-07-23 20:27:27 status unpacked libxext6:armhf 2:1.3.3-1
2016-07-23 20:27:27 status unpacked libxext6:armhf 2:1.3.3-1
2016-07-23 20:27:27 install libxi6:armhf <none> 2:1.7.4-1
2016-07-23 20:27:27 status half-installed libxi6:armhf 2:1.7.4-1
2016-07-23 20:27:28 status unpacked libxi6:armhf 2:1.7.4-1
2016-07-23 20:27:28 status unpacked libxi6:armhf 2:1.7.4-1
2016-07-23 20:27:28 install xinput-calibrator:armhf <none> 0.7.5+git20140201-1
2016-07-23 20:27:28 status half-installed xinput-calibrator:armhf 0.7.5+git20140201-1
2016-07-23 20:27:28 status unpacked xinput-calibrator:armhf 0.7.5+git20140201-1
2016-07-23 20:27:28 status unpacked xinput-calibrator:armhf 0.7.5+git20140201-1
2016-07-23 20:27:28 startup packages configure
2016-07-23 20:27:28 configure libxau6:armhf 1:1.0.8-1 <none>
2016-07-23 20:27:28 status unpacked libxau6:armhf 1:1.0.8-1
2016-07-23 20:27:28 status half-configured libxau6:armhf 1:1.0.8-1
2016-07-23 20:27:28 status installed libxau6:armhf 1:1.0.8-1
2016-07-23 20:27:28 status triggers-pending libc-bin:armhf 2.19-18+deb8u4
2016-07-23 20:27:28 configure libxdmcp6:armhf 1:1.1.1-1 <none>
2016-07-23 20:27:28 status unpacked libxdmcp6:armhf 1:1.1.1-1
2016-07-23 20:27:28 status half-configured libxdmcp6:armhf 1:1.1.1-1
2016-07-23 20:27:28 status installed libxdmcp6:armhf 1:1.1.1-1
2016-07-23 20:27:28 configure libxcb1:armhf 1.10-3 <none>
2016-07-23 20:27:28 status unpacked libxcb1:armhf 1.10-3
2016-07-23 20:27:28 status half-configured libxcb1:armhf 1.10-3
2016-07-23 20:27:28 status installed libxcb1:armhf 1.10-3
2016-07-23 20:27:28 configure libx11-data:all 2:1.6.2-3 <none>
2016-07-23 20:27:28 status unpacked libx11-data:all 2:1.6.2-3
2016-07-23 20:27:28 status half-configured libx11-data:all 2:1.6.2-3
2016-07-23 20:27:28 status installed libx11-data:all 2:1.6.2-3
2016-07-23 20:27:28 configure libx11-6:armhf 2:1.6.2-3 <none>
2016-07-23 20:27:28 status unpacked libx11-6:armhf 2:1.6.2-3
2016-07-23 20:27:28 status half-configured libx11-6:armhf 2:1.6.2-3
2016-07-23 20:27:28 status installed libx11-6:armhf 2:1.6.2-3
2016-07-23 20:27:28 configure libxext6:armhf 2:1.3.3-1 <none>
2016-07-23 20:27:28 status unpacked libxext6:armhf 2:1.3.3-1
2016-07-23 20:27:28 status half-configured libxext6:armhf 2:1.3.3-1
2016-07-23 20:27:28 status installed libxext6:armhf 2:1.3.3-1
2016-07-23 20:27:28 configure libxi6:armhf 2:1.7.4-1 <none>
2016-07-23 20:27:28 status unpacked libxi6:armhf 2:1.7.4-1
2016-07-23 20:27:28 status half-configured libxi6:armhf 2:1.7.4-1
2016-07-23 20:27:28 status installed libxi6:armhf 2:1.7.4-1
2016-07-23 20:27:28 configure xinput-calibrator:armhf 0.7.5+git20140201-1 <none>
2016-07-23 20:27:28 status unpacked xinput-calibrator:armhf 0.7.5+git20140201-1
2016-07-23 20:27:28 status half-configured xinput-calibrator:armhf 0.7.5+git20140201-1
2016-07-23 20:27:28 status installed xinput-calibrator:armhf 0.7.5+git20140201-1
2016-07-23 20:27:28 trigproc libc-bin:armhf 2.19-18+deb8u4 <none>
2016-07-23 20:27:28 status half-configured libc-bin:armhf 2.19-18+deb8u4
2016-07-23 20:27:28 status installed libc-bin:armhf 2.19-18+deb8u4
2016-07-23 20:27:29 startup packages configure
```

k-plan, 2016-07-23:
@Fourdee 

```
root@DietPi:~# cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop fsck.repair=yes rootwait
```

new and working:

```
root@DietPi:~# cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop fsck.repair=yes rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
```

k-plan, 2016-07-23:
@Fourdee 

- tested: switching to 'none' - reboot - white display
- tested: switching  form 'none' to 'waveshare32'  - reboot - login screen on LCD display

work now - done!

Fourdee, 2016-07-24:
@k-plan 

> https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-set_hardware#L165

Yep, i couldn't figure out a way of adding that to the end of the existing line, so it uses a pattern match.

> work now - done!

Excellent :+1: 

> Your bash scripts are really very cool - but understanding, reading and searching (e.g. typos) is living hell (pain in the ass).

Lol, yep. No spell checker in notepad++ at the moment. Probably a good time for me to try a plugin :)

I'll mark this as closed, but please reopen as needed.

closed 2016-07-24 by Fourdee

k-plan, 2016-07-24:
@Fourdee 

I think, you have miss problem/solution. 

You have a typo in your scripts (branch `master` as well in `testing`):

https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-set_hardware#L165

https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-set_hardware#L165

Fourdee - `dietpi-set_hardware` (_not working - please test it by yourself_):
```
#consoleblank=0 no effect 
### NOT WORK! ###
sed -i 's/rootwait /rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo /' /boot/cmdline.txt
```

must be (working - tested):

```
#consoleblank=0 no effect 
### work! ###
sed -i 's/rootwait/rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo /' /boot/cmdline.txt
```

The `blank` or `space` between `'s/rootwait     /rootwait fb,,,.` cause the problem.

-----

Testing (_please do yourself_):
```
root@DietPi:~# echo -e "dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop fsck.repair=yes rootwait" >> /root/cmdline.txt

root@DietPi:~# cat cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop fsck.repair=yes rootwait

root@DietPi:~# sed -i 's/rootwait /rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo /' /root/cmdline.txt

root@DietPi:~# cat cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop fsck.repair=yes rootwait

root@DietPi:~# sed -i 's/rootwait/rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo /' /root/cmdline.txt

root@DietPi:~# cat cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop fsck.repair=yes rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo

```

Or please tell me, how can it be explained better?

Fourdee, 2016-07-25:
@k-plan 

Got ya :) Spot on :+1:

```
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_hardware lcdpanel waveshare32 &> /dev/null
root@DietPi:~# cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop fsck.repair=yes rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo rootdelay=10
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_hardware lcdpanel none &> /dev/null
root@DietPi:~# cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop fsck.repair=yes rootwait rootdelay=10
```

Fixed with commit below.