MichaIng/DietPi #440: General | FStrim causing corrupt filesystem

Fourdee, 2016-07-23:
Although unproven, since this has been running as a daily task for our users, the number of filesystem corruptions has increased.

Fourdee, 2016-07-24:
Disabled as of v125.

Note:
Requires testing to prove stability before we add this back in.

Fourdee, 2016-09-28:
Leaving this as disabled. Not worth the risk of creating possible filesystem corruption to users system.

closed 2016-09-28 by Fourdee

TheOriginalMrWolf, 2021-02-03:
Hi,

Almost 4½ years later and it seems that fstrim is still disabled (DietPi v6.34.3, on Odroid N2) - any plans to re-enable on, say, a weekly basis?  Not having this causes significant slowdowns & additional wear due to write amplification.

I would hope that the underlying issue is resolved by now?

Thanks!