MichaIng/DietPi #442: Gogs don't work after install.

stdusm, 2016-07-23:
blank page on http://192.168.0.100:3000/.
output of **cat gogs_daemon.log**:

> 2016/07/24 02:38:05 [W] Custom config '/etc/gogs/custom/conf/app.ini' not found, ignore this if you're running first time
> 2016/07/24 02:38:05 [...s/setting/setting.go:367 NewContext()] [E] Fail to test 'ssh-keygen' command: exec: "ssh-keygen": executable file not found in $PATH (forgotten install?)

Fourdee, 2016-07-23:
@stdusm 

Thanks for the report :+1: 

I think this is due to you running Dropbear and the ```ssh-keygen``` binary might only be installed with OpenSSh server.

Can you paste the results of:
```
find / -type f -name ssh-keygen*
```

And
```
cat /DietPi/dietpi/.installed | grep 2
```

k-plan, 2016-07-23:
running Dropbear

```
root@DietPi:~# find / -type f -name ssh-keygen*
root@DietPi:~#
```

```
root@DietPi:~# cat /DietPi/dietpi/.installed | grep '=2$'
WEBSERVER_MYSQL=2
GOGS=2
MIDNIGHTCOMMANDER=2
GITCLIENT=2
SSHSERVER_DROPBEAR=2
LOGGING_RAMLOG=2
```

k-plan, 2016-07-23:
running OpenSSH:

```
root@DietPi:~# find / -type f -name ssh-keygen*
/usr/bin/ssh-keygen
/usr/share/man/man1/ssh-keygen.1.gz

root@DietPi:~# cat /DietPi/dietpi/.installed | grep '=2$'
WEBSERVER_MYSQL=2
GOGS=2
SSHCLIENT=2
MIDNIGHTCOMMANDER=2
GITCLIENT=2
SSHSERVER_OPENSSH=2
LOGGING_RAMLOG=2
```

[image]

stdusm, 2016-07-24:
**root@MediaServer:~# find / -type f -name ssh-keygen***

> root@MediaServer:~#

**root@MediaServer:~# cat /DietPi/dietpi/.installed | grep '=2$'****

> USBDRIVE=2
> DESKTOP_LXDE=2
> WEBSERVER_LLMP=2
> WEBSERVER_LLSP=2
> WEBSERVER_LIGHTTPD=2
> WEBSERVER_MYSQL=2
> WEBSERVER_SQLITE=2
> WEBSERVER_PHP=2
> WEBSERVER_MYADMINPHP=2
> TIGHTVNCSERVER=2
> OWNCLOUD=2
> MINIDLNA=2
> DELUGE=2
> FILESERVER_PROFTP=2
> LINUXDASH=2
> SUBSONIC6=2
> FAIL2BAN=2
> WIFIHOTSPOT=2
> GOGS=2
> GITCLIENT=2
> ALSABASE=2
> XSERVERXORG=2
> FFMPEG=2
> ORACLEJAVA=2
> SSHSERVER_DROPBEAR=2
> LOGGING_RAMLOG=2
> LOGGING_RSYSLOG=2

stdusm, 2016-07-24:
Installing OpenSSH solved the problem, thanks a lot.

Fourdee, 2016-07-24:
@stdusm 

> Installing OpenSSH solved the problem, thanks a lot.

Excellent :+1:.

Note to self: message prompt if dropbear.

k-plan, 2016-07-24:
> Note to self: message prompt if dropbear.

... _or deinstall dropbear and install openSSH, if gogs is selected for installation_ 😉

Fourdee, 2016-07-24:
Done:
[image]

Also updated gogs installation to use latest binary download links: https://dl.gogs.io/

closed 2016-07-24 by Fourdee