MichaIng/DietPi #443: Wifi Country Code change error

stdusm, 2016-07-24:
Always get: **"Country code (GB/US/etc..) could not been applied. Please check the country code and try again."**

**root@MediaServer:/etc/hostapd# cat hostapd.conf** 

> interface=wlan1
> driver=nl80211
> ssid=TestHotSpot
> hw_mode=g
> channel=3 
> macaddr_acl=0
> auth_algs=1
> ssid=TestHotSpot
> wpa=2
> wpa_passphrase=mySuperPassword
> wpa_key_mgmt=WPA-PSK
> wpa_pairwise=TKIP
> rsn_pairwise=CCMP

Adding line "_country_code=GB_" to _/etc/hostapd/hostapd.conf_ + services restart didn't solved the problem.
**Note:** I am using Banana Pi M3 (bcmdhd).

Fourdee, 2016-07-25:
@stdusm 

> Banana Pi M3

Which image did you use?

> Adding line "country_code=GB" to /etc/hostapd/hostapd.conf + services restart didn't solved the problem.

Try:
```
iw reg set GB
sleep 2
iw reg get
```

> ```interface=wlan1```

It may be wlan0 is preventing the reg change. Find the module ap621* in /etc/modules and comment it out:
```
lsmod
```Reboot.

Check if wlan1 is now wlan0:
```
ifconfig -a | grep wlan
```

Change ```interface=wlan0``` and restart services.

stdusm, 2016-07-25:
@Fourdee 

> Which image did you use? 

From here http://www.banana-pi.org/m3-download.html

------------------------------------------

> Try:

> iw reg set GB
> sleep 2
> iw reg get

Result :

```
root@MediaServer:/mnt/dietpi_userdata/.scripts/utils# iw reg set GB
root@MediaServer:/mnt/dietpi_userdata/.scripts/utils# sleep 2
root@MediaServer:/mnt/dietpi_userdata/.scripts/utils# iw reg get
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 40), (6, 20), (N/A), NO-IR
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, NO-IR
        (5170 - 5250 @ 40), (6, 20), (N/A), NO-IR
        (5735 - 5835 @ 40), (6, 20), (N/A), NO-IR
```

------------------------------------------

> Find the module ap621* in /etc/modules and comment it out:

No module a621, I used **modprobe bcmdhd** to enable wifi.

------------------------------------------

> lsmod

```
root@MediaServer:~# lsmod
Module                  Size  Used by
xt_multiport            2631  2 
xt_conntrack            3189  1 
iptable_filter          1710  1 
ipt_MASQUERADE          2384  1 
iptable_nat             5216  1 
nf_nat                 19859  2 ipt_MASQUERADE,iptable_nat
nf_conntrack_ipv4      15796  4 nf_nat,iptable_nat
nf_conntrack           91636  5 ipt_MASQUERADE,nf_nat,xt_conntrack,iptable_nat,nf_conntrack_ipv4
nf_defrag_ipv4          1571  1 nf_conntrack_ipv4
ip_tables              12777  2 iptable_filter,iptable_nat
x_tables               19160  6 ip_tables,ipt_MASQUERADE,xt_conntrack,xt_multiport,iptable_filter,iptable_nat
option                 20034  0 
usb_wwan               12345  1 option
bcmdhd                693235  0 
```

------------------------------------------

It seems that problem is in image or hardware that I am using, apologize for wasting your time with that

closed 2016-07-26 by stdusm