MichaIng/DietPi #445: Blynk Server for IoT

joaofl, 2016-07-26:
A very nice initiative for IoT is called [Blynk](http://www.blynk.cc/), and they provide their server code to be run in you own server for privacy concerns. Have it installed at my dietpi xu4, and would be great to have it listed at dietpi-software.

They provide an Android iOS app to interface with your raspberry/arduino/esp8266, through an onlline server of yours.

it requires java (openjdk8), and almost none initial setup

http://docs.blynk.cc/#blynk-server-requirements

oshank, 2017-01-22:
I second this request. Been looking into it for the last few days. I think it would be pretty cool to see this on the list. Should be relatively simple to set up I believe.

Fourdee, 2017-03-01:
Fails connection test...
```
			INSTALL_URL_ADDRESS='https://github.com/blynkkk/blynk-server/releases/download/v0.22.3/server-0.22.3.jar'
			/DietPi/dietpi/func/check_connection "$INSTALL_URL_ADDRESS"
```

Fourdee, 2017-03-01:
online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=5901#p5901

Fourdee, 2017-03-01:
Commit: https://github.com/Fourdee/DietPi/commit/045284afc43488822e83c549de2bb924362eaa32

Fourdee, 2017-03-02:
Just about to finish off the online doc, http://blynk.cc is down lol, typical.

Fourdee, 2017-03-02:
Completed.

joaofl, 2017-03-15:
@Fourdee 

I was testing the Blynk server today, but could not access the admin page from a remote pc, because by default, it is enable to local only. 

Also, by not having a config file, limits on doing other customizations.
So I did some minor changes to the dietpi-software install script to address this issue. 

I only missed adding a line to replace a line in the config file (sed -i   afraid of messing around).

I'm not exactly sure if I proposed the patches using the correct procedure, but anyway, I hope it helps.

Fourdee, 2017-03-15:
@joaofl 

Thanks for the report and changes 👍 

> So I did some minor changes to the dietpi-software install script to address this issue.

Would you be able to send a pull request for the changes, here on git? Please use the following **testing** branch when sending the pull request:
https://github.com/Fourdee/DietPi/tree/testing

joaofl, 2017-03-15:
I think I did it right this time.

https://github.com/Fourdee/DietPi/pull/810

Fourdee, 2017-03-18:
@joaofl 
Awesome 👍 
[image]

I'll update online docs once our hosting provider is back online, then mark this as completed.

Fourdee, 2017-03-18:
Online doc updated: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=5901#p5901

Great work @joaofl, marking this as completed 👍

closed 2017-03-18 by Fourdee

joaofl, 2017-03-19:
@Fourdee ,thanks for that. I hope I can contribute further on. :+1::+1::+1: