MichaIng/DietPi #448: Web Dashboard for DietPi

Fourdee, 2016-07-27:
Notes:
- Must not require a webserver stack (implies self contained website)

rhkean, 2016-07-27:
brainstorming.....  this can probably be done with python.  Couchpotato and NzbHydra use python and have decent simple web interfaces.  But, so does netdata and it's written in C.

tr0yspradling, 2016-07-28:
I plan on taking a stab at this with Python/Flask.

Fourdee, 2016-07-28:
@tr0yspradling 

> I plan on taking a stab at this with Python/Flask.

Excellent :+1: 

We will need to change alot of the DietPi code to accommodate the web interface, but either myself or @rhkean can do this alongside @tr0yspradling work. 
If you have any progress, let us know.

Fourdee, 2016-07-28:
@rhkean 

> But, so does netdata and it's written in C.

YMPD is also written in C: https://github.com/notandy/ympd
> Standalone MPD Web GUI written in C, utilizing Websockets and Bootstrap/JS http://www.ympd.org

WolfganP, 2016-07-28:
Great idea. I think that the system will benefit more if it's designed extensible from the start, so it can be tailored for custom system commands or headless operation (ie one button remote system shutdown/restart, modules commands -ie minidlna db rescan-, etc)

Fourdee, 2016-07-28:
@WolfganP 

Yep, ideally i'd like a system where users can either use the web dash, or, terminal for all DietPi features. But I/we will need help with this, I can do the bash/unix side, but i lack the skills for the web side.

rhkean, 2016-07-28:
> @WolfganP 
> Yep, ideally i'd like a system where users can either use the web dash, or, terminal for all DietPi features.

terrific idea.  The trick, I think, will come in finding a way to do this that maintains the small footprint that is at the heart of DietPi

WolfganP, 2016-07-28:
Ideally it should just be a different UI for using the existing installation / config scripts, with the ability to add user scripts / commands / recipes somehow (to avoid reinventing the wheel and having to reimplement the current / tuned dietpi-* specific scripts).

tr0yspradling, 2016-07-29:
I've created [DietPi-Dashboard](https://github.com/tr0yspradling/DietPi-Dashboard). 

I've decided to use CherryPy instead of Flask, but I'm curious, should we reverse proxy through NGINX or use the standalone server with CherryPy?

Fourdee, 2016-07-29:
@tr0yspradling 

That was quick, looking forward to seeing it progress :+1: 

> I've decided to use CherryPy instead of Flask, but I'm curious, should we reverse proxy through NGINX or use the standalone server with CherryPy?

I would prefer it didn't required a webserver, self contained in a standalone server would be ideal.

Fourdee, 2016-07-29:
@WolfganP 

> Ideally it should just be a different UI for using the existing installation / config scripts, with the ability to add user scripts / commands / recipes somehow (to avoid reinventing the wheel and having to reimplement the current / tuned dietpi-* specific scripts).

Yep. 
To start off with, we'll start small and see what the dash can do. In time, we can always add more to it. I have this vision of making DietPi completely web interfaced, but at this time, its just an idea. We'll see how it goes :)

tr0yspradling, 2016-07-29:
Right now I'm working on syncing live system information with [python-socketio](https://github.com/miguelgrinberg/python-socketio) and displaying it with [Chart.js](https://github.com/chartjs/Chart.js). Once I finish that, I'm going to add process & software package management.

k-plan, 2016-08-23:
Hi,

are where any news? Thanks.

Fourdee, 2016-08-24:
Note to self: 
- https://www.webtoolkit.eu/wt
- http://cppcms.com/wikipp/en/page/main

tr0yspradling, 2016-08-25:
I'll be updating the repository tonight. I'll post back here when I do.

tr0yspradling, 2016-08-25:
@Fourdee I considered using [Crow](https://github.com/ipkn/crow).

tr0yspradling, 2016-08-25:
I haven't been able to get as much done as I hoped, I had another project taking up most of my time. 

Check out [DietPi-Dashboard](https://github.com/tr0yspradling/DietPi-Dashboard) to see what's there. It needs cleaned up and reorganized a little, some of the major features aren't implemented yet.

Fourdee, 2016-08-25:
@tr0yspradling

> time
 
No worries :+1: 

> Check out DietPi-Dashboard to see what's there. It needs cleaned up and reorganized a little, some of the major features aren't implemented yet.

I'am a bit of a Python "n00b" (Its like reading an Alien language for me, with a few English letters and numbers in-between lol) , but i'll give it a whirl, ```run.py``` to launch?

Fourdee, 2016-08-25:
Notes: Quick install deps and run:
```
apt-get install -y python python3 #python-socketio?
wget https://github.com/tr0yspradling/DietPi-Dashboard/archive/master.zip -O package.zip
unzip package.zip
rm package.zip
chmod +x -R DietPi-Dashboard-master
DietPi-Dashboard-master/run.py
```

tr0yspradling, 2016-08-25:
The application depends on several python packages. The python package index can be installed with
`sudo apt install python-pip`
and the dependencies can be installed by running
`sudo pip install -r requirements.txt`

I'm creating a virtualenv and startup script for it now. I'll update when I've finished my modifications.

tr0yspradling, 2016-08-25:
Check [DietPi-Dashboard](https://github.com/tr0yspradling/DietPi-Dashboard). Executing `./start` should work.

Fourdee, 2016-08-26:
@tr0yspradling 

Tried on my test rig (Odroid C2, ARM64):
```
root@DietPi:~/DietPi-Dashboard-master# ./start
/usr/local/lib/python2.7/dist-packages/flask/exthook.py:71: ExtDeprecationWarning: Importing flask.ext.socketio is deprecated, use flask_socketio instead.
  .format(x=modname), ExtDeprecationWarning
Traceback (most recent call last):
  File "run.py", line 10, in <module>
    from project.server import sioServer
  File "/root/DietPi-Dashboard-master/project/server/__init__.py", line 3, in <module>
    from dashboard.views import dashboard_routes
  File "/root/DietPi-Dashboard-master/project/server/dashboard/views.py", line 9, in <module>
    from diagnosticsReport import DiagnosticsReport
  File "/root/DietPi-Dashboard-master/project/server/dashboard/diagnosticsReport.py", line 2, in <module>
    import cpuinfo
  File "/usr/local/lib/python2.7/dist-packages/cpuinfo/__init__.py", line 5, in <module>
    from cpuinfo import *
  File "/usr/local/lib/python2.7/dist-packages/cpuinfo/cpuinfo.py", line 1553, in <module>
    _check_arch()
  File "/usr/local/lib/python2.7/dist-packages/cpuinfo/cpuinfo.py", line 1510, in _check_arch
    raise Exception("py-cpuinfo currently only works on X86 and some ARM CPUs.")
Exception: py-cpuinfo currently only works on X86 and some ARM CPUs.
```

I'll power up a RPi and try it.

Fourdee, 2016-08-26:
@tr0yspradling 

I love where this is going. This is exactly what DietPi needs. Great work :+1:
[image]

tr0yspradling, 2016-08-26:
I'll have to go through and check the supported architectures for each supported platform. I didn't think of that. 

I should have done this from the start, but I'll go through and pull the data from the special file: `/proc/cpuinfo`

tr0yspradling, 2016-08-26:
We should start reporting issues to the dashboard repository so it's more organized.

Fourdee, 2016-08-26:
@tr0yspradling 

> We should start reporting issues to the dashboard repository so it's more organized.

Good call :+1: 
I'd like to add a few tickets with features I'd like to see, and discuss their implementation if thats ok?

tr0yspradling, 2016-08-26:
Absolutely. Check your email!

WolfganP, 2016-08-26:
Looks great guys, good work

bhaveshgohel, 2016-08-27:
@tr0yspradling Neat and clean, that's what needed! Awesome! :+1:

WolfganP, 2016-08-29:
@Fourdee Looking at your instructions, I think it's easier to install (and keep updated) via git

```
git clone https://github.com/tr0yspradling/DietPi-Dashboard.git
cd DietPi-Dashboard
pip install -r requirements.txt
./start
```
Anyways, I assume that the way to access the Dashboard is via a desktop browser and http://raspi-ip:9090 ?

Fourdee, 2016-08-30:
@WolfganP 

> I think it's easier to install (and keep updated) via git

Yep. 
We prefer to download the zip instead of using git clone in ```dietpi-software```. Reason being additional packages are required to use ```git clone ```.

> http://raspi-ip:9090

Yep, thats the plan. I think the current port is 8080, but this may change.

Fourdee, 2016-09-04:
@tr0yspradling 
Looks like you deleted your repo for this? https://github.com/tr0yspradling/DietPi-Dashboard

Heres the C++ version i've started: https://github.com/Fourdee/DietPi-Dashboard
Managed to pull all the DietPi-Software database into c++. Can now install software from it. Time to start on the web side.

mknrls, 2017-01-28:
@Fourdee so you still want the dashboard to be  self contained meaning without using a webserver?

Fourdee, 2017-01-28:
@mknrls 

> so you still want the dashboard to be self contained meaning without using a webserver?

Yep, self contained, that was the plan, try and keep it minimal. But at this stage, I am up for anything.

Fourdee, 2017-12-30:
As much as I'd love to see this in DietPi, we currently lack the man hours to achieve this. Just not feasible at the moment.

Closing.

ravenclaw900, 2021-08-17:
I could try and create this with Go and Svelte.  I'll probably use TailwindCSS instead of Bootstrap, though.

MichaIng, 2021-08-17:
Interesting find, I never knew that there has already been some work done on this idea, sadly discontinued. Would be awesome to revive it, basically some sort of webmin tailored for DietPi, which can potentially inherit the features of the Allo GUI as well, which, to be honest, requires a major rework to fit modern standards, also security wise. But while I learned to do HTML, CSS, JavaScript etc, I have not much experience with any real programming language, neither C++, not Go, while I usually learn and understand it fast, when needed.

Yep, Go seems to generally fit well, or Rust (while then I'd even less of help, given the larger syntax and logic differences 😄)? No preference on the style framework on my end. While our website uses bootstrap, I actually aim to remove it, when I find time. We use only a minimal subset and overwrite many of the defaults, so much wasted code we load into browsers. But first and easier will be to eliminate jQuery 😃.

ravenclaw900, 2021-08-17:
> We use only a minimal subset and overwrite many of the defaults

That's why I prefer Tailwind, it's a utility framework instead of a component, and it supports tree-shaking.

ravenclaw900, 2021-08-21:
Note to anyone that wants to compile this, make sure that you have Yarn and Go installed, then just run `make` in the project root.  This should compile a binary for your platform, however I wrote it with Linux/Unix commands, so I'm not sure how it will run on Windows.

ravenclaw900, 2021-09-02:
@MichaIng, @Joulinar, @fpetru, @StephanStS, could you review?  I plan to also add a management page, with generic tasks like shutting down and restarting the system, and a terminal.  Also, currently I'm just using the generic Tailwind lime colors, if you have a specific color code you'd like me to use please let me know.

Joulinar, 2021-09-02:
for colour code, I would recommend to use the same as on our web side/ docs.

Basically I like the idea. Especially to have a web terminal. 😃

GvY85, 2021-09-02:
This sounds like a great idea 👍 .

@ravenclaw900 can you perhaps write a (very minimal, only a list of cli commands is enough) manual on how to install/use this so more people can test/try/give feedback.

ravenclaw900, 2021-09-02:
Prereq:
```sh
dietpi-software install 9 16 17 188 # Install Node.js (webpage), Build-Essential (make), Git (git clone), and Go (binary), respectively
npm install -g yarn # Install Yarn package manager, for node dependencies
```
Now log out and log back in to register GOPATH
Compiling:
```sh
rm -rf DietPi-Dashboard # Remove possibly pre-download repository
git clone https://github.com/ravenclaw900/DietPi-Dashboard # Download source code
cd DietPi-Dashboard/src/frontend # Change directories
yarn # Install dependencies
cd ../.. # Change directories
make # Compile binary for your platform
./dietpi-dashboard # Run binary
```
Now just visit http://device.IP:8080.  Note that for DietPi-Software installs to work, the binary needs to be run as root.

Wow, I thought that it was a bug in the program that RAM and CPU were almost 0, stock DietPi is *really* lightweight.

Joulinar, 2021-09-02:
Nice, I did a test installation. Some observation

- process overview seems to be a heavy view. At least on my test VM it's causing high CPU usage
- I did 2 test installations from software catalogue, The output window on the bottom seems to be very small. It's quite hard to read the output.
- after software installations, the other 2 options `Statistics` and `Processes` are not working anymore. It's needed to perform a full fresh of the web page. This are the messages during page reload

```
2021/09/03 01:27:23 http: panic serving 192.168.0.53:56438: interface conversion: interface {} is nil, not []interface {}
goroutine 45 [running]:
net/http.(*conn).serve.func1()
        /usr/local/go/src/net/http/server.go:1801 +0xb9
panic({0x673720, 0xc00014e000})
        /usr/local/go/src/runtime/panic.go:1047 +0x266
github.com/ravenclaw900/DietPi-Dashboard/lib.ServeWebsockets({0x758fc8, 0xc0000c8380}, 0x0)
        /root/DietPi-Dashboard/src/backend/lib/sockets.go:107 +0x898
net/http.HandlerFunc.ServeHTTP(0x0, {0x758fc8, 0xc0000c8380}, 0x0)
        /usr/local/go/src/net/http/server.go:2046 +0x2f
net/http.(*ServeMux).ServeHTTP(0x0, {0x758fc8, 0xc0000c8380}, 0xc0002f3900)
        /usr/local/go/src/net/http/server.go:2424 +0x149
net/http.serverHandler.ServeHTTP({0xc00014ec00}, {0x758fc8, 0xc0000c8380}, 0xc0002f3900)
        /usr/local/go/src/net/http/server.go:2878 +0x43b
net/http.(*conn).serve(0xc000080320, {0x75a060, 0xc00006ef60})
        /usr/local/go/src/net/http/server.go:1929 +0xb08
created by net/http.(*Server).Serve
        /usr/local/go/src/net/http/server.go:3033 +0x4e8
```

Tested on Bullseye VM amd64

ravenclaw900, 2021-09-03:
1. Yeah, that's a bit resource heavy, I'm trying to figure that out
2. Enlarged window height
3. It kept reading messages into the same struct, without trying to overwrite them, thus it was probably still trying to un/install software whenever the page was changed.  Fixed
4. The graph is being generated in the browser, and it will keep going until it's closed

ravenclaw900, 2021-09-04:
I just pushed a commit with a terminal window, let me know how it looks.

ravenclaw900, 2021-09-06:
Alright, I added a few more features, and I'm hopeful for a `0.1.0` release.

Joulinar, 2021-09-06:
ok did some testing. looks quite nice. Joining the `terminal` view is giving following

2021/09/06 15:08:46 Request to /ws/term (terminal)
2021/09/06 15:09:31 Couldn't get data from frontend: websocket: close 1005 (no status)
2021/09/06 15:09:37 Request to /ws/term (terminal)
2021/09/06 15:09:39 Couldn't get data from frontend: websocket: close 1005 (no status)
2021/09/06 15:09:40 Request to /ws/term (terminal)
2021/09/06 15:09:41 Couldn't get data from frontend: websocket: close 1005 (no status)
2021/09/06 15:09:48 Request to /ws/term (terminal)
2021/09/06 15:09:49 Couldn't get data from frontend: websocket: close 1005 (no status)
2021/09/06 15:09:52 Request to /ws/term (terminal)
2021/09/06 15:09:54 Couldn't get data from frontend: websocket: close 1005 (no status)

ravenclaw900, 2021-09-06:
That's normal, it closes the connection when you leave the terminal page (as long as it's only happening when you leave the page).  I'll make it use the normal exit code.

ravenclaw900, 2021-09-06:
Version 0.1.0 released, see https://github.com/ravenclaw900/DietPi-Dashboard/releases/tag/v0.1.0 for precompiled binaries.

ravenclaw900, 2021-09-09:
I might try to rewrite part of the backend in Rust, just to see how it compares (Rust is supposed to be faster, use less memory, and have smaller binary sizes).  If it's better by a sizeable amount, I might consider switching the whole thing.

ravenclaw900, 2021-09-18:
New compilation instructions:
Prereq:
```sh
dietpi-software install 9 16 17 # Install Node.js (webpage), Build-Essential (make and gcc), and Git (git clone), respectively
npm install -g yarn # Install Yarn package manager, for node dependencies
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh # Install Rust (backend)
```

Compiling:
```sh
rm -rf DietPi-Dashboard # Remove possibly pre-download repository
git clone https://github.com/ravenclaw900/DietPi-Dashboard # Download source code
cd DietPi-Dashboard/src/frontend # Change directories
yarn # Install dependencies
cd ../.. # Change directories
make # Compile binary for your platform
./dietpi-dashboard # Run binary
```

ravenclaw900, 2021-09-18:
@Joulinar, could you test the new Rust binary. Performance seems to have improved greatly on things like the Processes page, for me from `103%` to `12%` CPU.

Joulinar, 2021-09-22:
@ravenclaw900 
there is an error message running `make`

```
root@DietPi4:/tmp/DietPi-Dashboard# make
cd src/frontend; yarn build
yarn run v1.22.11
$ rollup -c

src/main.ts → public/build/bundle.js...

warn - You have enabled the JIT engine which is currently in preview.
warn - Preview features are not covered by semver, may introduce breaking changes, and can change at any time.
(!) Plugin replace: @rollup/plugin-replace: 'preventAssignment' currently defaults to false. It is recommended to set this option to `true`, as the next major version will default this option to `true`.
created public/build/bundle.js in 51.1s
Done in 53.11s.
rustfmt --edition 2018 src/backend/src/*
/bin/sh: 1: rustfmt: not found
make: *** [Makefile:27: fmt] Error 127
root@DietPi4:/tmp/DietPi-Dashboard#
```

ravenclaw900, 2021-09-22:
Interesting, that should be installed when installing Rust. Could you try `rustup component add rustfmt`?

Joulinar, 2021-09-22:
way better now as you added the `source` line.

that's resource consumption while process view is open. Running on RPi4B 8GB, Bullseye, ARMv8

[image]

ravenclaw900, 2021-09-25:
@MichaIng, there's dark mode now.

ravenclaw900, 2021-09-26:
Version 0.2.0 released, see https://github.com/ravenclaw900/DietPi-Dashboard/releases/tag/v0.2.0 for precompiled binaries.

ravenclaw900, 2021-09-27:
Other than a file browser, I can't think of any other core features (other than `dietpi-config`, but it would require the whiptail to be parsed) that would need to be added. Any ideas?

Joulinar, 2021-09-28:
hmm what about the possibility to view status of DietPi services like `dietpi-services status`. Or to show current DietPi version + available updates, as well available apt package updates?

[19 more replies]

closed 2021-11-11 by MichaIng