MichaIng/DietPi #452: NewEgg Deal...

rhkean, 2016-07-28:
http://promotions.newegg.com/neemail/guerrilla/LP/16-Jul/index-landing_wave_27.html

Qty: 3 U1 - uSD - 16Gb  $10.99 + $2.88 shipping

@Fourdee not sure if you can take advantage of this
@xenfomation 
@k-plan

soooo.... totally abusing the Issues functionality of github here.... :laughing:

Fourdee, 2016-07-29:
@rhkean HMRC (the tax man) can take advantage of this if I was to buy one. lol

I personally use for all DietPi testing (excluding the Odroids, due to EMMC being miles faster):
Sandisk Extreme 16GB UHS 3. 
- 45MB/s~ write speed. 
- 90MB/s~ read speed
- £10.50 on amazon uk:
https://www.amazon.co.uk/SanDisk-Extreme-microSDHC-Frustration-Packaging/dp/B013P1M49A/ref=sr_1_1?ie=UTF8&qid=1469788948&sr=8-1&keywords=sandisk+extreme+16gb+microsd

I have 3 of those cards in use for roughly 8+ months, they take a hammering but still work as if new.

rhkean, 2016-07-29:
I use Samsung Pro's on my "production" systems.  On my test setups, unless I'm doing a performance or speed test, I use cheaper cards.  

I have several ADATA uSD cards that have survived multiple phone generations and are still going strong now that I've moved them to my SBC hobby.  :smile:

closed 2016-08-10 by Fourdee