MichaIng/DietPi #455: DietPi-Software | OMV Open Media Vault

Fourdee, 2016-07-31:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=561&p=2457#p2457

LarsNorgaard, 2016-08-08:
DietPi works on the BananaPi m1, but the diet script won't show the cpu temperature.
Also the min freq could be set lower, to save some mW.

```
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temp         |     Unknown
 Governor     |     ondemand
 Throttle up  |     60% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      912 Mhz      720 Mhz    912 Mhz
 CPU1         |      912 Mhz      720 Mhz    912 Mhz

 - Press any key to exit....
```

Fourdee, 2016-08-09:
@LarsNorgaard 
DietPi has not released a M1 image. Unfortunately, we do not have access to that device, so we can't make an image or support it fully.

I believe the CPU temp issue occurs on the BPi Pro (A20) also, that I can look into :)

LarsNorgaard, 2016-08-09:
@Fourdee The Banana Pi and Banana Pi Pro are identical except the wifi/bluetooth chip. Everything else is the same :)

Fourdee, 2016-08-09:
@LarsNorgaard 
There are some differences, at least in the .fex: https://www.diffchecker.com/yd8b2luc

But looks like mostly GPIO stuff.

Ok, lets create a ticket and i'll look into the CPU temp issue: https://github.com/Fourdee/DietPi/issues/465

> Also the min freq could be set lower, to save some mW.

We use ARMbian for the BPi images. AFAIK, the min clock speeds, voltage regulation are tweaked by ARMbian and proven stable. 
If you want to try lower values, you'll need to edit the .fex and test it. If its proven stable, let us know and i'll add it into DietPi.
https://raw.githubusercontent.com/igorpecovnik/lib/master/config/fex/bananapim1plus.fex

LarsNorgaard, 2016-08-09:
@Fourdee 
The Diff you linked to has the wrong Banana Pi model :)
https://www.diffchecker.com/diovxwvf

It's the Banana Pi and the Banana Pi Pro I'm referring to. But yeah, those two models aren't identical either, but only seems to be changes to the GPIO pins and what LCD display is supported.

Is there any way to build DietPi myself from sources but with the fex for the Banana Pi?

Fourdee, 2016-08-09:
@LarsNorgaard 
```
#-------------------------------------------------------------------------
//DietPi on a fresh install
#-------------------------------------------------------------------------

#Jessie , unified apt sources.
rm /etc/apt/sources.list.d/deb-multimedia.list

#  - C2
# cat << _EOF_ > /etc/apt/sources.list
# deb http://ftp.debian.org/debian jessie main contrib non-free
# deb http://ftp.debian.org/debian jessie-updates main contrib non-free
# deb http://security.debian.org jessie/updates main contrib non-free
# deb http://ftp.debian.org/debian jessie-backports main contrib non-free
# deb http://ftp.debian.org/debian jessie-proposed-updates contrib non-free main
# _EOF_

# 	C2	stretch
# cat << _EOF_ > /etc/apt/sources.list
# deb http://ftp.debian.org/debian stretch main contrib non-free
# deb http://ftp.debian.org/debian stretch-updates main contrib non-free
# deb http://security.debian.org stretch/updates main contrib non-free
# deb http://ftp.debian.org/debian stretch-backports main contrib non-free
# deb http://ftp.debian.org/debian stretch-proposed-updates contrib non-free main
# _EOF_
# apt-get update
# apt-get install busybox-static
# apt-get upgrade -y
# apt-get dist-upgrade -y
# apt-get autoremove --purge -y

# rpi stretch
# cat << _EOF_ > /etc/apt/sources.list
# deb http://archive.raspbian.org/raspbian stretch main contrib non-free rpi
# _EOF_
# apt-get update
# apt-get upgrade -y
# apt-get dist-upgrade -y
# apt-get autoremove --purge -y

# deb-src http://ftp.debian.org/debian jessie main contrib non-free
# deb-src http://ftp.debian.org/debian jessie-updates main contrib non-free
# deb-src http://security.debian.org jessie/updates main contrib non-free
# deb-src http://ftp.debian.org/debian jessie-backports main contrib non-free
# deb-src http://ftp.debian.org/debian jessie-proposed-updates contrib non-free main

# - Everything else (not including RPi)
cat << _EOF_ > /etc/apt/sources.list
deb http://ftp.debian.org/debian jessie main contrib non-free
deb http://ftp.debian.org/debian jessie-updates main contrib non-free
deb http://security.debian.org jessie/updates main contrib non-free
deb http://ftp.debian.org/debian jessie-backports main contrib non-free
_EOF_

# RPI UK for my testing, mirror director is slow, unstable and unreliable -------------------------
cat << _EOF_ > /etc/apt/sources.list
deb http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian jessie main contrib non-free rpi
_EOF_
# RPI UK for my testing, mirror director is slow, unstable and unreliable -------------------------

#Remove following Jessie
apt-get clean
apt-get update
apt-get purge bluez bluetooth rsync dialog dhcpcd5 libsqlite* libxapian22 lua5.1 netcat-* make makedev ncdu plymouth openresolv shared-mime-in* tcpd strace tasksel* wireless-* xdg-user-dirs triggerhappy python* v4l-utils traceroute xz-utils ucf xauth zlib1g-dev xml-core aptitude* avahi-daemon rsyslog logrotate man-db manpages vim vim-common vim-runtime vim-tiny mc mc-data

#+armbian
apt-get purge toilet toilet-fonts w-scan vlan weather-util* sysbench stress apt-transport-* cmake cmake-data device-tree-co* fping hddtemp haveged hostapd i2c-tools iperf ir-keytable libasound2* libmtp* libusb-dev lirc lsof ncurses-term pkg-config unicode-data rfkill pv mtp-tools m4 screen alsa-utils armbian-* autotools-dev bind9-host btrfs-tools bridge-utils cpufrequtils dvb-apps dtv-scan-table* evtest f3 figlet gcc gcc-4.8-* git git-man iozone3 ifenslave
apt-get purge -y linux-jessie-root-*

[... 218 lines cut ...]
_EOF_
systemctl enable rc-local.service
systemctl daemon-reload

#Shutdown SSH/Dropbear before reboot
cat << _EOF_ > /etc/systemd/system/kill-ssh-user-sessions-before-network.service
[Unit]
Description=Shutdown all ssh sessions before network
DefaultDependencies=no
Before=network.target shutdown.target

[Service]
Type=oneshot
ExecStart=/usr/bin/killall sshd && /usr/bin/killall dropbear

[Install]
WantedBy=poweroff.target halt.target reboot.target
_EOF_
systemctl enable kill-ssh-user-sessions-before-network
systemctl daemon-reload

dpkg-reconfigure tzdata
dpkg-reconfigure keyboard-configuration #Keyboard must be plugged in for this to work!
dpkg-reconfigure locales
```

k-plan, 2016-08-10:
@Fourdee 

Sorry, but what is the topic here?

> DietPi-Software | OMV Open Media Vault

or

> DietPi-Hardware |  Image for Banana Pi M1 

:question:  :confused:

Fourdee, 2016-08-10:
> Sorry, but what is the topic here?

I got carried away lol, feel free to slap me :+1: 
Should be DietPi-Software | OMV Open Media Vault

k-plan, 2016-08-13:
@Fourdee 

> I got carried away lol, feel free to slap me

No, I don't want to slap you, really.

Your former comment are absolutely interesting and cool. 👍 
Search for something like this and want to read it very much.
But only the title is wrong, so nobody will find this things back again.

So please modify title and make a new issue for OMV Open Media Vault.
_(You are the owner, I can't do such things)_

closed 2016-10-29 by Fourdee