MichaIng/DietPi #458: NanoPi Neo | Add support and image for DietPi

Fourdee, 2016-08-04:


rhkean, 2016-08-04:
> CPU: Allwinner H3

Run away!  Run away!

:laughing:

Fourdee, 2016-08-04:
@rhkean 

Lol, I cant, its too cute: https://twitter.com/DietPi_/status/761172953161752578
[image]

rhkean, 2016-08-04:
It'll be interesting to see just how "Pi-compatible" it actually is.

Fourdee, 2016-08-04:
@rhkean 

> It'll be interesting to see just how "Pi-compatible" it actually is.

Where did you see this, got a link?

Fourdee, 2016-08-04:
Image available : http://dietpi.com/downloads/images/DietPi_NanoPiNEO-armv7-(Jessie).7z

rhkean, 2016-08-04:
If you use "Pi" in the name, I expect a certain level of compatibility.   That being said, I reread the feature list... it just states pi-gpio compatibility.

I like the nanopi-M3 specs....  It looks like it has potential.   (then again, so does the C.H.I.P., but I won't have one of those for a few more months)

I wish more SBC's were adding USB3, though.

Fourdee, 2016-08-04:
Completed. image live in http://dietpi.com/download

closed 2016-08-04 by Fourdee

k-plan, 2016-08-04:
>> It'll be interesting to see just how "Pi-compatible" it actually is.

>Where did you see this, got a link?

http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO#Layout

> If you use "Pi" in the name, I expect a certain level of compatibility. 

Hmm, please specified "Pi-compatible"?

- Crappy network implementation via USB ?
- Over-clocking CPU by vendor to  produce more heat then agreeable?

Sorry :smile:  just tease the most selling SOC ever.

> I like the nanopi-M3 specs.... 

Really nice  Samsung S5P6818 Octa-Core Cortex-A53 CPU with 400M Hz - 1.4G Hz, but I don't like the idea for active cooling:
http://www.friendlyarm.com/index.php?route=product/product&path=70_82&product_id=133

It's is something like ODROID-XU4.

Fourdee, 2016-08-04:
@rhkean @k-plan 
All Raspberry Pi (but really its down to Broadcom) need to do is upgrade the ARM to USB3 hub. Then we can have gigabit network through the USB3 bus. Instead of through this crappy USB2 bus that all RPI's suffer with since the dawn of time.

> Really nice Samsung S5P6818 Octa-Core Cortex-A53 CPU with 400M Hz - 1.4G Hz, but I don't like the idea for active cooling:
It's is something like ODROID-XU4.

The XU4 is one hot puppy, thats for sure lol

k-plan, 2016-08-04:
@Fourdee 

> Completed. image live in http://dietpi.com/download

Congratulations. Very quick implementation of DietPi running Debian Jessie.
Have you used the [armbian nanopineo.fex file](http://forum.armbian.com/index.php/topic/1580-nanopi-neo/#entry12364)?

Until now, where was only the official nanopi-neo-core-qte-sd4g-20160704.img  running UbuntuCore with Qt-Embedded  and a armbian NanoPi Neo test image.

Fourdee, 2016-08-04:
@k-plan 
I built the image with ARMbian build tools, selecting nanopim1. Copied the nanopineo.bin to /boot/script.bin. Worked a treat.

rhkean, 2016-08-04:
@Fourdee --- Show off!!!

1 day!!!!   It took me 2 weeks to add Pine64 support