MichaIng/DietPi #459: Give credit to ARMbian for images built with it

Fourdee, 2016-08-05:
BPI Pro, BPI M2+, NanoPi Neo

eg:
dietpi-banner 
DietPi + armbian

k-plan, 2016-08-06:
@Fourdee 

I like the idea and it's correct to give them credits for working on the Debian device specific base system. 👍 

Hmm, but what is with all Odroid ?
Credits for meveric and hardkernel?

All Pine64? Credits for rhkean and longsleep?

All Orange Pi? Credits for t.kaiser and armbian?

VirtualBox ? Credits for xenfomation?

I think, it's don't narrow your effort to get DietPi on work for so many different devices.

_BTW: your wonderful british 🇬🇧   DietPi based on so many work of german  🇩🇪   coders (meveric, longsleep, t.kaiser).  Nice!  😄   (sorry, couldn't resist)_

Fourdee, 2016-08-06:
> Hmm, but what is with all Odroid ?
Credits for meveric and hardkernel?
All Pine64? Credits for rhkean and longsleep?
All Orange Pi? Credits for t.kaiser and armbian?
VirtualBox ? Credits for xenfomation?

Great idea :+1: 

I wanted to try and get this in very top of the banner, but it would mean increasing the width of the banner top section (It fits perfectly now in 320x240 eg: waveshare). 

so i'll pop it here:
[image]

k-plan, 2016-08-06:
@Fourdee 

> so i'll pop it here:

I think it's okay.

> ... , but it would mean increasing the width of the banner top section (It fits perfectly now in 320x240 eg: waveshare).

Please no .... 😨

Fourdee, 2016-08-06:
Ok, think i got everyone minus Xenformation. This is due to HW_MODEL being the same for all VM images, so no way to identify if its VMware/VirtualBox/Xenserver.

closed 2016-08-06 by Fourdee