MichaIng/DietPi #463: How to connect to wifi without internet?

alex9099, 2016-08-07:
Hi i recently installed diet pi and i have to say it is very good... but there is a problem, when using dietpi-config to connect to wifi it tries to download netplug but as it is obvious there is no wired connection (thats why i'm trying the wifi one)
What can i do? I can connect to a wired network but not that soon :(

k-plan, 2016-08-08:
Hi  alex,

can you please provide output from terminal:

`:~# ip a`

`:~# route -n`

`:~# cat /etc/network/interfaces`

How have you tested internet connectivity? Via` dietpi-config`?

On which device is DietPi running?

Fourdee, 2016-08-08:
@alex9099 

netplug is only installed to allow both eth and wlan devices to work simultaneously.

> Edit: also it says that it has a network connection (wifi: availible/enabled/connected) but it has no connectivity to the internet

Disable the eth adapter. Simply go into Ethernet menu and select Disable.
Also, under the WiFi menu, make sure you set the country code.

k-plan, 2016-08-08:
> netplug is only installed to allow both eth and wlan devices to work simultaneously.

netplug is more for automated user comfort., see: https://github.com/Fourdee/DietPi/issues/261#issuecomment-217988286

> Disable the eth adapter.

:smile:

Fourdee, 2016-08-14:
Marking as resolved. Please reopen if required.

closed 2016-08-14 by Fourdee

tido-, 2017-03-04:
my Zero with 4.4.44 Kernel and a WiFi dongle RT5370 on the USB-Hub is now online - after 2-3hours.
In order to do so, I was not allowed to change manually the `/etc/network/interfaces` everything had to be done via dietpi-config. Now in `interfaces` my WiFi password is written in plain Text. 
Only this way I was able to bring up the connection, all the information I stored about the access-point in `/etc/wpa_supplicant/wpa_supplicant.conf` is completely ignored (see below).

What is the reason you are not making use of wpa_supplicant ?
And why is it completely ignored by dietpi-config network setup ?

network={
  ssid="your SSID" 
  psk="your PW" 
  proto=RSN 
  key_mgmt=WPA-PSK 
  pairwise=CCMP 
  auth_alg=OPEN 
  wpa=2 
}
https://w1.fi/cgit/hostap/plain/wpa_supplicant/wpa_supplicant.conf
https://en.wikipedia.org/wiki/Wpa_supplicant

Fourdee, 2017-03-04:
@tido- 

> everything had to be done via dietpi-config

If you are using DietPi-Config to setup WiFi, currently, it will use solely ```/etc/network/interfaces```.

If you require a different implementation of WiFi creds, you can complete this manually without using ```dietpi-config```, in a method of your choosing, that suits your security requirements.