MichaIng/DietPi #471: DietPi-Config | Overclocking for RPi Zero v1.3

k-plan, 2016-08-13:
@Fourdee 

```
root@RPi-Zero:~# /DietPi/dietpi/misc/rpi_boardinfo

  - RPi Board Info -
 Revision         |     900093
 Released         |     Q2 2016
 Model            |     Zero
 Memory           |     512 MB
 Manufacturer     |     Sony

```

with default  `/boot/config.txt` from _RASPBIAN JESSIE LITE_

```
root@RPi-Zero:~# vcgencmd get_config int
arm_freq=1000
audio_pwm_mode=1
config_hdmi_boost=5
core_freq=400
disable_auto_turbo=1
disable_commandline_tags=2
enable_uart=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=300
hdmi_force_cec_address=65535
ignore_lcd=1
init_uart_clock=0x2dc6c0
over_voltage_avs=0x249f0
overscan_bottom=32
overscan_left=32
overscan_right=32
overscan_top=32
pause_burst_frames=1
program_serial_random=1
sdram_freq=450
temp_limit=85

```

[image]

But in `dietpi-config` it will show:
[image]

`dietpi-config` works, so you can/will overclock or underclock RPi Zero, but I think, do this, it is not correct. Read on raspberrypi.org form the developer, no over-clock options are presented in raspi-config as the stock 1000MHz settings are at the upper end of what the majority of the hardware is capable of.

[image]

Tested, with `raspi-config`, where is no overclocking option.

I think, undervolt and underclock will be okay,for people want to save power consume. It's no will damage your hardware.
But overclocking will not be a good idea, just as RPi3.

So better will do something like that:

[image]

Fourdee, 2016-08-14:
@k-plan 

Great find :+1: 

Also found a few other issues while i was fixing that lol:
- RPi 2/3/Zero will now show correct sdram speed (450mhz default) in overclocking menu

k-plan, 2016-08-14:
@Fourdee 

> Also found a few other issues while i was fixing that lol:

Yes! 😄 

Testing on RPi 3 v.128:

[image]

and now with v.129:

[image]

[image]

Looks good on RPi 3 and RPi Zero V1.3, very nice. Thanks.

But no RPi 2 or RPi Zero V1 for testing here.

Fourdee, 2016-08-14:
@k-plan 
> Looks good on RPi 3 and RPi Zero V1.3, very nice. Thanks.
But no RPi 2 or RPi Zero V1 for testing here.

Excellent :+1: 

RPi 2:
[image]

RPi 1 256mb:
[image]

closed 2016-08-14 by Fourdee