MichaIng/DietPi #473: RPi 3 | Build-in WiFi doesn't connect

k-plan, 2016-08-14:
@Fourdee 

- RPi 3 with DietPi v.128 (fresh install from DietPi_v127_RPi-armv6-(Jessie))

Problem:
- does not connect with build-in WiFi to accesspoint via `dietpi-config`
- reboot with correct configured WiFi

dmesg output:

```
[    4.248239] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[    4.276952] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    4.301378] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    4.301425] cfg80211: World regulatory domain updated:
[    4.301435] cfg80211:  DFS Master region: unset
[    4.301444] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    4.301459] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    4.301472] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    4.301487] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[    4.301522] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    4.301538] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    4.301551] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[    4.301563] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[    4.301575] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[    4.413775] cfg80211: Regulatory domain changed to country: DE
[    4.413834] cfg80211:  DFS Master region: ETSI
[    4.413847] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    4.413862] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    4.413877] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    4.413891] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    4.413904] cfg80211:   (5470000 KHz - 5725000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
[    4.413917] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[    5.049759] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[    5.049786] brcmfmac: brcmf_add_if: ignore IF event
[    5.054664] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    5.054808] brcmfmac: power management disabled
[    5.173580] brcmfmac: power management disabled
```

disable WiFi via `dietpi-config`

```
[  113.242141] brcmfmac: power management disabled
[  113.364322] WARNING: driver brcmfmac_sdio did not remove its interrupt handler!
[  113.544860] usbcore: deregistering interface driver brcmfmac

```

enable  WiFi via `dietpi-config`

```
[  188.292889] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[  188.436760] brcmfmac: power management disabled
[  188.448169] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[  188.448181] brcmfmac: brcmf_add_if: ignore IF event
[  188.451958] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  188.457264] brcmfmac: power management disabled
[  188.604403] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  188.604417] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[  192.154074] brcmfmac: power management disabled
[  194.565558] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  194.565582] brcmfmac: power management disabled
[  194.593420] brcmfmac: power management disabled
[  194.774183] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  194.774197] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
```

Fourdee, 2016-08-14:
@k-plan 
Thanks, i'll look into it :+1:

Fourdee, 2016-08-19:
@k-plan 

I can't replicate this. WiFi works everytime for me.

Although, did get some warnings in dmesg aswell:
```
root@DietPi:~# dmesg | grep brcmfmac

#Reboot with WiFi configured
[    4.472799] usbcore: registered new interface driver brcmfmac
[    4.628144] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2                                                                 016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[    4.649522] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    4.735922] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    5.453401] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[    5.453438] brcmfmac: brcmf_add_if: ignore IF event
[    5.459676] brcmfmac: power management disabled
[    5.563264] brcmfmac: power management disabled
[    6.030269] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[    6.030294] brcmfmac: brcmf_cfg80211_scan: scan error (-11)

#Works fine

#Disable Wifi
[  492.241784] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[  492.251863] brcmfmac: power management disabled
[  492.384403] WARNING: driver brcmfmac_sdio did not remove its interrupt handle                                                                 r!
[  492.555247] usbcore: deregistering interface driver brcmfmac

#Enable Wifi
[  618.924094] usbcore: registered new interface driver brcmfmac
[  618.999949] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2                                                                 016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[  619.137664] brcmfmac: power management disabled
[  619.149333] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[  619.149350] brcmfmac: brcmf_add_if: ignore IF event
[  619.158454] brcmfmac: power management disabled
[  619.305030] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  619.305045] brcmfmac: brcmf_cfg80211_scan: scan error (-11)

#Works fine
```

But good news, i fixed stats for RPi onboard wifi while I was there lol:
[image]

k-plan, 2016-08-19:
- Starting from the schrach: RPi 3 with DietPi v.128 (fresh install from DietPi_v127_RPi-armv6-(Jessie)) 
- `dietpi.txt` `config.txt` `cmdline.txt` without any changes, no automated install
- auto-update to v.128 on first bootup => auto reboot
- first install config dialog, no usb device, selecting only "MC" and "DietPi-Ramlog #2" for install     => auto reboot
-  `dietpi-config` selecting "WiFi"  => enable WiFi => auto-install `netplug` and ...

[image]

[image]

```
root@DietPi:~# dmesg
. . . . .
[  505.997113] usbcore: registered new interface driver brcmfmac
[  506.142226] cfg80211: World regulatory domain updated:
[  506.142241] cfg80211:  DFS Master region: unset
[  506.142246] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  506.142254] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  506.142261] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  506.142267] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[  506.142274] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  506.142282] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  506.142288] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  506.142294] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  506.142301] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[  506.158933] cfg80211: Regulatory domain changed to country: GB
[  506.158950] cfg80211:  DFS Master region: ETSI
[  506.158955] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  506.158963] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  506.158971] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  506.158978] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  506.158985] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
[  506.158991] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[  507.053881] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[  508.064171] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[  509.081269] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

```
root@DietPi:~# lsusb
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

```
root@DietPi:~# lsmod
Module                  Size  Used by
brcmfmac              186343  0
brcmutil                5661  1 brcmfmac
cfg80211              427855  1 brcmfmac
rfkill                 16037  2 cfg80211
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
fuse                   83461  1
ipv6                  347530  20
```

- manual reboot

now:
[image]

- `Scan and Connect` => nothing (terminal screen pops up for only 1 sec.)
[image]

```
[... 33 lines cut ...]
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

```
root@DietPi:~# iw wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr b8:27:eb:xx:xx:xx
        type managed
        wiphy 0

```

```
root@DietPi:~# ip a | grep wlan
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    inet 192.168.0.100/24 brd 192.168.0.255 scope global wlan0
```

> WiFi works everytime for me.

Cool, but didn`t work here. Possible, it's fu.. up my on board wifi. Will check it with Raspbian lite.

Fourdee, 2016-08-19:
@k-plan 

You probably have already checked this, but just incase:
Are you using ```dtoverlay=sdhost,overclock_50=xx``` in ```config.txt```?

k-plan, 2016-08-19:
@Fourdee 

> Are you using dtoverlay=sdhost,overclock_50=xx in config.txt?

No! As written before, all stock and no changes.

Fourdee, 2016-08-19:
@k-plan 
I only asked, because I've done it a few times. Wondering why WiFi didn't work lmao ;)

k-plan, 2016-08-19:
> Will check it with Raspbian lite.

Didn't work as well.

> it's fu.. up my on board wifi.

It's seem so, on board wifi is fu ... up.

Why? Hmm ... can't say exactly. Have tested the "new" RPi 3 Boot Mode MDS: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

```
root@DietPi:~# vcgencmd otp_dump
08:00000000
09:00000000
10:00000000
11:00000000
12:00000000
13:00000000
14:00000000
15:00000000
16:00280000
17:3020000a
18:3020000a
19:ffffffff
20:ffffffff
21:ffffffff
22:ffffffff
23:ffffffff
24:ffffffff
25:ffffffff
26:ffffffff
27:00002727
28:f7c1c426
29:083e3bd9
30:00a02082
31:00000000
32:00000000
33:00000000
34:00000000
35:00000000
36:00000000
37:00000000
38:00000000
39:00000000
40:00000000
41:00000000
42:00000000
43:00000000
44:00000000
45:00000000
46:00000000
47:00000000
48:00000000
49:00000000
50:00000000
51:00000000
52:00000000
53:00000000
54:00000000
55:00000000
56:00000000
57:00000000
58:00000000
59:00000000
60:00000000
61:00000000
62:00000000
63:00000000
64:00000000
65:00000000
66:00009daa
```
But anyway. Have now blacklisted `blacklist brcmutil` and `blacklist brcmfmac`.
When tested with a USB WiFi stick and this one works immediately. 

I am so sorry Fourdee, was my fault. Now I have a RPi 3 with non on board Wifi.

closed 2016-08-19 by k-plan

Fourdee, 2016-08-19:
@k-plan 

> Didn't work as well.
I am so sorry Fourdee, was my fault. Now I have a RPi 3 with non on board Wifi.

No worries :+1:
A Raspberry Pi with a hardware failure? Not something you hear often, might be worth poking RPi for a free replacement ;)?

k-plan, 2016-09-07:
> A Raspberry Pi with a hardware failure? Not something you hear often, might be worth poking RPi for a free replacement ;)?

.

> Hallo ,
wir senden Ihnen einen neuen Raspberry Pi, als Ersatz für den Defekten zu. 
Ihr Rasppishop.de Team,
Mit freundlichen Grüßen,
Team Administration

:smiley:  :+1:

bhaveshgohel, 2016-09-08:
@k-plan @Fourdee : I just tested yesterday with the latest build fresh install and it work out-of-the-box on RPi3! :smiley: