MichaIng/DietPi #474: DietPi-Software | Add optimized installation for Uae4arm

Fourdee, 2016-08-15:
> https://github.com/Chips-fr/uae4arm-rpi
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=23#p2562

midwan, 2016-08-16:
UAE4All2 does not have JIT, 68040 or Picasso96 emulation, among other things. UAE4Arm is a newer build based on UAE4All but adds several new features (but also maybe some bugs):
https://github.com/lubomyr/uae4arm/blob/master/Readme_Pandora.txt

The main differences to UAE4ALL:
 - Different 68000 core (newcpu instead of FAME/C) with support for 32 bit addressing mode
 - Support for 68030 and 68040 cpus
 - FPU (68881, 68882 and internal 68040)
 - JIT
 - Autodetect Amiga ROMs
 - Up to 5 HDDs
 - Lot of new audio options
 - Picasso96
 - GUI
 - bsdsockets
 - RDB hardfiles
 - Support for rp9
 - CD32 support
 
What's missing compared to WinUAE:
 - Cycle exact emulation of cpu and blitter
 - SCSI support
 - AHI
 - Builtin debugger
and some more...

Fourdee, 2016-08-16:
@midwan 

Great to have you onboard for this one, cant wait :+1: 

> I am also interested in getting it to work with my Pine64.

Does uae4arm-rpi run on X11 or FB? 
We are still yet to implement the X11 mali drm drivers for Pine 64: https://github.com/Fourdee/DietPi/issues/380

> I can easily compile the latest sources and produce binaries for Pi 1, 2 and 3 to upload in the specified location.

Excellent, I'll host these on http://dietpi.com so we can use them in the installation. I've sent you an invite to a dropbox share we can use as a temp upload location.

> I can also take care of the instructions after installation (where each things goes, how to configure the emulator, etc.)

Awesome, once you have this ready, let me know and I'll give you access to edit the online forum docs, or I can copy/paste it in.

> How will we handle the amiga emulator boot option? In AmiBerry I used a systemd service configured to run very early, to have it boot as quick as possible into UAE - with some system tweaks I managed to do that in under 2.7 seconds from cold boot. ;-)

Impressive boot time, really is. 
We may have issues reaching that with DietPi with the current autoboot system. Mainly due to the dietpi-service which sets up Ramlog and Ramdisk during boot. The current "autoboot" section is triggered by /etc/rc.local (idle systemd service that runs very last): https://github.com/Fourdee/DietPi/blob/master/dietpi/login#L49-L99

I think we should be able to use your systemd service as a special case for this software installation. You will need to add ```After=dietpi-service.service``` under ```[Unit]``` for your service.

Once i've got the compiled binaries, i'll crack on with adding the installation code into DietPi-Software.

midwan, 2016-08-16:
uae4arm-rpi is customized (from the main uae4arm which uses SDL) to use dispmanx on the Pi, for direct access to the display hardware. It gives somewhat better performance than using SDL. The code has pieces that control this based on a declared variable, so the SDL parts are also still in place from what I've seen.

I'm not sure what we can use for the Pine yet, as I haven't seen much of the low-level stuff available there. Probably going for the SDL approach would be the most realistic for now. We can work on improving that later I guess. :)

I'll upload a fresh compile for each target later today and test out the boot times we can get on a real Pi 3.

midwan, 2016-08-17:
I've just added the uae4arm-rpi binaries in the dropbox folder. I have compiled a version for RPI1, 2 and 3 so you'll find 3 executables named accordingly. I've also included the expected folder structure, so that this archive can simply be extracted somewhere and it's ready to go (you still need to add the ROMs and disks/HDFs/whatever of course).

How do we plan to keep them up-to-date? I've forked the uae4arm-rpi project with the aim of improving it, so new versions will be coming soon. We should have some mechanism to upgrade the binaries when necessary.

Fourdee, 2016-08-17:
@midwan 

> I've just added the uae4arm-rpi binaries in the dropbox folder.

Excellent :+1: 

I'd like to make a few changes to ```adfdir.conf``` if thats ok?:
```
path=/etc/uae4arm-rpi
config_path=/etc/uae4arm-rpi/conf/
rom_path=/mnt/dietpi_userdata/uae4arm-rpi/kickstarts/
```
This will allow us to install to ```/etc/uae4arm-rpi```. ```/mnt/dietpi_userdata``` allows users to have their roms stored on USB drive/SDcard/custom location as set in ```dietpi-software```. DietPi automatically symlinks this directory if located off SDcard: http://dietpi.com/phpbb/viewtopic.php?f=8&t=478
Fileserver choices (proftpd and samba server) also point to ```/mnt/dietpi_userdata``` by default.

> How do we plan to keep them up-to-date? I've forked the uae4arm-rpi project with the aim of improving it, so new versions will be coming soon. We should have some mechanism to upgrade the binaries when necessary.

Ok, so we can do a few things:
- If the installation code does not change, we can simply overwrite ```uae4arm-rpi.zip``` with the new one on dietpi.com. Any user which then installs it, will be using the updated version
- If the installation code changes, we will have to give it a version number eg: ```uae4arm-rpi_v2.zip```. this way, we can modify the installation code to use the latest version, when that version of DietPi is then released, when users update dietpi, the updated installation code and file will be used.
 - DietPi also has a patch system, so we can target current uae4arm installations and apply an update. If we need to reinstall the program, we can use ```dietpi-software uninstall 108``` (to uninstall it), then ```dietpi-software install 108``` to (install it). ```dietpi-software list``` gives a list of software indexes.

Fourdee, 2016-08-17:
I'll get the dietpi-software code added later today for the installation. I'll then give you the instructions to test the installation on the testing branch :)

midwan, 2016-08-17:
OK with modifying the .conf file, it's a default template anyway. :)

Up to now, the archive has been the same (only the binaries inside it changed, directory structure was the same). I'm thinking if in the future we should give the users the option to choose the version installed, between say a cutting-edge but experimental version and an older but more stable one. In that case, we'll need to version the binaries at least (or the whole archive). I'm OK with both approaches.

If we wanted to be really special, we could also add the option to download and compile directly from source. I have that option in AmiBerry (takes care of installing required dev packages, clones project from github, runs proper make statement depending on Pi platform detected).

One thing to note: I found that the current version of uae4arm-rpi does NOT work properly if the user upgrades the Pi firmware to the latest available (currently 4.4y). It works OK with the previous one which is the one that currently comes with a default Jessie installation. We should have a warning about that or even an option to roll back a firmware upgrade to the previous one, until we (or Chips-fr) fixes this problem.
I've already opened a ticket about this on Chips-fr project, but so far I haven't seen much activity so it may take a while for it to be fixed (unless I manage to fix it myself).

Fourdee, 2016-08-17:
> I'm thinking if in the future we should give the users the option to choose the version installed, between say a cutting-edge but experimental version and an older but more stable one.

Sounds good, I think we should use a version id on the binary.zip. Allows us to control the updates, and generally good practice to separate the versions.

> Source code option

Yep, i'd be up for that aswell. 
We can create two installations, one for binary and one for source. Lets get the binary installation done 1st and go from there. Depending on when v129 is released, if need be, we can look at adding the source compile installation option at a later date (eg: for DietPi v130)

> One thing to note: I found that the current version of uae4arm-rpi does NOT work properly if the user upgrades the Pi firmware to the latest available (currently 4.4y). 

Ok, i'll take a look and see if we can automatically install a previous kernel version during installation.

midwan, 2016-08-17:
It works fine with the latest kernel and system updates (so you can do apt-get upgrade and apt-get dist-upgrade without fear), I was referring to the GPU firmware upgrade only: rpi-update (https://github.com/Hexxeh/rpi-update)

Fourdee, 2016-08-17:
@midwan 
I get a black screen after selecting start from the menu, is this the issue with 4.4 kernel you were mentioning earlier, or possibly something else?

I also tried with 4.1.21, same issue.

Fourdee, 2016-08-17:
Ok initial sourcecode is done. Some notes for testing:
- No change to kernel version, until we know which kernel version is stable.
- I've enabled the auto start boot service for testing purposes.

###  Heres how to install it on DietPi.
#### Method 1 - Automated installation from fresh image:
- Write DietPi image 
- Save this file 
[dietpi.txt](https://github.com/Fourdee/DietPi/files/423119/dietpi.txt)
to /boot/dietpi.txt
- Plug in ethernet
- Power on, watch the magic

#### Method 2 - Standard installation:
- Write DietPi image
- Before powering on, follow these instructions: https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md#steps-to-use-the-dietpi-testing-branch
- When DietPi-Software appears, select DietPi optimized menu, select Uae4Arm (located under Emulation and Gaming)
- Go start install

Fourdee, 2016-08-17:
@midwan 
I completely forgot GPU memory split, 128MB gpu required for uae4arm?

midwan, 2016-08-17:
Yeap, it won't work with the default 16MB (I tested it also, you get a black screen and the system hangs). Using 128MB is enough to make it work in my tests, it boots into my Workbench 3.1 HDF normally. No change in the current kernel is required.

I'll get the testing branch version and test things out there as well. From a first investigation, it seems that the dietpi.service takes roughly 4 seconds in bootup, so if we start it after that it will be much slower than the AmiBerry approach. 

To compare, here's a startup graph from AmiBerry after the tweaks I made:
[image]

Fourdee, 2016-08-17:
@midwan 
- 128MB GPU will be set during installation
- Lets try starting uae4arm before dietpi-service. It should be ok as it doesn't use ```/var/log```.

midwan, 2016-08-18:
What would the implications be if the user has set /mnt/dietpi_userdata to a network share? I mean of course they won't be available until the networking service is up, but do you see anything that would block uae4arm from starting if it's configured that way?

Unless there's something that would interfere with that, then it's ok if it runs in the background (in parallel with uae), the same as the rest of the services.

In other words, we can have a uae service running very early without blocking the rest of the services that will also run in parallel. Networking will be available when the relevant service is up, which is probably OK considering that the Workbench would have just booted by then also (and the user will not need network before Workbench starts up anyway). And that's just the worst-case scenario I can think of, if they are using just for launching a game, then networking is probably not even needed for that session.

Do you see a problem with this approach? Like I said, I'm not sure what the implications are with using /mnt/dietpi_userdata on a network share.

Fourdee, 2016-08-18:
> What would the implications be if the user has set /mnt/dietpi_userdata to a network share? I mean of course they won't be available until the networking service is up, but do you see anything that would block uae4arm from starting if it's configured that way?

I think the only issue would be for the user, example:
- System boots
- uae4arm menu is loaded
- User is loading roms from a network share (/mnt/dietpi_userdata, or any other network location), and the network filesystem service has not completed yet. The user may say "where the hell did my roms go" :)
It probably wouldn't happen, but there is a chance it can.

If we wanted to avoid this, I believe we can simply add ```After=remote-fs.target``` to the service. In theory this shouldn't increase the boot time if the user does not have a network mount, however, we'd need to test to verify.

Fourdee, 2016-08-18:
@midwan 
Any objections if we use?
- ```/mnt/dietpi_userdata/uae4arm-rpi/rom/``` = ROM directory
- ```/mnt/dietpi_userdata/uae4arm-rpi/kickstarts/``` - Kickstarts
Until this point i thought ROM/Kickstats were the same thing lol :)

Fourdee, 2016-08-18:
@midwan 
RPi 3 on DietPi install (used GIMP to convert the .svgz, not sure what happened there lol):
[image]

midwan, 2016-08-18:
I did some quick tests today:
- Updating to the latest firmware does not break uae4arm, at least not in DietPi. Perhaps in my earlier tests I had forgotten to check the memory split and that's why I got the black screen problem. In any case, it does not seem to be an issue.
- I tested a new compile of uae4arm in which I had added an extra resolution in the Picasso96 modes: 1920 x 1080 (1080p). It seemed silly that the native resolution of the HDMI output would not be there by default. It works fine in my tests, you just need to untick the "Keep 4:3 aspect ratio" in the configuration screen of uae (otherwise the display is stretched).
- There seems to be some bug(s) regarding keyboard and mouse input in uae4arm. Namely, sometimes the Input devices are set both to mouse (without you changing the configuration for that) and the result is that the mouse does not work at all. I had to restart the emulator multiple times to get it to work again when that happened. The second issue is that keyboard input is sluggish in general, when detecting the keypresses. This results in weird scenarios where, you press the Left Windows key (Left Amiga in the emulation) and it stays pressed even though you don't hold it down. Or if you keep a key pressed for a while, the stream of input continues for way more than when you release it again.

These problems are not apparent in the other ports of UAE4Arm such as the Android port, which is more often updated with bug fixes. I was planning to compare the sources and see if we can implement any of those fixes in the rpi port. Additionally, I wanted to go through the code in general and see how much I can fix (certainly several compiler warnings about deprecated operations), perhaps also getting pieces from the other various UAE ports where possible. It's a project that will take time though, which is why I mentioned that about the option to select which version of UAE to install. ;-)

I still have to test the "testing" branch of DietPi, including the uae service (either what you have done, if it's available, or setting it up myself). I hope to do this tonight if time permits, otherwise for sure during the weekend. I'll post my results and any comments here.

midwan, 2016-08-18:
I've just tested the "testing" branch, good work in automating everything! :)

I compared the startup times with systemd-analyze and converted the plots from both DietPi "testing" and my own AmiBerry:

DietPi:
[image]

AmiBerry:
[image]

In DietPi, uae4arm.service starts around the 3s mark. In AmiBerry it does around 2.7s. The difference is not that big, but I noticed that the kernel takes about twice the time to load.
Also, do you object to having the boot text optionally hidden? So that the first thing that comes up after powering up would be the emulator config/environment directly.

Meanwhile, I will experiment a bit with the boot options, to see if I can make it a little faster.

midwan, 2016-08-18:
Good news, I managed to speed it up further! 👍 

Adding `boot_delay=0` in the `config.txt` saves us 1 second, since by default it waits for 1 second until the SDcard is ready before it boots. I haven't seen any problems with my cards with setting this to 0, but they may be an issue potentially with low quality cards.

After modifying the boot_delay, I went on to further speed things up in the booting process. This time by modifying `/boot/cmdline.txt` as follows:

Adding `loglevel=3` (only display critical errors in startup)
[image]

Adding `logo.nologo` (disables the boot logo in the console)
[image]

And finally, `console=tty3` (changing the default console to tty3 so we don't get any boot messages shown):
[image]

In my environment, this brought down the whole process to 4 seconds less than where I started. And see how fast uae4arm.service is started there, that's even faster than my AmiBerry... ;-)

There might be more speed optimizations possible, but this is already quite promising.

midwan, 2016-08-18:
One more modification which seemed to speed things up a bit, but not very consistently during reboots:
Set the default boot mode to console:
```
systemctl set-default multi-user.target
ln -fs /lib/systemd/system/getty@.service /etc/systemd/system/getty.target.wants/getty@tty1.service
```

The best results I got after a few tries, were these:
[image]

you may want to test this one on your end as well though. ;)

Fourdee, 2016-08-19:
@midwan 
Impressive results :+1:  If i'am reading this right, roughly 2.42~ - 2.46~ seconds boot time?

Ok, so, I'am going to work on the ```dietpi-autostart``` options today. Based on your results, heres what I've planned:
#### Option 1: Uae4arm (fast boot)
< 3 seconds. Fastest possible boot time to uae4arm. May be unstable with certain SDcards.
- Your startup systemD service
- ```boot_delay=0``` in the ```config.txt```
- ```logo.nologo``` in ```cmdline.txt```
- ```loglevel=3``` in ```cmdline.txt```

#### Option 2: Uae4arm (standard boot)
compatible/stable 12+ seconds boot.
 - Normal DietPi boot, selected application launched at the very end of boot.

#### The items I'd like to leave out:
- ```console=tty3```
I really dislike this. Although i'am all up for minimizing the amount of "on screen gibberish" some of our novice Linux users don't really need or want to see,  I strongly believe we should never disable it entirely.

midwan, 2016-08-19:
Regarding the `console=tty3` option, it doesn't actually disable it entirely. It just moves them to tty3, which can be accessed with Ctrl-Alt-F3 if you want to. 
It doesn't have to be enabled by default, but what about having it as an option (for those novice Linux users who don't want to see those messages)? Perhaps only an option if you select the "fast boot" method?

Fourdee, 2016-08-19:
> Regarding the console=tty3 option, it doesn't actually disable it entirely. It just moves them to tty3, which can be accessed with Ctrl-Alt-F3 if you want to

Yep, but most novice users wouldn't know how to do that :) If the boot fails, they would be staring at their monitor with a blank screen saying "its broken" lol.

> Perhaps only an option if you select the "fast boot" method?

Ok, lets make "fast boot" to your specs. We can always add a whiptail message, highlighting the benefits of using this mode, and, possible drawbacks (eg: ```boot_delay=0 ``` might break SDcard boot)?

midwan, 2016-08-19:
I have never seen boot fail due to `boot_delay=0` (yet), but theoretically it would probably mean that the boot device was not found because it needed more time to initialize, and the whole process would stop there - unless it retries after a bit.
The worst case scenario would be that the user can't boot from that card anymore. They'd need to plug it to another working machine and edit the config.txt file from there, removing/commenting that line (or setting it back to the default value of 1).

So the possible drawbacks as far as I know are:
- Might fail to boot if the sdcard needs more time to initialize. In that case you need to edit the config.txt file and remove the `boot_delay=0` line.
- Boot messages will be shown in the tty3, so if something goes wrong and it's displayed there, you'll need to use Ctrl-Alt-F3 to see them.

We would also add this information in the documentation/wiki somewhere, so people are aware of it or can look it up if something goes wrong.

What do you think?

Fourdee, 2016-08-19:
> We would also add this information in the documentation/wiki somewhere, so people are aware of it or can look it up if something goes wrong.
What do you think?

Perfect:

[image]
[image]
[image]
[image]

midwan, 2016-08-19:
👍 Looks good!

Fourdee, 2016-08-19:
@midwan 
I just need to add in these options to the code: https://github.com/Fourdee/DietPi/issues/474#issuecomment-240992216

Heres the updated automated installation config for uae4arm with autoboot 6 set (fast boot):
[dietpi.txt](https://github.com/Fourdee/DietPi/files/426988/dietpi.txt)

midwan, 2016-08-19:
Thanks, I'll test it out and get back to you!

Fourdee, 2016-08-19:
> There seems to be some bug(s) regarding keyboard and mouse input in uae4arm.

Yep, confirmed. Seems very inconsistent in which combination of settings actually works, and when. Either that, or I've misunderstood the menu lol

Fourdee, 2016-08-21:
@midwan 
I've just released v129 and nudged this to v130 so we can continue work on it. 
I just want to make sure we get the installation perfect, and, gives us some extra time.

midwan, 2016-08-22:
Sure, no problem. :)
Could we add some sort of reference to AmiBerry in this?

Fourdee, 2016-08-23:
@midwan 

> Could we add some sort of reference to AmiBerry in this?

Yep no worries.
I'll also mention the collab with AmiBerry in the changelog, and the online docs. If you can think of anything else, and/or have a preferred wording/description, please let me know.

I believe all that is left on this ticket is:
:u5408: Online documentation.
:u5408: Possibility of compiling for other devices (eg: ARM64 C2/PineA64)
:u5408: Final testing and tweaking.

midwan, 2016-08-23:
Great!
I've been working on setting up an environment to allow me to further develop uae4arm these days, and yesterday I reached the first milestone: a working cross-compiler with the ability to remote debug the emulator. Now I can start fixing some of those bugs and try to merge in the latest WinUAE sources from the various pieces.

I have a Pine64 so I will try to port it over to that as well.

Where can I begin working on the documentation? Is there an online resource I can edit, or shall I prepare it locally and then we'll upload it?

Fourdee, 2016-08-23:
@midwan 

> I reached the first milestone: a working cross-compiler with the ability to remote debug the emulator. Now I can start fixing some of those bugs and try to merge in the latest WinUAE sources from the various pieces.

Excellent :+1:, sounds like a awesome setup. Would be great to see any improvements to Uae4ARM.

> I have a Pine64 so I will try to port it over to that as well.

Brilliant, I can test the Odroid C2 at my end (ARM64).

> Where can I begin working on the documentation? Is there an online resource I can edit, or shall I prepare it locally and then we'll upload it?

Usually, I just add a new post for the documentation on our PhpBB forums: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p5 (I'd like to have a better/cleaner system for the online docs, but it works for now)
I can also link to another url (eg: Google docs) if you prefer.
If not, send me a draft of the documentation and i'll transfer to the forums.

midwan, 2016-08-28:
I've just updated the uae4arm binaries on Dropbox. The latest version incorporates some bugfixes and some code that should help fix the LED problems with a Keyrah adapter if you have it.

They are also smaller in size, since I compressed the executable (it will decompress itself during runtime).
I haven't changed the RPI-1 binary, since I'm still trying to compile that one correctly. The changed ones are for the RPI-2 and 3, but all 3 of them are included to keep the archive consistent.

Fourdee, 2016-08-28:
@midwan 

> fix the LED problems with a Keyrah adapter if you have it.

Nope, but I ended up browsing http://amigakit.com/ for a good half hour lol.

> They are also smaller in size, since I compressed the executable (it will decompress itself during runtime).
I haven't changed the RPI-1 binary, since I'm still trying to compile that one correctly

Excellent :+1:  i've uploaded the new zip to dietpi.com. Next time you install from testing branch, it will use your updated zip.
I'll run a test install today on RPi 3.

Any updates on documentation, or still in the works?

midwan, 2016-08-28:
> Excellent 👍 i've uploaded the new zip to dietpi.com. Next time you

I'll work on the docs later today, should not take too long.

Fourdee, 2016-08-28:
@midwan 

Appears Uae4ARM doesn't support/display symlinks when browsing filesystem. 

So, datapaths for user uae4arm data (roms/kickstarts etc) is:
- SDcard only = ```/mnt/dietpi_userdata/uae4arm-rpi```
- USB drive installed via DietPi = ```/mnt/usb_1/uae4arm-rpi```

Fourdee, 2016-08-28:
@midwan 

### Questions:
- AROS KS ROM (built-in) does not work for me. Just get red lines on screen. Is this expected?
- Can we provide Kickstarter roms files (eg: 1.3) with our installation, is it legal?

From my testing. To successfully run uae4arm:
- Start uae4arm via ```systemctl start uae4arm-rpi``` (if not set for autostart during boot)
- Select "Load autostart.uae" (This never auto loads inputs for me)
- Choose a kickstarter rom. (I cant get AROS builtin to work)
- Choose a floppy disk .adf 
- Run

midwan, 2016-08-28:
the AROS KS ROM does work, but it has some dependency on the CPU you choose (I believe it doesn't work with 68000, try with 68020 or higher). It's not 100% compatible though, so some things will not quite work until it has further evolved.

The original Kickstart ROMs are still under copyright I'm afraid, the current owner is Cloanto which offers them for purchase as part of the Amiga Forever package. Independent license deals are also possible, but they will come at a cost. If you own Amiga Forever, then you can use the ROMs contained there with any emulator. It's the same situation with WinUAE on Windows or any other platform really. Until we get a fully compatible AROS rom (if ever) or this license situation changes, we're stuck with that.

For running uae4arm:
- you can provide a command line parameter to load a specified config file automatically. The parameter is "-f <conf file name>". So you can open the config window once, set things up the way you want them and save the configuration. After that you can auto-load it with that parameter.
- You can also configure uae4arm to Hide the configuration window on startup, i.e. start direct into emulation. This combined with the previous option makes it possible to auto-start a configured Workbench environment, once you set it up.

Of course, if you want to boot from an ADF for a game, then this doesn't apply - you have to enter the configuration to choose which floppy to load. But you could have a A500 configuration auto-loaded for example, if you wanted to.

So for my environment, since I mostly boot into a customized 3.9 Workbench (such as AmiKit), I have the following setup:
- autostart UAE, or run it manually once.
- Set up a configuration for the highest possible CPU speed, AGA chipset, select either HDF or directory based drives (and copy over my pre-installed environment), select ROM 3.1.
- Save the configuration with an appropriate name, say "autostart".
- After I've tested the setup and verified everything works, I then change my start to:

`uae4arm-rpi3 -f conf/autostart.uae`

Of course, you can have your autostarting service with the above line to begin with, and provide a preset "autostart" configuration with the most common settings. Then the end-users can always modify that configuration accordingly.

midwan, 2016-08-28:
Hi Dan,
I'm attaching the first documentation file here, hope that's ok.
I've marked some areas with yellow so we can work on those and update them accordingly. Let me know if we should include any additional info or if you have any comments on this. It's a first draft, we could improve it further with some input (and coffee). :)

The format is .odt, I assume that's OK? If not, let me know what format you'd prefer instead.
[AmiBerry.zip](https://github.com/Fourdee/DietPi/files/441258/AmiBerry.zip)

Fourdee, 2016-08-28:
@midwan 

> UAE4arm is compiled with predefined search paths for the "data" folders, expecting them to be in the same folder the binary is in. So the structure should look like this:
uae4arm-rpi
data/
conf/
kickstarts/
savestates/
screenshots/

Yep, as dietpi uses a single location for userdata ```/mnt/dietpi_userdata```, during installation we create symlinks to replace the following folders:
```
kickstarts/
savestates/
screenshots/
```
So in general terms, those folders exist in ```/etc/uae4arm-rpi/```
But as I mentioned earlier, uae4arm (at least the front end) cannot see or browse symlinks. So could be an issue when our users have a USB drive or custom location for their DietPi user data location.
Although, as per: https://github.com/Fourdee/DietPi/commit/b060a49c1ec5996fa80ead470768a261127a1b5e DietPi will change all conf locations during installation, to now match the actual location.

> The original Kickstart ROMs are still under copyright I'm afraid
The AROS KS ROM does work, but it has some dependency on the CPU you choose (I believe it doesn't work with 68000, try with 68020 or higher).

Shame, a true definition of "milking every last drop for money".

> uae4arm-rpi3 -f conf/autostart.uae

We have this set in the service already, but still fails to load inputs and bindings: https://github.com/Fourdee/DietPi/blob/testing/dietpi/conf/uae4arm-rpi.service#L8

Also, we create a symlink from ```uae4arm-rpi``` to the device specific binary, during DietPi installation. So either ```/etc/uae4arm-rpi/uae4arm-rpi``` or ```/etc/uae4arm-rpi/uae4arm-rpi[MyDeviceNumber]``` works.

> I'm attaching the first documentation file here, hope that's ok.

Excellent, I'll check it over :+1:

Fourdee, 2016-08-28:
@midwan 
Nice documentation :+1: 

Based on your documetnation, I tried to replicate what a new user would need to do, step by step.
I ended with up this: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=64#p64

Aside from 'First Run Setup', all items in RED are areas I'd like to discuss/improve. If you have any other areas you'd like to discuss, please let me know.

midwan, 2016-08-28:
Good work there! 👍 

Regarding the items in RED:
- Donations: Sure, no problem with that. :)
- Disclaimer: kind of obvious really, we could just mention that we don't supply any Kickstart ROMs or other copyrighted material (that should keep the lawyers away), that should cover it.
- Image: I was planning to make a proper logo, but haven't gotten to it yet. Can we update it later if necessary?
- Kickstarts path: Any path will work in the end, what the predefined setting helps with is whether UAE will automatically find them when it first starts up or not. You can always go in the configuration and change the paths, they will be saved as such from then on. Or, alternatively I can easily compile a version with another predefined path for the various components, so that it ties together with DietPi better. Let me know which way you prefer.
- ADFs: I'm a bit skeptical about the wording here. ROMs are normally Kickstart ROMs in the Amiga world, since they refer to physical chips (which could only be read from). ADFs are just floppy disk images, which can be read/written to as well. You can create empty ADFs from within the emulators, write stuff to them etc. So maybe calling them ROMs might be confusing for people?
Also, there are many ADFs which are offered for free from their creators nowadays, not everything is under copyright. So we should rephrase that last sentence a bit, IMHO.
Also, there's no predefined path that UAE expects to find ADFs in, it can be anywhere.
- Hard drive images: UAE can work with either hardfiles (HDFs) or a normal directory containing Amiga files. Those can also be placed anywhere you want, the emulator allows you to browse for the path to them in order to use them. HDFs can be created in WinUAE (and/or other emulators) and just copied over. They are virtual hard disks, containing filesystem, files, etc. They have a bit lower performance than normal directories, since read/write operations have to happen within that single file.
Normal directories are the other option. You can just create an empty directory, select it in uae and it will be used as an Amiga hard drive. You can use it to install Workbench, copy files etc. Of course you can have multiple such directories, each representing a separate disk (or partition) in your emulated system.
For example, I have the following setup:
```
<uae4arm>/dir/System
<uae4arm>/dir/Work
```

System contains my Workbench 3.9 installation, while Work contains my applications and extra software I have there. I have selected them both in my UAE configuration, assigning a Device Name (DH0: and DH1: respectively) and Volume Label (System and Work respectively). Here's a screenshot of what the config looks like from within uae:
[image]

Finally, it would be ideal to fix the problem about the bindings you mentioned. I'll work on that and see what I can do, the aim should be to eventually have a system that can boot directly even without going in the config screen at all.

midwan, 2016-08-29:
Here's an updated logo, since a friend was kind enough to design it for me:
[image]

Fourdee, 2016-08-29:
@midwan 

Epic image. I may ask your friend to create one for DietPi lol! It looks excellent :+1: 

> You can always go in the configuration and change the paths, they will be saved as such from then on. Or, alternatively I can easily compile a version with another predefined path for the various components, so that it ties together with DietPi better. Let me know which way you prefer.

We apply the paths directly to the ```/etc/uae4arm-rpi/conf/adfdir.conf``` and ```/etc/uae4arm-rpi/conf/autostart.uae``` during installation, but it doesn't seem to find or list any content, unless the user browses Roms and Kickstarts.
I also tried the "Rescan Rom" button, has no effect on listed Kickstarts or ADF's.

> Finally, it would be ideal to fix the problem about the bindings you mentioned. I'll work on that and see what I can do, the aim should be to eventually have a system that can boot directly even without going in the config screen at all.

This would be great :+1: 
It seems to load the CPU/sound/display settings fine, but the Input settings are never loaded, until the user selects "load" on the autostart.uae configuration.

I updated the zip file to contain the paths that DietPi will use ```/mnt/dietpi_userdata/uae4arm-rpi``` in the config files: ```adfdir.conf``` and ```autostart.uae```
http://dietpi.com/downloads/binaries/all/uae4arm-rpi_v1.zip 
You may want to copy these and use them when creating updated zips.

> Hard drive images: 

Excellent :+1: Makes sense now.
I think I may leave this out for now, too much documentation can cause confusion for the user.

> DFs: I'm a bit skeptical about the wording here. ROMs are normally Kickstart ROMs in the Amiga world, since they refer to physical chips (which could only be read from). 

Yep :+1: I'am so used to calling emulator games "Roms" (n64 etc)
I think we should then change the folder ```rom``` to ```floppy_images```?

midwan, 2016-08-29:
I'm not sure the adfdir.conf file is actually used, I found it there in the original archive so I kept it, but I will have to check the source code to be 100% sure. It may be irrelevant and we can throw it away if that's the case.

The *.uae files are certainly used and whatever is in them should be loaded/respected though. If there's a bug in loading the input settings, we'll have to get it fixed.

We could call the floppy images folder `floppy_images `or `adf`, depends on how long you like your folder names I guess. :) Either one is good I believe.

I'll prepare a custom compile with the DietPi paths later today, to see if that helps at all.

Fourdee, 2016-08-29:
@midwan 

As far as I can tell, ```adfdir.conf``` gets used for:
- Default paths (left hand side menu) 
```
config_path=/etc/uae4arm-rpi/conf/
rom_path=/mnt/dietpi_userdata/uae4arm-rpi/kickstarts/
```
- When user browses filesystem (eg: ... button)
```
path=/mnt/dietpi_userdata/uae4arm-rpi/
```

I've updated the paths in this file and now Kickstarts are being listed automatically on 1st run! 
The paths in this file require a slash after the last directory ```/```, else it doesnt work.

Fourdee, 2016-08-29:
@midwan 

I think we can improve performance in Uae4ARM by setting 480p resolution during installation.
Framebuffer size has a direct impact on RAM performance as it shares the same bandwidth: https://www.raspberrypi.org/forums/viewtopic.php?p=104973&sid=44bbe222489dcc082fcdf6faeccef711#p104973

So in theory, the smaller the resolution, the faster the memory throughput on RPi. Its not much, but every little counts. And the Uae4ARM menu looks great in 480p (full screen).

midwan, 2016-08-29:
Not a bad idea, as long as the user is still able to use higher resolutions if they want to.
For example, I use my Workbench at FullHD resolution - it's great for working on graphics apps, coding, multitasking, etc.

For games, you certainly don't need more than 480p of course.

On another matter, there seems to be a bug in Raspbian installations which makes the Scroll Lock key non-functionaly (the LED does not even turn on on the keyboard). I've managed to locate a fix for that, but I haven't checked if you already have addressed this in DietPi.
The solution is to copy the attached file in `/etc/udev/rules.d` and then run `udevadm control --reload-rules` and reboot.

Can we include this fix in DietPi? It should benefit everyone if it works.
[50-leds.rules.zip](https://github.com/Fourdee/DietPi/files/443205/50-leds.rules.zip)

Fourdee, 2016-08-30:
@midwan 

> Not a bad idea, as long as the user is still able to use higher resolutions if they want to.
For example, I use my Workbench at FullHD resolution - it's great for working on graphics apps, coding, multitasking, etc.

As far as I can tell, the emulator supports max 768x270 resolution in the GUI. I did try raising these to 1080p (1920x1080) in ```autostart.aue```, but locked up emulator on start.
480p is 858x480p, so still higher than max uae4arm rendering resolution.

The RPi always outputs at 1080p, but the framebuffer size determines internal (actual visible) resolution.

> Can we include this fix in DietPi? It should benefit everyone if it works.

Yep, I'll get this added for all RPi's :+1: 
```ACTION=="add", SUBSYSTEM=="leds", ENV{DEVPATH}=="*/input*::scrolllock", ATTR{trigger}="kbd-scrollock"```

midwan, 2016-08-30:
The config window is hardcoded to show at a specific resolution, but the actual Amiga screen that opens up is not, at least if you use Picasso96. If you use the Picasso96 RTG drivers from Workbench, you can open up a screen up to the maximum resolution your host system provides (1080p in the case of Pi).

I think it should be able to switch to a higher resolution even if your starting framebuffer size is smaller, but we need to test this to be sure. :)

The test should go like this:
- Boot in 480p
- Run uae4arm, boot a Workbench installation which has Picasso96 setup (I have such an environment handy)
- Try to change resolution from within the emulation, see if it can go higher than 480p without problems.

If the above scenario works, we're good to go. For games, you won't need to change resolution to a higher one anyway, only lower (e.g. 320x256) so we should be covered already.

If you have a configuration that starts with 480p, I can test the rest.

Fourdee, 2016-08-30:
> If you have a configuration that starts with 480p, I can test the rest.

@midwan

Configuration for the RPi framebuffer?

This will set RPi framebuffer to 480p
```
sed -i '/framebuffer_width=/c\framebuffer_width=854' /DietPi/config.txt
sed -i '/framebuffer_height=/c\framebuffer_height=480' /DietPi/config.txt
reboot
```

1080p if you want to return to this.
```
sed -i '/framebuffer_width=/c\framebuffer_width=1920' /DietPi/config.txt
sed -i '/framebuffer_height=/c\framebuffer_height=1080' /DietPi/config.txt
reboot
```

Fourdee, 2016-08-30:
Note to self:
Move and symlink ```conf/```` to userdata dir. Basically means uninstall = No loss of saved configs.

midwan, 2016-08-31:
Confirmed: user can switch resolution normally within the emulation, even to higher ones than 480p when the framebuffer is set to such.

So we can add that modification as well.

I've located a bug in uae (or perhaps it's Raspbian) with the CapsLock/NumLock keys. If you start the emulation from a console (i.e. no X is loaded) then those keys do not respond as expected (Caps Lock does not turn on the LED, and does not turn input into capitals). If you start the emulator from within X however, even from a console window there, it works fine.

I'm experimenting with a few changes but so far I haven't had much success.

Fourdee, 2016-08-31:
@midwan 

Xinit should work for this, try:
```
systemctl stop uae4arm-rpi
dietpi-software install 6
cd /etc/uae4arm-rpi
xinit ./uae4arm-rpi
```

midwan, 2016-08-31:
It does indeed! Thanks for that!

Now I can cleanup the code a bit, finally!

Fourdee, 2016-08-31:
@midwan 
Excellent. 
I'll create a script to detect if ```xinit``` needs to be run, then add to the ```uae4arm-rpi``` service.

[53 more replies]

closed 2016-09-06 by Fourdee