MichaIng/DietPi #475: DietPi | Debian Stretch

Fourdee, 2016-08-15:
Look into creating and supporting Debian Stretch in DietPi. This will allow for more "recent" software version installations.

>

k-plan, 2016-08-15:
Puhh, 

reading, Debian Stretch will start over after Kernel 4.10 is released.

https://wiki.debian.org/DebianStretch

If I will see the timeline, it will be end of 2017 until official Stretch release.
But good to know, you have a milestone.

Fourdee, 2016-10-09:
```dietpi-drive_manager``` fails to start with Stretch, when ```auto``` entries for non-existent drives are enabled in ```/etc/fstab```

Fourdee, 2016-10-11:
Notes:
- ```dropbear``` now supports SCP/SFTP by default (YAY!)
- PHP upgraded to php7

Fourdee, 2016-10-13:
🈯️ webserver stacks / nextcloud / owncloud

Zone22, 2016-11-03:
I think I'll hold off on my projects until this is implemented. Otherwise it requires a reformat.

Fourdee, 2016-11-03:
@Zone22 

This wont be ready for a while, need to test and fix every installation. I may release the image under alpha, but wont be for a few weeks.

k-plan, 2016-12-10:
@Fourdee 

> Alpha testing

http://dietpi.com/phpbb/viewtopic.php?f=9&t=986&p=4279#p4279

Fourdee, 2016-12-18:
@k-plan 
Many thanks for testing results 👍 , i've updated the list in first post and i'll get those fixed.

Fourdee, 2016-12-20:
Note to self: test RoonBridge installer and ```dietpi-justboom```

Fourdee, 2016-12-21:
Typical, trying to do some RPi testing and ``` http://archive.raspberrypi.org```  is unstable/offline lol:
```
The following packages will be upgraded:
  libraspberrypi-bin libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel
4 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 34.5 MB of archives.
After this operation, 33.8 kB of additional disk space will be used.
Err:1 http://archive.raspberrypi.org/debian jessie/main armhf raspberrypi-kernel armhf 1.20161215-1
  500  Internal Server Error [IP: 46.235.227.11 80]
Err:2 http://archive.raspberrypi.org/debian jessie/main armhf libraspberrypi-bin armhf 1.20161215-1
  500  Internal Server Error [IP: 46.235.227.11 80]
```

Fourdee, 2016-12-21:
🈴 HiFi:
```
E: Package 'libavcodec56' has no installation candidate
E: Package 'libavformat56' has no installation candidate
E: Unable to locate package libavresample2
E: Package 'libavutil54' has no installation candidate
E: Package 'libbinio1ldbl' has no installation candidate
E: Unable to locate package libjson-c2
E: Package 'libnfs4' has no installation candidate
E: Unable to locate package libvpx1
E: Package 'libwildmidi1' has no installation candidate
E: Unable to locate package libx264-142
```

k-plan, 2016-12-21:
@Fourdee 

New Alpha testing DietPi v140:

http://dietpi.com/phpbb/viewtopic.php?f=9&t=986&p=4455#p4455

You DietPi fixes are well done! 😃 

Put this stuff to the forum, to get some more user and concern for Stretch alpha testing.
But if you prefer it, I can put it as well on github. 😕

Fourdee, 2016-12-22:
@k-plan 

> But if you prefer it, I can put it as well on github.

Forums are fine, please keep giving me a nudge on here (notification) so I don't miss it 😃 

Great testing, thanks @k-plan. Stretch is a long road, lots of testing as you know, but with your help (and hopefully others), we'll get there much quicker 👍 

@k-plan  Merry Xmas! 🎄 😃

Fourdee, 2017-01-04:
Looks like the bluetooth issue exists on jessie aswell. lacking ```pi-bluetooth``` packages. Will get that patched.

Fourdee, 2017-01-05:
WiFi:
```
root@DietPi:~# dmesg | grep brcm
[ 7414.760610] usbcore: registered new interface driver brcmfmac
[ 7414.761782] brcmfmac_sdio mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2
[ 7415.765022] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[ 7416.777708] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

``

Fourdee, 2017-01-19:
Test NeoVim packages available in Stretch:
https://github.com/Fourdee/DietPi/issues/692#issuecomment-272726938

Fourdee, 2017-02-03:
RPi 3 onboard WiFi, not functional after reboot:
http://dietpi.com/phpbb/viewtopic.php?f=9&t=986&p=5434#p5407

Fourdee, 2017-02-05:
VNC4, changed to tigervnc:

```
ii  tigervnc-standalone-server     1.7.0+dfsg-2                armhf        Standalone virtual network computing server
ii  vnc4server                     4.1.1+X4.3.0+t-1            armhf        transitional package

root@DietPi:~# vncpasswd
-bash: vncpasswd: command not found
```

kronflux, 2017-03-02:
I'd like to note that(although I'm using a manually updated PineA64) v145 produced an error with Fail2Ban for me.
Apt-get has failed:
│  - Apt string = fail2ban --no-install-recommends
│  - Error code = 100
the console produced this:
Job for fail2ban.service failed because the control process exited with error code.
See "systemctl status fail2ban.service" and "journalctl -xe" for details.
invoke-rc.d: initscript fail2ban, action "start" failed.
Process: 7398 ExecStart=/usr/bin/fail2ban-client -x start (code=exited, status=255)

Fourdee, 2017-03-08:
ToDo:
- Update image (RPi Zero W support)

Fourdee, 2017-03-10:
@kronflux 

Looks like an issue with apt, or a bug with the ```fail2ban``` package. Fails to install on RPi 3, at a basic ```apt-get install``` level:
```
Setting up fail2ban (0.9.6-1) ...
Created symlink /etc/systemd/system/multi-user.target.wants/fail2ban.service → /lib/systemd/system/fail2ban.service.
Job for fail2ban.service failed because the control process exited with error code.
See "systemctl status fail2ban.service" and "journalctl -xe" for details.
invoke-rc.d: initscript fail2ban, action "start" failed.
● fail2ban.service - Fail2Ban Service
   Loaded: loaded (/lib/systemd/system/fail2ban.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2017-03-10 13:50:55 UTC; 29ms ago
     Docs: man:fail2ban(1)
  Process: 4390 ExecStart=/usr/bin/fail2ban-client -x start (code=exited, status=255)

Mar 10 13:50:55 DietPi systemd[1]: fail2ban.service: Unit entered failed state.
Mar 10 13:50:55 DietPi systemd[1]: fail2ban.service: Failed with result 'ex…de'.
Hint: Some lines were ellipsized, use -l to show in full.
dpkg: error processing package fail2ban (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for libc-bin (2.24-9) ...
Processing triggers for systemd (232-18) ...
Errors were encountered while processing:
 fail2ban
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

WolfganP, 2017-04-08:
@Fourdee As a suggestion, as I don't know if you're using a custom build for ffmpeg, but if yes; it may be worthy to include the openmax functionality to include hardware assisted encoding (as in https://github.com/legotheboss/YouTube-files/wiki/(RPi)-Compile-FFmpeg-with-the-OpenMAX-H.264-GPU-acceleration)

Fourdee, 2017-04-09:
@WolfganP 

> it may be worthy to include the openmax functionality to include hardware assisted encoding (as in 

Legend 👍  this should help Emby server performance which uses FFmpeg.

I'll need to compile them. Jessie ARMv6 uses one we compiled, ARMv7 is deb multimedia repo, stretch is rpi repo.

I'll check flags on FFmpeg for stretch before hand, may already be enabled (hoping). But we'll do it for Jessie aswell.

MichaIng, 2017-06-02:
I faced two problems with this image:

1. During dietpi-software installation process some warning is shown about "`--force-yes is deprecated`" and options with "`allow`" should be used instead.

2. I use dietpi to run nextcloud on my pi2. Chose LAAP webserver stack to assure that mariadb will be used. It is not yet possible to chose preferred database like webserver itself -> Separate issue already opened about that. Nevertheless depending or that or not, dietpi-services no tries to handle two database services "mysql" AND "mariadb". Actually mysql is not installed, at least not via apt and it does not show up in dietpi-software either. So I can't remove it. dietpi-services starts "mysql" first successfully and tries to start "mariadb" afterwards which fails after long trial time. Nextcloud works fine, using the available service. It IS actually mariadb, which can be seen via modules, version check and nextcloud web ui. I am wondering about this "mariadb" service. DietPi-Process_Tool also shows one entry about that beside : `Webserver MariaDB (1086)` Never saw mariadb running as another service name than mysql. Did something change there or is it just some typing bug introduced with this dietpi image?

MichaIng, 2017-06-02:
About missing `apc.php` with PHP7+, which results in some dummy file will be used instead:
> 7317`			# - PHP7+ - File does not exist`
> 7318`			else`
> 7319
> 7320`				wget http://dietpi.com/downloads/conf/apc.php -O /var/www/apc.php`

There is some `apc.php` on github which is working well with my dietpi rpi stretch PHP7 setup: https://github.com/krakjoe/apcu/blob/master/apc.php

Fourdee, 2017-06-03:
@MichaIng 

> There is some apc.php on github which is working well with my dietpi rpi stretch PHP7 setup: 

Thanks for this 👍 
I've tested with Jessie and seems compatible. I'll update the code with new link.

xanview, 2017-06-10:
Is there a beta image for the Asus Tinker Board by any chance? :P

Fourdee, 2017-06-11:
@xanview 

> Is there a beta image for the Asus Tinker Board by any chance? :

Yep: http://dietpi.com/downloads/images/DietPi_AsusTinkerBoard-armv7-(Jessie).7z

Its named Jessie, however its Stretch.

Fourdee, 2017-06-12:
- Update Stretch image: http://dietpi.com/phpbb/viewtopic.php?f=9&t=986&start=20#p7498

xanview, 2017-06-12:
That's awesome, thank you! - I'm using it as a headless server and so far latest image is working beautifully.

noplanman, 2017-06-25:
Regarding the fail2ban issue, what solved it for me was simple:
`$ touch /var/log/auth.log`
before the install, then it went through successfully 👍

Fourdee, 2017-06-25:
@noplanman 

> $ sudo touch /var/log/auth.log
before the install, then it went through successfully 👍

Legend 👍 Many thanks for the fix. I'll get this added to ```dietpi-software```. Works with Jessie aswell, so we'll add for all distros.

mr-bolle, 2017-07-08:
Hi Fourdee, many thanks for your work. 

I install webmin and the Module webminstats v2.16, if i open this Module ''Historic System Statistics" i get the Error Message "Error - Missing Content-Type Header"

I could solve this problem while I install **apt-get install libcgi-pm-perl**

many thanks again

mr-bolle, 2017-07-08:
I install DietPi_RPi-armv6-(Stretch) image, and install over dietpi-software Fail2Ban, after that the service don't start 

$ /etc/init.d/fail2ban status
Failed to start server : Starting fail2ban (via systemctl): fail2ban.serviceJob for fail2ban.service failed because the control process exited with error code. See "systemctl status fail2ban.service" and "journalctl -xe" for details. failed!

> $ journalctl -xe
-- 
-- Unit fail2ban.service has finished shutting down.
Jul 08 16:08:04 nc-raspi2 systemd[1]: Starting Fail2Ban Service...
-- Subject: Unit fail2ban.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit fail2ban.service has begun starting up.
Jul 08 16:08:05 nc-raspi2 fail2ban-client[1739]: ERROR  Failed during configuration: While reading from '/etc/fail2ban/jail.conf' [line 52]: option 'port' in section 'pam-generic'
Jul 08 16:08:05 nc-raspi2 systemd[1]: fail2ban.service: Control process exited, code=exited status=255
Jul 08 16:08:05 nc-raspi2 systemd[1]: Failed to start Fail2Ban Service.
-- Subject: Unit fail2ban.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit fail2ban.service has failed.
-- 
-- The result is failed.
Jul 08 16:08:05 nc-raspi2 systemd[1]: fail2ban.service: Unit entered failed state.
Jul 08 16:08:05 nc-raspi2 systemd[1]: fail2ban.service: Failed with result 'exit-code'.
Jul 08 16:08:05 nc-raspi2 systemd[1]: fail2ban.service: Service hold-off time over, scheduling restart.
Jul 08 16:08:05 nc-raspi2 systemd[1]: Stopped Fail2Ban Service.
-- Subject: Unit fail2ban.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit fail2ban.service has finished shutting down.
Jul 08 16:08:05 nc-raspi2 systemd[1]: fail2ban.service: Start request repeated too quickly.
Jul 08 16:08:05 nc-raspi2 systemd[1]: Failed to start Fail2Ban Service.
-- Subject: Unit fail2ban.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit fail2ban.service has failed.
-- 
-- The result is failed.
Jul 08 16:08:05 nc-raspi2 systemd[1]: fail2ban.service: Unit entered failed state.
Jul 08 16:08:05 nc-raspi2 systemd[1]: fail2ban.service: Failed with result 'exit-code'.

**Solution:** 
apt-get purge fail2ban && apt-get install fail2ban

$ /etc/init.d/fail2ban status
  fail2ban.service - Fail2Ban Service
   Loaded: loaded (/lib/systemd/system/fail2ban.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2017-07-08

[https://github.com/Fourdee/DietPi/issues/475#issuecomment-285673712](url)

sal666, 2017-07-21:
Please, update berryboot testing image. Last one in http://dietpi.com/downloads/testing/ is from December.

MichaIng, 2017-08-17:
Finally Stretch became stable for Raspbian: https://www.raspberrypi.org/downloads/raspbian/

WolfganP, 2017-08-17:
Will it change anything in Dietpi? I'm under the impression that @Fourdee updated kernel components a while ago (or maybe I messed with my system a lot :-)

MichaIng, 2017-08-17:
Yeah the kernel is updated together with dietpi regularly, but this has not much to do with the Distro version.

Raspbian still based on Debian Jessie until yesterday, thus DietPi for RPi did the same, if you didn't flash the beta above.
Stretch brings a lot of new package major versions, that were frozen in Jessie, most importantly for my use cases PHP7.0 (instead of PHP5.6), which also changes locations of configuration files etc. This will be the same for some other packages I guess.
So after full upgrade all DietPi images to Stretch, much code about version checks etc. can be cleaned up 😄 .

rdslw, 2017-08-22:
Can you provide DietPi testing build for Pine64? I'll give it a shot and test.

molnarti, 2017-09-02:
stretch will have to be installed from scratch, or an update from jessie will be possible?

Fourdee, 2017-09-03:
@molnarti 
> stretch will have to be installed from scratch, or an update from jessie will be possible?

Stretch installs, will require a clean installation with the Stretch image.

Its not feasible for us to upgrade existing systems, as we have no way of knowing if the user has applied manual changes to linux settings. User input is also required for distro upgrade, which again, could change depending on what the user has installed/configured (we cant predict outcome).

k-plan, 2017-09-03:
@Fourdee 

> 🈺  RPi Monitor: Memory missing: http://dietpi.com/phpbb/viewtopic.php?f=9&t=986 (external bug)

Where is a new upcoming version: RPi-Monitor v2.12-r0

https://github.com/XavierBerger/RPi-Monitor-deb/tree/master/packages

Tested with `2017-08-16-raspbian-stretch.img` and it work.

[image]

Because of new strech ethernet interface naming `enx(+mac-address)`, user have to edit `/etc/rpimonitor/template/network.conf` but it works as desired.

atiensivu, 2017-09-03:
I upgraded my Pine64 install to Stretch and then, since it is my experimental build, went up to sid. 
Works fine overall but I don't have a bunch of packages installed, so it might be a premature 'works fine'. 

I find the Stretch ethernet naming a bit gross but I'm sure there is a good reason for it.

Fourdee, 2017-09-04:
@atiensivu 

> I find the Stretch ethernet naming a bit gross but I'm sure there is a good reason for it.

We revert to wlanx/ethx naming in DietPi v155:
https://github.com/Fourdee/DietPi/blob/master/dietpi/patch_file#L2733-L2734

Bridouz, 2017-09-27:
Is it possible to boot with DietPi Stretch for the first time with an enabled wifi configuration ?

With DietPi Jessie the `dietpi.txt` file managed to do that but I don't manage to activate wifi on first boot with Stretch ? Am I missing something ?

Fourdee, 2017-09-27:
@Bridouz 

> With DietPi Jessie the dietpi.txt file managed to do that but I don't manage to activate wifi on first boot with Stretch ?

Thanks for the report, i'll try to replicate 👍

Bridouz, 2017-09-27:
DietPi Stretch (v151) installed with ethernet connexion. Wifi is not working out of the box.

By enabling wifi in DietPi-Config Raspbian wants to install `netplug`
A reboot after the installation of `netplug` and WIFI SSID are available.

However I tried (after the wifi setting) to disable Ethernet. By doing it it removes `netplug` and the Pi is now not accessible from either Ethernet or Wifi.

Bridouz, 2017-09-28:
@Fourdee 

Certbot is now packaged on Stretch and I can confirm it works with the `webroot` plugin out of the box.

Would you authorize me to propose a PR with changes reflecting that ?

Fourdee, 2017-09-28:
> Certbot is now packaged on Stretch and I can confirm it works with the webroot plugin out of the box.
Would you authorize me to propose a PR with changes reflecting that ?

Yep by all means 👍

MichaIng, 2017-09-28:
Yes certbot from repo also working well for me on raspbian stretch.

Bridouz, 2017-09-28:
@MichaIng Thanks for mentioning `python-certbot-nginx`. I've just tested it and it's really nice, it works really great without any user input (except the fact I had to add my ddns domain to nginx.conf).

The ssl/cipher config is rated B from Qualys SSL Labs. Not bad but could be enhanced (with Mozilla recommendations https://mozilla.github.io/server-side-tls/ssl-config-generator/)

What do you think about Certbot : 

- Using `certbot certonly --webroot` which works but requires a *well-known* location in server config
- Using certbot --nginx/apache which does the work without any user input + add a bit of ssl/cipher to server config file (which could not be bad afterall)

MichaIng, 2017-09-29:
@Fourdee Does the problem with kodi still exist? I am wondering, because stretch is already official released also for raspbian with kodi in default repo.

`apt install kodi --dry-run` ran without error, I hesitate to try the real installation with 100+ dependency packages 😛.

kotamotov, 2017-10-06:
Hi Fourdee and thank you for your work! 
Please explain what Linux kernel is being used for DietPi_SparkySBC-armv7-(Stretch).7z image?

Fourdee, 2017-10-07:
@kotamotov 

Hi, its 3.10 currently:
https://github.com/Fourdee/DietPi/issues/1176#issue-263619322

Bridouz, 2017-10-09:
@Fourdee

No wifi out of the box on Dietpi Stretch V156 image.

In dietpi-config onboard wifi is active but wifi module is not. Need to activate it (which install netplug) reboot and then I'm able to scan SSID.

Strange because on latest Raspbian image wifi works OOTB withou `netplug` package.

MichaIng, 2017-10-09:
Maybe due to recent debian 9.2 release? https://www.debian.org/News/2017/20171007

Fourdee, 2017-10-10:
@Bridouz 

> In dietpi-config onboard wifi is active but wifi module is not. Need to activate it (which install netplug) reboot and then I'm able to scan SSID.

Confirmed, in testing i found 8/10 reboots results in scan failure:
```dmesg```
```
[    4.309924] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[    4.309934] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
```

Its really early on in boot, and we do nothing differently for Jessie/Stretch in DietPi, not sure on cause yet.

Same errors with ```netplug``` removed.

molnarti, 2017-10-12:
i tried to install nextclout on the most recent RPi stretch image and got the following error

`2017-10-12 23:48:32 (1.03 MB/s) - ‘/var/www/apc.php’ saved [38125/38125]

 [Info] Apt-get is processing, please wait...

E: Command line option 'i' [from -intl] is not understood in combination with the other options.
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.

 [Info] Apt Log:
E: Command line option 'i' [from -intl] is not understood in combination with the other options.

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues`

MichaIng, 2017-10-12:
@molnarti Are you using testing branch? Try it with the newest version of dietpi-software, just replace your version of it. This error occurred during our Nextcloud installation optimization and was fixed with the recent PRs.

Fourdee, 2017-10-13:
@MichaIng 

>  @Fourdee is it expected that the Stretch images also contain testing branch, as there is sometimes stuff still in development that has not necessarily something to do with Stretch? Also the image is named v156 but contains as mentioned testing code from v157?

Correct, when generating images I use the testing branch. Allows us to apply fixes to the image that are in devolpment. 

But yes, seems this one caught the Nextcloud dev at a bad stage, regardless, will be fixed with v157 release, ETA next week.

@molnarti 

**You can fix this issue (before the v157 is released) by doing the following:**
```
wget https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-software -O /DietPi/dietpi/dietpi-software
```
Nextcloud should now install.

[26 more replies]