MichaIng/DietPi #478: DietPi-Software | pi-hole/AdminLTE update hint

k-plan, 2016-08-21:
@Fourdee 

[image]
really annoying.

Fourdee, 2016-08-21:
> really annoying.

@k-plan 

Yep. I'am going to have a rethink about Pihole on DietPi and how we install it. Possibly get rid of our Pihole/fork and use their master branches.

Whats the update link target URL?

k-plan, 2016-08-21:
@Fourdee 

> Whats the update link target URL?

https://github.com/pi-hole/AdminLTE/releases

changes: https://github.com/pi-hole/AdminLTE/wiki/Changelog

_new logo, used ram in footer, cpu temp. in header, ..._

k-plan, 2016-08-21:
@Fourdee 

> really annoying.

... is the alert popup on every refresh : https://github.com/pi-hole/AdminLTE/blob/master/header.php#L96

... and three PayPay donation links on one single webpage!
... and loading "https://www.paypalobjects.com/en_US/i/scr/pixel.gif" on every refresh!

I`m back on my old (modded) Pi-hole 2.6.3 ...

Fourdee, 2016-08-25:
> Yep. I'am going to have a rethink about Pihole on DietPi and how we install it. Possibly get rid of our Pihole/fork and use their master branches.

Now i remember why we have a fork.  
- Their install script (https://github.com/pi-hole/pi-hole/blob/master/automated%20install/basic-install.sh) only allows for Lighttpd.
- ```/var/www/html```, we use ```/var/www/pihole```
- We dont need to install Git packages.

Ideally, I'd like to rewrite their basic_install.sh script, allow for either Lighttpd/Nginx/Apache2, would make this installation much simplier. 
But for me at the moment, its quicker to fork, make some minor changes (eg: disable the update check) and update Pihole forks every < 1month.

@k-plan 
Removed the update notifications:
[image]

k-plan, 2016-08-25:
@Fourdee 

tested on Pine64 with nginx:

- update notification has gone :smile: 
- click on `Pi-hole` in top right corner doesn't open a pop-up like on your screen-shot
- `Temp:` didn't work on Pine64

[image]

Fourdee, 2016-08-25:
@k-plan 
No menu seems to be related to Nginx.

Temp issue related to PineA64, fine on C2.

I dont have a Pine setup at the moment, whats the valuess of:
```
cat /sys/class/thermal/thermal_zone0/temp
cat /sys/devices/platform/sunxi-i2c.0/i2c-0/0-0034/temp1_input
```

Fourdee, 2016-08-27:
@k-plan 

> click on Pi-hole in top right corner doesn't open a pop-up like on your screen-shot

> No menu seems to be related to Nginx.

Yep, something I broke on our fork. Will take a look.

k-plan, 2016-08-27:
@Fourdee 

> I dont have a Pine setup at the moment, whats the values of:

sorry, but my Pine64 is 50 miles away and offline. So i can do it earliest next week.

But, if I remember it correct, it was:

```
cat /sys/class/thermal/thermal_zone0/temp
52
```
Verify it next week.

Fourdee, 2016-08-27:
@k-plan 
Turns out Java script needs quotations lol 

Fixed :)

> cat /sys/class/thermal/thermal_zone0/temp
52

PiHole must be pulling temp in a different way, I'll test on Pine here and check.

Fourdee, 2016-08-27:
Done, temp readout now working on Pine A64,  I'll send a PR to Pihole with above changes.

closed 2016-08-27 by Fourdee

Fourdee, 2016-08-27:
> I'll send a PR to Pihole with above changes.

https://github.com/pi-hole/AdminLTE/pull/129