MichaIng/DietPi #479: DietPi-Software | Revisit and test Letsencrypt with Lighttpd and Nginx

Fourdee, 2016-08-21:
>

montreuilg, 2016-09-12:
My goals are to set up the dietpi3s as RFM95 lora/ethernet gateways using lighttpd and SQLite, so I'm ready to help if can. :-)

Fourdee, 2016-09-12:
@montreuilg 

> so I'm ready to help if can. :-)

Excellent :+1:
If you would be willing to test a LLSP installation using aar0nism's guide: http://dietpi.com/phpbb/viewtopic.php?f=9&t=603&p=2625#p2624, then report results, this will help speed things up.

Fourdee, 2016-09-12:
Lighttpd code is done. Just tested with 2 certs. Had to buy a 99p domain lol.

I'll send the commit and let users know when its ready for testing.

[image]

[image]

Fourdee, 2016-09-12:
@montreuilg 

Good to go. 

#### If you (and/or anyone else) could test the following using the testing branch, i'd be grateful:
- Setup a new installation of DietPi for testing branch: https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md
- ```dietpi-software``` select LLSP for installation
- ```dietpi-software``` select Certbot for installation
- Start the installation
- After reboot, run ```dietpi-letsencrypt```, enter your details and create the cert.

Fourdee, 2016-09-16:
Successful in my testing. Marking this as closed.

Please reopen if required. 

NB: No Nginx support yet, if we get requests for this, i'll look into it.

closed 2016-09-16 by Fourdee

montreuilg, 2016-09-19:
Thanks but, Oups, i am not on the same beat as you guys are it seems! I was going to check that today, i had a few hours of playtime... 🔢 :-)

I'll reinstall and keep on trying to make that small tft working then :-(

 One thing I'll take time to look at and i think you should too for developpers: [RethinkDB](https://www.rethinkdb.com/docs/quickstart/) It would be great to have it as part of an IOT offer lets say ;-)

k-plan, 2016-12-23:
@Fourdee 

> NB: No Nginx support yet, if we get requests for this, i'll look into it.

New request:
http://dietpi.com/phpbb/viewtopic.php?f=9&t=1045

LexiconCode, 2016-12-25:
Excellent, thank you @Fourdee @k-plan

LexiconCode, 2016-12-25:
Alternate Certbot clients that might more easily integrate in to dietpi and all three web services.
https://github.com/acmephp/acmephp
https://github.com/kuba/simp_le

sootstone, 2017-06-05:
Certbot does now provide a plugin for nginx since last October [https://certbot.eff.org/docs/using.html#nginx](https://certbot.eff.org/docs/using.html#nginx)

It's still technically in alpha, but may be something worth considering?

**I have not tested this!** Might give it a go though.

sootstone, 2017-06-09:
Worked fine for me with standard server block:

```
server {
      listen 80;
      listen [::]:;
      server_name domain.com;
}
```

Basically does the same as the --apache plugin.