MichaIng/DietPi #482: DietPi | Create image for NanoPi M2

Fourdee, 2016-08-26:
Many thanks to @k-plan for sending the board (and some Haribo :laughing: ) :+1:

Fourdee, 2016-08-26:
No ARMbian support for this one (yet), so i'll try the official Debian images.

Fourdee, 2016-08-26:
Image is done and available on http://dietpi.com/download

> http://dietpi.com/phpbb/viewtopic.php?f=8&t=620
DietPi for NanoPi M2 is based on the official Friendly ARM image, which, has been completely stripped and optimized for DietPi.

Need to try and fix:
- CPU temp readouts

Fourdee, 2016-08-27:
Marking this as done. Unfortunately, no CPU temp readouts available at this time.

closed 2016-08-27 by Fourdee

k-plan, 2016-08-27:
@Fourdee 

yes, I didn't find any CPU Temp. value. Perhaps something in the prebuild debian image is missing.

And the ext4 boot partition is as well annoying.

> Feels like 65'c+ on touch.

Yes, correct. My [Pyrometer](https://en.wikipedia.org/wiki/Pyrometer) shows up as well something like that. Better to put on a heatsink for Samsung Cortex-A9 S5P4418 CPU.

But anyway, nice job Fourdee. 👍

Fourdee, 2016-08-27:
@k-plan 

> And the ext4 boot partition is as well annoying.

Yep, I know, same here. 
If i get some time, i'll look at compiling from scratch using: http://wiki.friendlyarm.com/wiki/index.php/NanoPi_M2#Make_Your_Own_OS_Image and hopefully set vfat and fix cpu temp readout.

> My Pyrometer shows up as well something like that. Better to put on a heatsink for Samsung Cortex-A9 S5P4418 CPU.

I'll add a Pyrometer to my wish list :+1: I've got a spare RPi heatsink, tiny, but should be better than nothing :)