MichaIng/DietPi #484: Got display problem after kodi exit

467815891a, 2016-08-29:
I setup v129 on my Rasberrypi2.Use VGA to HDMI Cable connect to my monitor(1366*768).
I modify /boot/config.txt and force Raspberrypi output with 1366*768 resolution.It works well.
I sure my Raspberrypi got enough current and VGA to HDMI Cable got External power.
I used dietpi-software to install kodi. kodi plays video perfectly.But after I exit kodi,I only saw my desktop in a flash,and soon it jump to a "abnormal" desktop.The resolution and color are both changed Forgive my poor english,I can't describe it,but I took picture.
This is normal my Desktop before I run kodi: http://5aimiku.com/wp-content/uploads/2016/08/2016082913214040.jpg
This is "abnormal" Dsktop after exit kodi: http://5aimiku.com/wp-content/uploads/2016/08/2016082913210273.jpg
I try to change Desktop environment frome LDXE to XFACE4,turn Overscan on or off,but It doesn't change anything.I need some advice...

Fourdee, 2016-08-29:
@467815891a 

Looks like a possible bug with the "black screen kodi" workaround we use on RPi: https://github.com/Fourdee/DietPi/blob/master/dietpi/misc/start_kodi#L34-L40

I'll take a look. Thanks for report :+1:

467815891a, 2016-08-30:
Thanks for reply, I am very appreciated for your work!

Fourdee, 2016-08-30:
@467815891a 

Yep, seems the workaround is no longer needed. Possibility added to Kodi binary.

I'd appreciate it if you can confirm this update has resolved the issue.
Please run the following command:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/misc/start_kodi -O /DietPi/dietpi/misc/start_kodi
```
Then launch Kodi from your desktop, followed by exit.

Fourdee, 2016-08-31:
Tested and working for me on RPi3. 

@467815891a I'll mark as closed, please reopen if required.

closed 2016-08-31 by Fourdee