MichaIng/DietPi #488: DietPi C2 | fdisk utility cannot create a partition table > 2 TB.

Zone22, 2016-09-01:
I have a 4 TB external hard drive hooked up to Odroid - C2. When formatted only 2 terabytes are available. Some research illuminates that technically the fdisk utility cannot create a partition table larger than 2 TB.

Technical details
>  The size of this disk is 3.7 TiB (4000787025920 bytes). DOS partition table format can not be used on drives for volumes larger than 2199023255040 bytes for 512-byte sectors. Use GUID partition table format (GPT).

The solution is to use GNU parted command with GPT. It supports Intel EFI/GPT partition tables. You
must include GPT support in kernel in order to use GPT. If you don’t include GPT support in Linux kernelt, after rebooting the server, the file system will no longer be mountable or the GPT table will get corrupted.

In short you need to recompile the kernel. Set CONFIG_EFI_PARTITION to y to compile this feature.  Then use a different utility besides fdisk.

Resources
http://www.cyberciti.biz/tips/fdisk-unable-to-create-partition-greater-2tb.html
GPT fdisk
http://www.rodsbooks.com/gdisk/

maximm07, 2016-09-10:
I also have this problem with 5TB drive..any chance this enhancement getting into v.131?

Fourdee, 2016-09-10:
@maximm07 @Zone22 

I'll take a look in v131, but cant promise anything. Code should be easy, but, we may hit issues with kernels not having GPT enabled on various devices. Could be a nightmare.

Fourdee, 2016-09-10:
RPi kernel GPT support = :u6307: 
```
root@DietPi:~# parted /dev/sda
GNU Parted 3.2
Using /dev/sda
Welcome to GNU Parted! Type 'help' to view a list of commands.
(parted) print
Model: SanDisk Cruzer Fit (scsi)
Disk /dev/sda: 8003MB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags:

Number  Start   End     Size    File system  Name        Flags
 1      1049kB  8002MB  8001MB  ext4         DietPi_USB
```

Fourdee, 2016-09-10:
Odroid C2:

[image]

[image]

Reboot:

```
root@DietPi:~# df -h | grep usb
/dev/sda1       7.3G   17M  6.9G   1% /mnt/usb_1
root@DietPi:~# parted
GNU Parted 3.2
Using /dev/sda
Welcome to GNU Parted! Type 'help' to view a list of commands.
(parted) print
Model: SanDisk Cruzer Fit (scsi)
Disk /dev/sda: 8003MB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags:

Number  Start   End     Size    File system  Name     Flags
 1      1049kB  8002MB  8001MB  ext4         primary
```

Zone22, 2016-09-10:
do you need this tested on the pine 64 and Odroid xu4?

Fourdee, 2016-09-10:
@Zone22 

> do you need this tested on the pine 64 and Odroid xu4?

Would be great if you could. :+1: Needs lots of user testing.
Use blank USB drives with no personal data, might be a few bugs we need to iron out.

Its on testing branch at the moment, so you'll need this: https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md

Zone22, 2016-09-12:
I'll test as soon as my power supply comes in 2 to 4 days.

maximm07, 2016-09-13:
I can test with C2 and 5TB WD drive. What kind of tests are required?

Fourdee, 2016-09-14:
@maximm07

> I can test with C2 and 5TB WD drive. What kind of tests are required?

Excellent :+1: 

#### Please do the following to test your drive:
This test will format your drive, so don't continue if you have data on the drive you want to keep.
- Unplug the USB drive from the system before hand.
- Follow these steps to install a DietPi image using the testing branch: https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md
- Once the system is finished installing, run ```dietpi-drive_manager```
 - The drive should appear at the top of the list, select it
 - Go to ```format```
 - Select ```GPT``` and confirm format
- Once the format is finished, check it shows ```5TB~``` free space.

Fourdee, 2016-09-16:
I'll mark this as closed, successful in my testing. 

@Zone22 @maximm07 
Pending results from your tests, please reopen if you experience any issues.

closed 2016-09-16 by Fourdee