MichaIng/DietPi #489: DietPi-Software C2 | Fail2Ban blocks SSH access by default.

Zone22, 2016-09-01:
The current set up of Fail2Ban seems to be blocking SSH access upon install on C2. 
Uninstalling Fail2Ban resolves the issue.

Fourdee, 2016-09-01:
@Zone22 
Will take a looks, thanks for reporting :+1:

Fourdee, 2016-09-02:
Had a RPi 3 testbed setup, was unable to replicate with:
- rpi 3 
- dropbear

Will test C2.

@Zone22 
Which SSH server did you have installed?

Fourdee, 2016-09-02:
@Zone22 

Working fine on C2 + OpenSSH for me.

What is status of service:
```
systemctl status fail2ban -l
```

And logfile contents:
```
cat /var/log/fail2ban.log
```

Zone22, 2016-09-02:
I'm backing up the SD card to an image. Then I will reimage a fresh install and report back soon.

Zone22, 2016-09-02:
Actually I get a crash upon reboot during install which I didn't notice Because the C2 wasn't hooked up to a monitor. Which gave the appearance of Fail2Ban blocking SSH when it was a system crash. After a hard reset everything functions as expected. I've sent a bug report linking back to this post so you have all the logs.  Reference CODE: f794c177-1044-4061-9246-f53665a51567-0  

Replication procedures.
1. Fresh Image of DietPi_v127_OdroidC2-arm64-(Jessie)
2. Hook up external hard drive USB
3. Let the automatic update and reboot complete.
4. Follow the wizard for using andreformatting external hard drive
5. in dietpi-software change preferences to
- SSH Server           : OpenSSH  
- File Server          : ProFTP  
- Webserver Preference : Apache2 

6. under Software Optimized   Select DietPi Optimized Software for installation select
[*] 73   Fail2Ban: prevents brute-force attacks with ip ban   
7.  select Go  >> Start Installation.  
8.  Wait for the packages to be installed and the system will restart.
9.  The system will restart then upon boot crash with stack trace.
10. Then do a Hard reset and everything functions as expected.

Requested output of commands after crash.
```
root@DietPi:~# systemctl status fail2ban -l
● fail2ban.service - LSB: Start/stop fail2ban
   Loaded: loaded (/etc/init.d/fail2ban)
   Active: active (running) since Fri 2016-09-02 11:32:16 EDT; 3min 29s ago
  Process: 916 ExecStart=/etc/init.d/fail2ban start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/fail2ban.service
           └─927 /usr/bin/python /usr/bin/fail2ban-server -b -s /var/run/fail2ban/fail2ban.sock -p /var/run/fail2ban/fail2ban.pid

Sep 02 11:32:16 DietPi fail2ban[916]: Starting authentication failure monitor: fail2ban.
Sep 02 11:32:16 DietPi systemd[1]: Started LSB: Start/stop fail2ban.
root@DietPi:~# cat /var/log/fail2ban.log
2016-09-02 11:32:16,484 fail2ban.server [927]: INFO    Changed logging target to /var/log/fail2ban.log for Fail2ban v0.8.13
2016-09-02 11:32:16,487 fail2ban.jail   [927]: INFO    Creating new jail 'ssh'
2016-09-02 11:32:16,488 fail2ban.jail   [927]: INFO    Jail 'ssh' uses poller
2016-09-02 11:32:16,565 fail2ban.jail   [927]: INFO    Initiated 'polling' backend
2016-09-02 11:32:16,569 fail2ban.filter [927]: INFO    Added logfile = /var/log/auth.log
2016-09-02 11:32:16,571 fail2ban.filter [927]: INFO    Set maxRetry = 6
2016-09-02 11:32:16,576 fail2ban.filter [927]: INFO    Set findtime = 600
2016-09-02 11:32:16,578 fail2ban.actions[927]: INFO    Set banTime = 600
2016-09-02 11:32:16,738 fail2ban.jail   [927]: INFO    Creating new jail 'dropbear'
2016-09-02 11:32:16,739 fail2ban.jail   [927]: INFO    Jail 'dropbear' uses poller
2016-09-02 11:32:16,741 fail2ban.jail   [927]: INFO    Initiated 'polling' backend
2016-09-02 11:32:16,744 fail2ban.filter [927]: INFO    Added logfile = /var/log/auth.log
2016-09-02 11:32:16,747 fail2ban.filter [927]: INFO    Set maxRetry = 6
2016-09-02 11:32:16,751 fail2ban.filter [927]: INFO    Set findtime = 600
2016-09-02 11:32:16,753 fail2ban.actions[927]: INFO    Set banTime = 600
2016-09-02 11:32:16,839 fail2ban.jail   [927]: INFO    Jail 'ssh' started
2016-09-02 11:32:16,854 fail2ban.jail   [927]: INFO    Jail 'dropbear' started
root@DietPi:~#

```

Fourdee, 2016-09-02:
@Zone22 

>Replication procedures.
1. Fresh Image of DietPi_v127_OdroidC2-arm64-(Jessie)
2. Hook up external hard drive USB
3. Let the automatic update and reboot complete.
4. Follow the wizard for using andreformatting external hard drive
5. in dietpi-software change preferences to
SSH Server : OpenSSH
File Server : ProFTP
-Webserver Preference : Apache2 
\6. under Software Optimized Select DietPi Optimized Software for installation select
[*] 73 Fail2Ban: prevents brute-force attacks with ip ban
7. select Go >> Start Installation.
8. Wait for the packages to be installed and the system will restart.
9. The system will restart then upon boot crash with stack trace.

Unable to replicate the crash. Everything working.
I just wondering if the image may be corrupt, i'll run a fscheck on image.

EMMC or SD?
Official HK PSU?

Fourdee, 2016-09-02:
Image checks out fine, I run it prior to image resize. Terminal error is due to vfat not supporting fscheck:
[image]

Zone22, 2016-09-02:
EMMC or **SD** a class 10 Samsung
As for the power supply I think that might be our problem although I need to test. I had thought my Motorola charger was to DC 5Volt / 2A output when indeed it is DC 5Volt / 1,100 mA = 1.1A.

Fourdee, 2016-09-02:
> DC 5Volt / 1,150 mA = 1.15A.

Highly possible cause.

> I wonder if there's some sort of stress testing tool for ARM (like prime 95 but for PC) that could be integrated into DietPi-launcher. That would be useful for ferreting out power supply issues and overclocking.

@Zone22 

Stress Test. A really really good idea, great suggestion! :+1: https://github.com/Fourdee/DietPi/issues/492

Fourdee, 2016-09-04:
@Zone22 
I'll mark this as closed due to 1A~ PSU. Please reopen if required.

closed 2016-09-04 by Fourdee