MichaIng/DietPi #490: DietPi-Software | Explain how to add new installations to the sourcecode (for contributors)

Fourdee, 2016-09-01:
As per @zone22's suggestion: https://github.com/Fourdee/DietPi/issues/168#issuecomment-244120375

Fourdee, 2016-09-02:
## Adding new software to DietPi-Software.
### For this example we will add Owncloud to DietPi-Software:

To get a full list of DietPi-Software installation options:
```
dietpi-software list
```

#### Obtain a unique installation Index for this software:
Run the following command, make a note of this number.
```
dietpi-software list | grep 'index Current'
```
A unique Index of ```47``` will be used in this guide.

#### Choose a category index for the software
https://github.com/Fourdee/DietPi/blob/96dbdc5b39957872e8e7d4230e0d25d7d8f8e765/dietpi/dietpi-software#L265-L283
```4``` = Cloud / Backups

#### Check pre-reqs. Does it require other software to be installed?:
https://github.com/Fourdee/DietPi/blob/96dbdc5b39957872e8e7d4230e0d25d7d8f8e765/dietpi/dietpi-software#L357-L368
In this example, Owncloud will require:
```
aSOFTWARE_REQUIRES_WEBSERVER[$index_current]=1
aSOFTWARE_REQUIRES_MYSQL[$index_current]=1
```

#### Add the new Owncloud entry:
You should end up with the entry below:
https://github.com/Fourdee/DietPi/blob/96dbdc5b39957872e8e7d4230e0d25d7d8f8e765/dietpi/dietpi-software#L709-L719

#### Add the installation code:
Inside function ```Install_Dietpi_Software()```
https://github.com/Fourdee/DietPi/blob/96dbdc5b39957872e8e7d4230e0d25d7d8f8e765/dietpi/dietpi-software#L2666-L2698

#### Add the configuration code:
Inside function ```Install_Apply_Configs()```
https://github.com/Fourdee/DietPi/blob/96dbdc5b39957872e8e7d4230e0d25d7d8f8e765/dietpi/dietpi-software#L5322-L5388

Thats it. 
If successful, your new entry will be available from dietpi-software optimized menu and ```dietpi-software list```

closed 2016-09-02 by Fourdee

Zone22, 2016-09-06:
Note to others you can use FTPs to transfer your edited files into an existing Dietpi installs in for testing purposes. An alternative to using your own get repository testing branch.

Fourdee, 2016-09-06:
@Zone22 
>What are the files such as DietPi-Software written as? 

Bash / Unix Shell.

As for debugging, the only one I know and use is: http://www.shellcheck.net/

svh1985, 2019-09-09:
`dietpi-software list | grep 'index Current'` does not work anymore. :)

MichaIng, 2019-09-11:
@svh1985 
You can use `dietpi-software free` now to obtain the first free software ID that can be used for the new implementation.
See the new HowTo in wiki: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

foxy82, 2021-06-04:
Note: `Install_Dietpi_Software()` is now called: `Install_Software() `
And: `Install_Apply_Configs()` is now called: `Configure_Software()`

MichaIng, 2021-06-04:
Many thanks. With last DietPi update, it has changed quite a bit. I'll update that page.

Joulinar, 2021-06-04:
I guess whole wiki section we would need to rework. https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title
Or should it go to docs?

MichaIng, 2021-06-04:
I'm currently updating the Wiki. I think for now it's okay to keep such contributor/dev infos in the Wiki, as there is no developer section in the docs.