MichaIng/DietPi #492: DietPi-Config | Add Stress Test

Fourdee, 2016-09-02:
As per @zone22's suggestion: https://github.com/Fourdee/DietPi/issues/489

Must test:
- CPU
- Filesystem IO
- RAM

Fourdee, 2016-09-04:
```dietpi-config``` > tools > Stress Test
[image]

closed 2016-09-04 by Fourdee

k-plan, 2016-09-04:
@Fourdee 

nice work! 😄  👍 

[image]

[image]

[image]

[image]