MichaIng/DietPi #493: Embed Gitter into README.md

Zone22, 2016-09-03:
Gitter is a chat and networking platform that helps to manage, grow and connect communities through messaging, content and discovery. Utilize Gitter https://gitter.im/ on DietPi Github page.

Fourdee, 2016-09-04:
@Zone22 
Great idea, but i'am not sure at the moment. 
If this will help reduce the time I have to spend supporting our users across multiple sites, i'am all in. But at the moment I have enough on my plate keeping these monitored and providing replies:
- Git
- Forums: http://dietpi.com/phpbb/
- RPi forum: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=100976
- C1: http://forum.odroid.com/viewtopic.php?f=114&t=11962&p=83691#p83691
- C2: http://forum.odroid.com/viewtopic.php?f=138&t=19948
- XU4: http://forum.odroid.com/viewtopic.php?f=98&t=15700
- OPi PC: http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=662&extra=page%3D1
- Twitter: https://twitter.com/dietpi_

We had IRC a while back, but I had to drop it, was getting silly. Spent more time chatting than actually doing work on DietPi.

k-plan, 2016-09-04:
@Fourdee 

> If this will help reduce the time I have to spend supporting our users across multiple sites, i'am all in.

No, I think it will enlarge support effort.

It is 	currently  more when we can fulfill.

But gitter is nice for chatting with other DietPi supporter and developer. (closed group)

Zone22, 2016-09-05:
I think if you were to purposely drive communication from the various Forums to Gitter you could actually decrease the support effort. If you only use the Forums for posting information and only offering support through Gitter and Git. Although at the end of the day you guys should do what you think is best. 

If you guys wanted to create a developer and supporter group using  [Slack](https://slack.com/) which allows for closed groups unlike Gitter that would be most interesting to me.

Fourdee, 2016-09-05:
> I think if you were to purposely drive communication from the various Forums to Gitter you could actually decrease the support effort.

@Zone22 

Good point :+1: 

I'll set Gitter up and give it a spin :)

closed 2016-09-27 by Zone22