MichaIng/DietPi #494: DietPi-Software | UrBackup Server and Client

Zone22, 2016-09-03:
The Server and Client has Arm and Arm64 builds

UrBackup is an easy to setup Open Source client/server backup system, that through a combination of image and file backups accomplishes both data safety and a fast restoration time.

File and image backups are made while the system is running without interrupting current processes.

UrBackup also continuously watches folders you want backed up in order to quickly find differences to previous backups. Because of that, incremental file backups are really fast.

Your files can be restored through the web interface, via the client or the Windows Explorer while the backups of drive volumes can be restored with a bootable CD or USB-Stick (bare metal restore).

A web interface makes setting up your own backup server really easy. For a quick impression please look at the screenshots here. 

https://www.urbackup.org/

Fourdee, 2016-09-04:
@Zone22 
Looks good :+1:  i'll take a look in v131.

Fourdee, 2016-09-07:
notes: 
compile needed for ARM64 and armv6

Zone22, 2016-09-12:
Just a quick note the client needs compiling there are ARM64 and armv7 server packages

Fourdee, 2016-09-17:
Was about to start testing the server:
[image]

Fourdee, 2016-09-17:
Annoyances:
- You cant stop backups on the client, with the client program
- Clicking the Stop button on a active backup, most of the time, doesn't stop the backup
- Clients ignore the server settings of, disabled, auto file and image backups
- Client always wants to resume, runs indexing, then checks if the server activity actually exists. Then stops.
- Various Client hangs, requires urbackup service restart.

This program is riddled with bugs.

k-plan, 2016-09-17:
@Fourdee 

Yes, server looks good.
Client for Windows (_yes, it's not my world_) is crap! Sorry.

Fourdee, 2016-09-17:
Server:
- ARMv7+ / AMD64 = Deb
- ARM64 = Sourcebuild | Unable to find a config setting in ```/etc/default/urbackupsrv``` or another config location for backup directory (set to ```C:\``` by default)

Fourdee, 2016-09-17:
To Do:
- 🈴 Client installation option
- 🈴 Uninstall code for both client and server
- 🈴 Online Docs
- 🈴 Would be nice: ARM64. Find a way to set default backup directory.

Zone22, 2016-09-17:
Well I'm open to an alternative if it seems to be as robust at least of the server-side.

Fourdee, 2016-09-19:
@Zone22 

> Well I'm open to an alternatives.
The following are open source multiplatform solutions but further investigation is needed for arm support

They sound good, but they all need testing to find the "best one". If you can pick one alternative to investigate, create a ticket and we will take a look at adding it in the future.

Fourdee, 2016-09-19:
Marking this as closed: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=65#p65

closed 2016-09-19 by Fourdee

Zone22, 2016-09-21:
@k-plan I see your frustration and also there updating the client soon. Take a look at their blog!
http://blog.urbackup.org/