MichaIng/DietPi #495: DietPi-Software | OMV Open Media Vault

k-plan, 2016-09-04:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=561&p=2457#p2457

> https://github.com/Fourdee/DietPi/issues/455#issuecomment-238925807

_Please note:_
https://en.wikipedia.org/wiki/OpenMediaVault#Release_history

- _Current openmediavault v2.2.6 (Stone Burner) only  support Debian 7 (Wheezy)_ 

- _Next version openmediavault v3.x (Erasmus) will support Debian 8 (Jessie), but it is still in beta for now ..._

Fourdee, 2016-09-05:
Erasmus binaries: http://packages.openmediavault.org/public/dists/erasmus/main/

k-plan, 2016-12-23:
New request:

http://dietpi.com/phpbb/viewtopic.php?f=12&t=1040

Fourdee, 2017-01-06:
Requires:
- ProFTPD
- Samba Server
```
echo "deb http://packages.openmediavault.org/public erasmus main" > /etc/apt/sources.list.d/openmediavault.list
apt-get update
apt-get install -y --force-yes openmediavault-keyring 

apt-get update
apt-get install -y openmediavault postfix # Must automate postfix whiptail, "no configuration" option?
omv-initsystem
```

```
root@DietPi:~# debconf-get-selections | grep postfix
postfix postfix/tlsmgr_upgrade_warning  boolean
postfix postfix/bad_recipient_delimiter error
postfix postfix/root_address    string
postfix postfix/mydomain_warning        boolean
postfix postfix/protocols       select
postfix postfix/relayhost       string
postfix postfix/procmail        boolean
postfix postfix/retry_upgrade_warning   boolean
postfix postfix/chattr  boolean false
postfix postfix/destinations    string
postfix postfix/mailname        string  /etc/mailname
# Install postfix despite an unsupported kernel?
postfix postfix/kernel_version_warning  boolean
postfix postfix/sqlite_warning  boolean
postfix postfix/rfc1035_violation       boolean false
postfix postfix/main_mailer_type        select  No configuration
postfix postfix/mynetworks      string  127.0.0.0/8 [::ffff:127.0.0.0]/104 [::1]/128
postfix postfix/mailbox_limit   string  0
postfix postfix/relay_restrictions_warning      boolean
postfix postfix/not_configured  error
postfix postfix/recipient_delim string  +
```

```
root@DietPi:~# debconf-get-selections | grep openmediavault
openmediavault  openmediavault/run-initsystem   note
```

Fourdee, 2017-01-06:
Ok done, just needs some testing.

Fourdee, 2017-01-10:
Marking as completed. 
Although, as OMV can take control of some DietPi features, until our users are able to test all areas of OMV's features, we won't know for sure if OMV, and all its plugins are fully compatible on a DietPi system.

closed 2017-01-10 by Fourdee