MichaIng/DietPi #500: DietPi-Software | Add Gitlab CE

jmarginier, 2016-09-06:
Hi,
May be this software package interest some peoples (me included) : 
https://about.gitlab.com/downloads/#raspberrypi2

With arm64/aarch64 support for odroid-c2 plz :)

midwan, 2016-09-06:
This is really good! Thanks for the heads-up!

Zone22, 2016-09-07:
Me 3

bhaveshgohel, 2016-09-07:
The steps provided in the [link](https://about.gitlab.com/downloads/#raspberrypi2) is also work on Pi3. i have tested it on dietpi.

phonicmouse, 2017-04-03:
Isn't this too heavy for a poor raspberry?

JoshuaJones, 2017-05-02:
@phonicmouse I've ran GitLab on a Pi3 running DietPi with passable results. It can chug/lag in my experience.

phonicmouse, 2017-05-02:
@JoshuaJones Whoa, I'll try it then.

tatoosh, 2017-10-22:
Any news to implement gitlab into dietpi?

Fourdee, 2017-10-26:
@tatoosh 

> Any news to implement gitlab into dietpi?

Not something I can take on at the moment. Hopefully one of our source contributors can pick it up.

closed 2017-12-30 by Fourdee