MichaIng/DietPi #501: DietPi + AmiBerry | Auto usbmount

Fourdee, 2016-09-07:


Fourdee, 2016-09-07:
@midwan

#### Q: Are the users trying to:
- Copy their .rom/.adf from a USB drive to the SD card?

or

- Run .rom/.adf directly from an existing USB drive?

or 

- Both?

midwan, 2016-09-07:
The most common scenario will be to:
- Plug in USB device which contains ROMs, ADFs, HDFs, etc.
- Copy those on the SD Card for use with the emulator

Many people prefer to get a large sd card (e.g. 8 or 16GB) and store everything in it, so they don't have to plug in anything extra after that.

Of course, the option to have the external USB device used "as-is" is very good, but most users are not familiar with that at the moment. I believe once they pick up on that, they might start using it more and more (maybe I should write a howto specifically for that).

Fourdee, 2016-09-07:
#### Notes:

:u6307: | Existing USB drive setup with DietPi
- Automatically mounts USB drive as per fstab, when plugged in. 
- Mount location is correct ```/mnt/usb_1```

NB: need to check dedicated USB drive scripts will work with this. EG: do they scrape ```df /mnt/usb_1```?

:u6307: : | No existing USB drive setup with DietPi
- Automatically mounts USB drive to ```/media/usb0```

Fourdee, 2016-09-07:
Notes to self:
Increases filesystem total by 606KB.
Install by default on all DietPi systems?

midwan, 2016-09-07:
Please check if you've also got filesystem support available, for FAT16, FAT32, NTFS...

Fourdee, 2016-09-07:
NTFS support for USBmount on Jessie:
```
		apt-get install usbmount -y
		sed -i '/^FILESYSTEMS=/c\FILESYSTEMS="vfat ext2 ext3 ext4 hfsplus ntfs-3g fuseblk"' /etc/usbmount/usbmount.conf
		cat << _EOF_ > /etc/udev/rules.d/usbmount.rules
KERNEL=="sd*", DRIVERS=="sbp2",         ACTION=="add",  PROGRAM="/bin/systemd-escape -p --template=usbmount@.service \$env{DEVNAME}", ENV{SYSTEMD_WANTS}+="%c"
KERNEL=="sd*", SUBSYSTEMS=="usb",       ACTION=="add",  PROGRAM="/bin/systemd-escape -p --template=usbmount@.service \$env{DEVNAME}", ENV{SYSTEMD_WANTS}+="%c"
KERNEL=="ub*", SUBSYSTEMS=="usb",       ACTION=="add",  PROGRAM="/bin/systemd-escape -p --template=usbmount@.service \$env{DEVNAME}", ENV{SYSTEMD_WANTS}+="%c"
KERNEL=="sd*",                          ACTION=="remove",       RUN+="/usr/share/usbmount/usbmount remove"
KERNEL=="ub*",                          ACTION=="remove",       RUN+="/usr/share/usbmount/usbmount remove"
_EOF_

		cat << _EOF_ > /etc/systemd/system/usbmount@.service
[Unit]
BindTo=%i.device
After=%i.device

[Service]
Type=oneshot
TimeoutStartSec=0
Environment=DEVNAME=%I
ExecStart=/usr/share/usbmount/usbmount add
RemainAfterExit=yes
_EOF_

systemctl daemon-reload
```

Fourdee, 2016-09-07:
Ok could probably do this with just systemD using auto FStype:
```
/dev/sda1       /mnt/usb_1      auto     defaults,noatime,nofail,permissions,x-systemd.automount  0       0
```

Then udev to simply run ```mount -a```

Fourdee, 2016-09-07:
Only downside is:
- Cant use ```permissions``` mount attribute thats used only by NTFS drives. As this prevents other FS types from mounting (eg: ext4).

Fourdee, 2016-09-07:
Done, automount works with:
- NTFS
- VFAT
- EXT*

closed 2016-09-07 by Fourdee