MichaIng/DietPi #502: DietPi + AmiBerry | Use OpenSSH server by default on image

Fourdee, 2016-09-07:
Requires new image: https://github.com/Fourdee/DietPi/issues/474#issuecomment-245259988

Fourdee, 2016-09-07:
#### notes:
- Updated from 4.4.16 > 4.4.19 firmware
- Tested image installation :u6307: 
- Uploaded image download: http://dietpi.com/downloads/images/DietPi-AmiBerry_RPi-armv6-(Jessie).7z

@midwan

Done, new image that will install OpenSSH server by default is available: http://dietpi.com/downloads/images/DietPi-AmiBerry_RPi-armv6-(Jessie).7z

Please note, usb auto mount (https://github.com/Fourdee/DietPi/issues/501) wont be available until we release v131:

midwan, 2016-09-07:
>    images/DietPi-AmiBerry_RPi-armv6-(Jessie).7z

Awesome, thank you for the quick response!

closed 2016-09-07 by Fourdee