MichaIng/DietPi #503: DietPi-Software | Mode: Automated reinstall ?

k-plan, 2016-09-07:
@Fourdee 

just updated to v130

```
Welcome to DietPi-Software

 DietPi-Software
-----------------------------------------------------
 Mode: Automated reinstall
 Please wait...

 [Info] 6: Xserver is not currently installed
 [Info] The program must be installed, before reinstall can be used
 [Ok] No changes applied for: Xserver
 [Ok] Patch 130 completed

```

Hmm, new feature?
It it possible to use this for update (Automated reinstall) more installed software, e.g. pi-hole without loosing  approach and data?

Think it is in some cases better when do a `dietpi-software => uninstall ` and trigger a brand new installation again.

Fourdee, 2016-09-07:
@k-plan 

> It it possible to use this for update (Automated reinstall) more installed software, e.g. pi-hole without loosing approach and data?

Its mainly for patching purposes and making my life easier ;). 
So instead of having to check if the user has program X installed, we can just run ```dietpi-software reinstall 108```. If the program is installed, it will run the installation procedure again (over the top of the existing one), effectively updating it.
Its untested on Pihole, but yes, thats the idea behind reinstall.

> Think it is in some cases better when do a dietpi-software => uninstall and trigger a brand new installation again.

Yep, for end users, they shouldn't be using ```dietpi-software reinstall```.

k-plan, 2016-09-08:
@Fourdee 

> Its mainly for patching purposes and making my life easier ;). 

Cool and very nice implemented. :smiley: 

> Its untested on Pihole and other programs at the moment, but yes, thats the idea behind reinstall.

Ask for pi-hole, because you hold a own DietPi fork of it.
Since v.2.9 they have a new update feature:

https://github.com/pi-hole/pi-hole/wiki/Changelog

> [NEW] - A much requested update method! Updating to 2.9 will require you use the standard install method one last time, but for future releases, the script will detect existing installs and not force you to have to enter all the settings again! Also added the command pihole -up to update everything from the command line with no prompts at all, so you can automate it if you like! 

```
root@NeoPi:~# pihole -h
::: Control all PiHole specific functions!
:::
::: Usage: pihole [options]
:::             Add -h after -w (whitelist), -b (blacklist), or -c (chronometer)  for more information on usage
:::
::: Options:
:::  -w, whitelist            Whitelist domains
:::  -b, blacklist            Blacklist domains
:::  -d, debug                Start a debugging session if having trouble
:::  -f, flush                Flush the pihole.log file
:::  -ud, updateDashboard      Update the web dashboard manually
:::  -up, updatePihole           Update Pi-hole
:::  -g, updateGravity        Update the list of ad-serving domains
:::  -s, setupLCD             Automatically configures the Pi to use the 2.8 LCD screen to display stats on it
:::  -c, chronometer          Calculates stats and displays to an LCD
:::  -h, help                 Show this help dialog
:::  -v, version              Show current versions
:::  -q, query                Query the adlists for a specific domain
:::  uninstall                Uninstall Pi-Hole from your system :(!
```

So, either user will use this, or better we get something similar to update with your fork and disable `pihole -up` and `pihole -ud`

Otherwise continuous mischief come about. (see update warnings: http://dietpi.com/phpbb/viewtopic.php?p=2791#p2791)

Fourdee, 2016-09-08:
> So, either user will use this, or better we get something similar to update with your fork and disable pihole -up and pihole -ud

Yep, sounds good :+1: 

Ok, so next PiHole release, i'll aim for the reinstall feature and see if we can pull it off. Basically, automatic Pihole upgrades during our updates.

> http://dietpi.com/phpbb/viewtopic.php?p=2791#p2804

I'll make a note a look into ```pihole -v``` and missing version numbers in the footer. But it will have to wait until the next PiHole release. Can do it all at the same time.

k-plan, 2016-09-08:
> Can do it all at the same time.

Totality agree! :100:  

Yes Sir, please do one thing after the other. :+1:

k-plan, 2016-09-09:
> ... missing version numbers in the footer.

https://github.com/pi-hole/AdminLTE/blob/master/footer.php#L7

[image]

😀  😀  😀  

But please don`t stress you. It must and should be "_all-for-fun_ "

Fourdee, 2016-10-26:
PiHole now uses official installer, making this as closed. https://github.com/Fourdee/DietPi/issues/583
Reinstall features of DietPi for all programs are still in devolpment.

closed 2016-10-26 by Fourdee