MichaIng/DietPi #505: /DietPi - empty directory

Bolo92, 2016-09-08:
Hi,

After reboot my raspberrypi with DietPi v129 or v130 won't get the IP address from DHCP.
When i plugin HDMI cable to raspberry i noticed that many errors shows up and one of these is that there is no files in /DietPi/ directory.

Recently i made a few changes to Linux, but i don't remember to remove some files or similar.
I changed /etc/fstab (add automatic mount via hubic) and i added /var/run to tmpfs. Which probably was not good idea. Many commands wasn't worked. Now i commet this line out.
`#tmpfs                  /var/run                tmpfs   defaults,noatime,nosuid,nodev,mode=0755,size=2m         0 0`

Now i changed DHCP to static IP address so raspberry after reboot has IP address and i can log in via SSH.
How can i fix this? Make a image of this SD Card, flash with new DietPi, run once and copy /DietPi directory to PC. Write image of previous DietPi install (without a /DietPi directory) and copy the files?
[image]

Sorry for my english.

Fourdee, 2016-09-09:
> If /DietPi directory is tmpfs which script copies the files to tmpfs after reboot?
I was thinking that i made a mistake with /etc/fstab and ovverride mount of /DietPi directory, but i found in repo oryginal file and compare with mine.

```/DietPi``` is mounted to RAMdisk during boot. It copies the files from ```/boot```. So looks like those files are missing, or, removal of mount point in ```/etc/fstab```

If ```/boot/dietpi/*``` has been deleted, I'd recommend a reinstall as all your DietPi settings and installed software list will be lost. 
You could try and replace the DietPi scripts in ```/boot```, but, you may still experience issues afterwards (eg: lost settings).

Bolo92, 2016-09-09:
Yeah, unfortunately /boot/dietpi/* is empty :/
Shit, i was thinking about backups ;)

> I'd recommend a reinstall as all your DietPi settings and installed software list will be lost. 

DietPi settings i can recreate. What about the installed software - you mean that i lost only the software list in DietPi-software script?

Fourdee, 2016-09-09:
@Bolo92 

> you mean that i lost only the software list in DietPi-software script?
No. ```/boot/dietpi/.installed``` file contains the status of what software is installed.

Important DietPi files:
- ```/boot/dietpi.txt``` | DietPi settings
- ```/boot/config.txt``` | RPi settings
- ```/boot/dietpi/.*``` | Unique files for your installation, that DietPi generates to store settings and various information.

I highly recommend you reinstall.

Bolo92, 2016-09-09:
@Fourdee Ok, i will reinstall DietPi after i make a backup of /etc and similar to have backups of my configuration files.
Thanks for your time and help ;)
Cheers

closed 2016-09-09 by Bolo92