MichaIng/DietPi #509: DietPi-Config | Move SD to USB

Zone22, 2016-09-12:
The proposal involves moving all data a USB medium with the exception of the boot loader on the SD card. This differs from the current system which involves Symbolic links to store application data on external media. With the proposal the entire OS would reside on the external medium. 
This has a few benefits for high read write applications that utilize databases. In addition it simplifies backup. One can simply clone the external hard drive and not have to worry about the SD card.

Fourdee, 2016-09-12:
@Zone22 

Good suggestion :+1: 
My issue is i have a C2 + EMMC that does 140MB/s write speed lol. So for me, the EMMC is a USB drive drive. Its also faster than a USB 2.0 port + drive (USB2.0 = 30-40MB/s R/W Max).

But yes, had a few people ask about this in the past, so i'll look into it when possible, will need different code for devices eg: RPi/Odroid.

Zone22, 2016-09-12:
> However, I honestly can't recommend EMMC enough as vast improvement over SDcards. Random R/W is also vastly superior. EMMC is a well worth purchase, and then some. 

I agree in fact I have it for my odroid xu4. Although it costs $22.95 +8 dollars shipping For the 8 GB version.

In the event of the features implemented depending on what the read/write performances for odroid xu4 USB 3.0. I might if it's worth switching EMMC to the C2.

That being said I think we both agree that there is some different use cases for the feature.

k-plan, 2016-09-12:
Hmm, nice idea. It will work with Odroid and RPi devices.

http://dietpi.com/phpbb/viewtopic.php?p=1720#p1720

http://dietpi.com/phpbb/viewtopic.php?p=1655#p1655

But you will get in trouble, if your device will start via:

 
```
boot.scr
script.bin
uInitrd
initrd.img-3.x.xx-sunxi
```

like Orange Pi (PC), NanoPi Neo, Orange Pi One and many other Allwinner SOC do (BananaPi, ...).

After you have change `setenv bootargs "root=/dev/mmcblk0p2  ... "` in `boot.cmd` , you will have do  recompile with:

`mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr`

and `script.bin`, `initrd.img-3.x.xx-sunxi` as well as `uInitrd` must support the new ` bootargs`  ...

k-plan, 2016-09-12:
@Fourdee 

> I honestly can't recommend EMMC enough as vast improvement over SDcards.

Hmm, but only the new v.5 EMMC with a Odroid C2.

But this in your C1+ and you get a nasty surprise ...

Most of my on-board-EMMC by manufacture are slowly but surely compared to e.g SanDisk Extreme microSDHC, Class 10, U3 card.

Fourdee, 2016-09-13:
For reference, John kindly shared his experience with going from SD to USBHDD: 
http://dietpi.com/phpbb/viewtopic.php?f=9&t=655&p=2875#p2842

k-plan, 2016-09-13:
@Fourdee 

Hmm, have you test [RPi-3 Boot from Mass Storage Device (MSD)](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/) ?

First, it is still BETA ( https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/ )
and only works with RPi 3 after patching One-Time Programmable (OTP) memory.
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

Cons:
- only woks with RPi 3
- OTP changes can not undo
- only a few  flash drive will work

> Some flash drives have a very specific protocol requirement that we don’t handle; as a result of this, we can’t talk to these drives correctly. An example of such a drive would be the Kingston Data Traveller 100 G3 32G.
These bugs exist due to the method used to develop the boot code and squeeze it into 32kB. It simply wasn’t possible to run comprehensive tests.

- problems with USB-HDD (spinning disk)

> Some flash drives power up too slowly. There are many spinning disk drives that don’t respond within the allotted two seconds. It’s possible to extend this timeout to five seconds, but there are devices that fail to respond within this period as well, such as the Verbatim PinStripe 64GB.

- it will increase boot time about 5 sec. 

https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/#comment-1261788

https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/#comment-1261786

Yes, it's possible to speed it up by putting a empty FAT formatted SD card in the slot, but so I can as well stay on this, which will work on every RPi: http://dietpi.com/phpbb/viewtopic.php?p=1655#p1655

After testing (_and fuc... up my RPi 3on-board  wifi by this way_) I will say   :-1:

Fourdee, 2016-09-13:
> Hmm, have you test RPi-3 Boot from Mass Storage Device (MSD) ?

I've not tested any methods yet lol :wink:, but will certinally try this one when I make a start on it.

I'am trying to find a schematic of the RPI's ARM, no luck so far. 
I'am curious to know if the SDIO is routed through the shared USB hub on the ARM. Ethernet goes through this already, so it would be interesting if SDIO doesn't, could imply SD is actually a "better" option than USB on the RPi.

k-plan, 2016-09-13:
> You can try other, lower overclock settings, as long as the number is an integer divisor of the core clock (default is 400 MHz on the Pi 3)

http://www.jeffgeerling.com/blog/2016/how-overclock-microsd-card-reader-on-raspberry-pi-3

Hmm, if I read this correct SD reader clock speed  is bundled at VideoCore IV subsystem (400 MHz)

If SD reader went through  usb hub, `dtoverlay=sdhost,overclock_50=100` will as well overclock usb host.

Zone22, 2016-10-24:
@Fourdee 
Would there be a possibility to add this to v135?

Fourdee, 2016-10-24:
@Zone22 

Not for v135, stacked up already. I'll add it to v136 and take a look after v135 is released.

Fourdee, 2016-11-07:
Ok, going with @k-plan 's implementation (http://dietpi.com/phpbb/viewtopic.php?f=9&t=396&p=1713#p1720) to cover RPi and Odroid devices.

There will be some limitations (subject to any changes):
- SD/EMMC will still be required for booting
- To allow for drive cloning, we wont use the UUID, so this might limit the system to 1 USB drive 
- ```/dev/sda1''' must be formatted to ext4 before hand (eg: user data must be located on SD)

Fourdee, 2016-11-12:
Testing:
- RPi
- Odroid (any, same code for all)

Fourdee, 2016-11-12:
Completed.

closed 2016-11-12 by Fourdee