MichaIng/DietPi #512: DietPi | Add option to format external USB drive as FAT32/NTFS

midwan, 2016-09-13:
This suggestion came from several AmiBerry users, but I believe it might interest others as well.
Instead of offering only ext4 as the filesystem for the external USB drive, an option to format it and use it as FAT32 or NTFS should also be available.

That way, users could easily also use it on a Windows machine.

If this causes some features not to work, e.g. symlinks inside the USB drive, we could have a relevant warning and maybe disable such applications from utilizing it? Not sure if there's any such scenario though.

Fourdee, 2016-09-14:
Ties in with: https://github.com/Fourdee/DietPi/issues/271

Fourdee, 2016-09-14:
> If this causes some features not to work, e.g. symlinks inside the USB drive, we could have a relevant warning and maybe disable such applications from utilizing it? Not sure if there's any such scenario though

Yep, symlinks is the main factor, ```dietpi-backup``` requires symlinks and will prevent use on non EXT drives. Aside from that, I think we should be fine.

Minor considerations:
NTFS = Higher CPU usage and slower performance
FAT32 = 4GB filesize limit | 2TB max volume

Fourdee, 2016-09-14:
[image]

Fourdee, 2016-09-14:
@midwan 
Completed. ```dietpi-drive_manager``` can now format ext4/ntfs/fat32.

I'll take a look at https://github.com/Fourdee/DietPi/issues/513. Most likely prompt user and run ```dietpi-drive_manager``` so they can setup drive.

closed 2016-09-14 by Fourdee

midwan, 2016-09-15:
@Fourdee you have a typo there in the screenshot, in case you didn't notice:
Recommended if you plan to use this drive **soley** on this system

should be **solely** I believe?