MichaIng/DietPi #513: DietPi + AmiBerry | Prompt users to format and use external USB device during first boot

midwan, 2016-09-13:
The prompt that exists in a standard DietPi installation, would also be useful in an AmiBerry autoboot installation.

Users should be prompted to format/use an external USB device if they wish, during installation.
Preferably, this should be combined with #512 also.

Fourdee, 2016-09-14:
Ties in with: https://github.com/Fourdee/DietPi/issues/271

Fourdee, 2016-09-14:
Ok, i've added a question prompt, which enables the user to run the new ```dietpi-drive_manager``` during installation. 
This program allows for mount/dismount/format etc.

note to self: create an online doc for this program, it is a bit more advanced than the previous system.

Fourdee, 2016-09-16:
Online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=673&p=2898#p2898

@midwan 
Users can follow 'How do i setup a dedicated drive for DietPi user data?' in the link above, when they are prompted during installation to setup a USB drive.

Fourdee, 2016-09-16:
@midwan 
Video example of what user will see (minus the ssh commands), and do:
https://youtu.be/U7YhVpAV3dI

Fourdee, 2016-09-18:
Tested with v130 + testing branch (v131). Prompt working fine. Marking as closed.

closed 2016-09-18 by Fourdee