MichaIng/DietPi #514: Image | OrangePi One

Fourdee, 2016-09-15:
@k-plan 
Think you might have something to do with this turning up in the post today? :+1: :tada:

k-plan, 2016-09-15:
@Fourdee 

Yes. Have a look in your survey directory.
It's not perfect, we have to work on it. (serial, leds, resolution, GPU support,....)

Fourdee, 2016-09-15:
@k-plan 

Found it, looks good, going through your notes now :+1: 

Do you have a cross compiler setup? I can give you a virtual machine with one setup for ARMbian and walk you through the steps to create a image?

k-plan, 2016-09-15:
@Fourdee 

no thanks. No clue for cross compiler.  :smiley: 

 virtual machine? -only vbox-

H3 almost identical. You have done all the work. Your uboot (uInitrd) and zImage  works.

[image]

Fourdee, 2016-09-15:
@k-plan 
> You have done all the work. Your uboot (uInitrd) and zImage works.

I did that before I read your notes, you did all the work also lol.

I've reverted the source code changes, want you to do them and get credit for it :+1: Do you know how to send a pull request with git? If not i'll walk you through it.

I'll update the ```/boot``` image with your changes also.

===

> no thanks. No clue for cross compiler. 

Honestly, its easy with ARMbian build tools, this is all you do on the VM:
```
apt-get -y install git
git clone https://github.com/igorpecovnik/lib --depth 1
cp lib/compile.sh .

#Modify compile.sh for
# - en_GB
sed -i 's/en_US/en_GB/' compile.sh

#Modify lib/configuration.sh
# + BOOTSIZE="64"
# bootsize exists in latest version, change value
# - packages
sed -i '/BOOTSIZE=/c\BOOTSIZE=64' lib/configuration.sh

#Build with ARMbian
./compile.sh

#Image then located in output/images/*.img
```

k-plan, 2016-09-15:
@Fourdee 

> Do you know how to send a pull request with git?

git? No Thanks. :smiley: 

Anti-coder don't need git. Sorry, I haven't got a clue.

Better you do this in your way. No skill in coding stuff or tools.
I only can "google" and "try and error".

> I did that before I read your notes, 

Sorry. But if I will told this to you before, you don't have a "Christmas surprise" now. :santa: :snowman: 

:wink:

Fourdee, 2016-09-15:
@k-plan 

> Anti-coder don't need git. Sorry, I haven't got a clue.

Yes you do, your notes are spot on.

VNC session and i'll get you setup? :)

k-plan, 2016-09-15:
@Fourdee 

> Yes you do, your notes are spot on.

Notes in German you can get at much as you want.

But coding and writing in English - no way. I am so sorry, take more time and knowledge I ever will have.

Yes, I am very unhappy that you will be a single-player, a one-man-show at the moment, but I can not help you out.
Ich tue was ich kann, aber leider ist im Moment nicht mehr drin. Wie sieht es mit Deutsch aus?

Fourdee, 2016-09-15:
@k-plan 

No worries. :+1: 
I know the language barrier is a pain, but you do great. 
If you change your mind, let me know and we'll sort it :)

Fourdee, 2016-09-15:
Ok notes:
:u5408: | Desktop (probably due to sunxi_no_mali_mem_reserve)
:u5408: | Kodi, we havn't implemented this for any H3 device yet.

k-plan, 2016-09-15:
@Fourdee 

>If you change your mind, let me know and we'll sort it :)

You will be the first how will know it. :100: 

Will you really want to support this Orange Pi One? It was a gift, without a commitment.

I personally think, we don't to get bogged down with to many devices and to many different software.
You can/will/should  make for your self some thoughts of generally target course.

(e.g. Platform for playing games, file server, productive tools, ...)

Fourdee, 2016-09-15:
@k-plan 

> It was a gift, without a commitment.

Yep, thank you :+1: 
I actually enjoy adding them into DietPi, gives me a chance to have some fun hehe

> I personally think, we don't to get bogged down with to many devices and to many different software.

Agree. 
I think I could improve the source to reduce the workload, especially on all the H3 devices we have.

We could also try having just one H3 image, then tell users how to update the script.bin for their device. Would save me having to make 5+ images for each device (very time consuming).

k-plan, 2016-09-15:
@Fourdee 

>  Kodi, we havn't implemented this for any H3 device yet.

Have tested Kodi (OpenELEC for OrangePi) and think H3 devices not a good choice.

Burn down your device and much problems at the moment.

http://wiki.orange314.com/Openelec_for_Orange_Pi

http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=648&extra=page%3D1

- cooling
- HDMI-CEC
- addons
- sound
- ....

But https://github.com/jernejsk/OpenELEC-OPi2/  helps me out a lot, to understand what's going on with H3 (.fex , boot.cmd, ..)

k-plan, 2016-09-15:
@Fourdee 

> We could also try having just one H3 image, then tell users how to update the script.bin for their device. Would save me having to make 5+ images for each device (very time consuming).

That's was the idea behind it. Reduce work. H3 image without `script.bin` but in this way, you get some new problems. (e.g. identify devices via `/etc/.dietpi_hw_model_identifier`)
Yes, possible to put it on `dietpi.txt` or 	somewhere else, but so you will get new trouble.

> Yep, thank you

No, not for this.

> I actually enjoy adding them into DietPi, gives me a chance to have some fun hehe

Yes that's the most important. FUN !  :smiley: 

If it make no fun any more, we should leave the project.  :+1:

k-plan, 2016-09-15:
Notes:

- Serial (unsure with parameter in `boot.cmd`)

`setenv bootargs "console=tty0 .....`

- Display output (DVI / HDMI)
Have you tested HDMI display with "my" `script.bin`? 
Can't test with original  `script.bin.bac`. Have only a DVI display for testing. Family LCD TV with HDMI input is forbidden! 

- LED:
Have set it now like : 

```
OrangePi-ONE:~# cat /etc/rc.local
#!/bin/bash
echo -e "$(cat /proc/uptime | awk '{print $1}') Seconds" > /var/log/boottime

#### switch on red LED for sd-card activity
echo 1 >/sys/class/leds/red_led/brightness
echo "mmc0" >/sys/class/leds/red_led/trigger

if (( $(cat /DietPi/dietpi/.install_stage) == 1 )); then

    /DietPi/dietpi/dietpi-services start

fi
/DietPi/dietpi/dietpi-banner 0
echo -e " Default Login:\n Username = root\n Password = dietpi\n"

exit 0
 
```

plus changes in `script.bin` to switch on red LED while booting up:

```
[leds_para]
leds_used = 1
green_led = port:PL10<1><default><default><0>
green_led_active_low = 1
red_led = port:PA15<1><default><default><1>
red_led_active_low = 0
```

But anyway.

Let first bring out Fourdee "DietPi v131" before we work on another new device.
Sunday will come now very quick.

Fourdee, 2016-09-15:
> That's was the idea behind it. Reduce work. H3 image without script.bin but in this way, you get some new problems. (e.g. identify devices via /etc/.dietpi_hw_model_identifier)
Yes, possible to put it on dietpi.txt or somewhere else, but so you will get new trouble.

Yep, good point. I think I can pull it off. Had a look at the code eariler, should be doable with 1 HW_MODEL for all H3 devices. Yes we lose the board name, but it would save me many hours of image creation. 
Just need to confirm with ARMbian if ```script.bin``` is the only unique item per all H3 boards. Really depends on that.

@k-plan 
PS. Whats your favorite alcoholic beverage? Do you like Aniseed, like black jacks (https://en.wikipedia.org/wiki/Black_Jack_(confectionery)?

notes:
h3 img:
- ```rm /etc/init.d/*```
- uboot: ```\usr\lib\linux-u-boot-orangepione_5.17_armhf\*.bin ??
- script.bin ```/boot/script.bin```

Fourdee, 2016-09-15:
@k-plan 
Asking the devs at ARMbian for hopefully some answers to my questions: http://forum.armbian.com/index.php/topic/2033-question-h3-unified-imaging/

k-plan, 2016-09-15:
@Fourdee 

> PS. Whats your favorite alcoholic beverage? 

Sorry I don't drink (and drive).

> Do you like Aniseed, like black jacks 

Black Jack?  Vampire Vape?  

Only know is this: http://vampirevape.de/product/black-jack-aroma-30ml/

[image]

:smiley:

Fourdee, 2016-09-16:
@k-plan 

>  but I think, we should: 1. - v131 / 2. - Orange Pi One only image / 3. - H3 universal Image for NanoPi Neo, Orange Pi (PC), Orange Pi One / ... and if this will work have a look for other H3 build boards, DietPi supported at moment.

Yep, good plan :+1: 
I'll release OPi One as a standard DietPi image with v131. Then after v131, will take a look unified H3 image.

> Black Jack? Vampire Vape?

You vape? what mg (eg: 18mg)?

Fourdee, 2016-09-16:
Note to self:
:u5408:  Run a final test on OPi One image + installation.

k-plan, 2016-09-16:
@Fourdee 

> http://dietpi.com/phpbb/viewtopic.php?f=8&t=676
> ==>>http://dietpi.com/download

Tomorrow?

-------------

OT @Fourdee 

> You vape? 

yes.

> what mg (eg: 18mg)?

:fearful:  Puh, no, it's too much. Sorry I'm a wuss.  :smiley: 
Mixed up my own base liquids with some flavour.

Fourdee, 2016-09-16:
> Tomorrow?

Most likely Sunday, depends how many tickets I nudge to v132 lol :)

> no, it's too much for me. Sorry I'm a wuss.

Not wuss :smile: . Wish i didn't have to vape if i'am honest (smoker 20 day) :disappointed:

k-plan, 2016-09-17:
> We could also try having just one H3 image, ....

Hmm, now I think it wasn't such a good idea.
Images are use `apt.armbian.com`. My test OPi-one based on your OPi (PC) image.
So I get a update by `apt-get upgrade` today.
Not really sure, if this will work correct under every circumstances.
(e.g. `zImage` and  `script.bin` are no link on FAT, `uInitrd`... ) 

Here is the output:

```
root@OrangePi-ONE:~# apt-get upgrade
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Paketaktualisierung (Upgrade) wird berechnet... Fertig
Die folgenden Pakete werden aktualisiert (Upgrade):
  linux-firmware-image-sun8i linux-image-sun8i linux-u-boot-orangepipc-default
3 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 12,4 MB an Archiven heruntergeladen werden.
Möchten Sie fortfahren? [J/n] n

root@OrangePi-ONE:~# uname -a
Linux OrangePi-ONE 3.4.112-sun8i #14 SMP PREEMPT Tue Jul 5 16:28:14 CEST 2016 armv7l GNU/Linux

root@OrangePi-ONE:~# cat /proc/version
Linux version 3.4.112-sun8i (root@pluton) (gcc version 5.3.1 20160413 (Ubuntu/Linaro 5.3.1-14ubuntu2) ) #14 SMP PREEMPT Tue Jul 5 16:28:14 CEST 2016

root@OrangePi-ONE:~# ls -lah /usr/lib/linux-u-boot-orangepipc_5.16_armhf/u-boot-sunxi-with-spl.bin
-rw-r--r-- 1 root root 377K Jul  5 17:48 /usr/lib/linux-u-boot-orangepipc_5.16_armhf/u-boot-sunxi-with-spl.bin

root@OrangePi-ONE:~# ls -lah /boot/
insgesamt 12M
drwxr-xr-x  3 root root  16K Jan  1  1970 .
drwxr-xr-x 21 root root 4,0K Sep 13 10:46 ..
-rwxr-xr-x  1 root root 3,3K Sep 13 10:55 boot.cmd
-rwxr-xr-x  1 root root 2,4K Sep 11 19:55 boot.cmd.bak
-rwxr-xr-x  1 root root 3,4K Sep 13 10:55 boot.scr
-rwxr-xr-x  1 root root 2,4K Sep 11 19:55 boot.scr.bak
-rwxr-xr-x  1 root root  93K Sep 10 20:12 config-3.4.112-sun8i
-rwxr-xr-x  1 root root 2,0K Sep 13 10:57 config.txt
drwxr-xr-x  5 root root 4,0K Sep 13 10:57 dietpi
-rwxr-xr-x  1 root root 7,3K Sep 13 10:57 dietpi.txt
-rwxr-xr-x  1 root root 2,4M Aug  6 17:51 initrd.img-3.4.112-sun8i
-rwxr-xr-x  1 root root 6,3K Sep 11 17:25 README.md
-rwxr-xr-x  1 root root  36K Sep 12 10:00 script.bin
-rwxr-xr-x  1 root root  36K Sep 10 20:13 script.bin.bak
-rwxr-xr-x  1 root root  18K Sep 12 10:00 script.fex
-rwxr-xr-x  1 root root  18K Sep 10 20:13 script.fex.bak
-rwxr-xr-x  1 root root 2,1M Jul  5 16:28 System.map-3.4.112-sun8i
-rwxr-xr-x  1 root root 2,4M Aug  6 17:25 uInitrd
-rwxr-xr-x  1 root root 4,8M Jul  5 16:28 zImage

root@OrangePi-ONE:~# apt-get upgrade
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Paketaktualisierung (Upgrade) wird berechnet... Fertig
Die folgenden Pakete werden aktualisiert (Upgrade):
  linux-firmware-image-sun8i linux-image-sun8i linux-u-boot-orangepipc-default
3 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 12,4 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 934 kB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] j
Holen: 1 http://apt.armbian.com/ jessie/main linux-firmware-image-sun8i armhf 5.20 [113 kB]
Holen: 2 http://apt.armbian.com/ jessie/main linux-image-sun8i armhf 5.20 [12,1 MB]
Holen: 3 http://apt.armbian.com/ jessie/main linux-u-boot-orangepipc-default armhf 5.20 [156 kB]
Es wurden 12,4 MB in 3 s geholt (3.922 kB/s).
(Lese Datenbank ... 21869 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../linux-firmware-image-sun8i_5.20_armhf.deb ...
Entpacken von linux-firmware-image-sun8i (5.20) über (5.16) ...
Vorbereitung zum Entpacken von .../linux-image-sun8i_5.20_armhf.deb ...
Entpacken von linux-image-sun8i (5.20) über (5.16) ...
Vorbereitung zum Entpacken von .../linux-u-boot-orangepipc-default_5.20_armhf.deb ...
Entpacken von linux-u-boot-orangepipc-default (5.20) über (5.16) ...
linux-firmware-image-sun8i (5.20) wird eingerichtet ...
linux-image-sun8i (5.20) wird eingerichtet ...
update-initramfs: Generating /boot/initrd.img-3.4.112-sun8i
linux-u-boot-orangepipc-default (5.20) wird eingerichtet ...

root@OrangePi-ONE:~# ls -lah /usr/lib/linux-u-boot-orangepipc_5.20_armhf/u-boot-sunxi-with-spl.bin
-rw-r--r-- 1 root root 412K Sep 15 07:28 /usr/lib/linux-u-boot-orangepipc_5.20_armhf/u-boot-sunxi-with-spl.bin

root@OrangePi-ONE:~# ls -lah /boot/
insgesamt 12M
drwxr-xr-x  3 root root  16K Sep 17 10:16 .
drwxr-xr-x 21 root root 4,0K Sep 13 10:46 ..
-rwxr-xr-x  1 root root 3,3K Sep 13 10:55 boot.cmd
-rwxr-xr-x  1 root root 2,4K Sep 11 19:55 boot.cmd.bak
-rwxr-xr-x  1 root root 3,4K Sep 13 10:55 boot.scr
-rwxr-xr-x  1 root root 2,4K Sep 11 19:55 boot.scr.bak
-rwxr-xr-x  1 root root  93K Sep 14 20:29 config-3.4.112-sun8i
drwxr-xr-x  5 root root 4,0K Sep 13 10:57 dietpi
-rwxr-xr-x  1 root root 7,3K Sep 13 10:57 dietpi.txt
-rwxr-xr-x  1 root root 2,4M Sep 17 10:16 initrd.img-3.4.112-sun8i
-rwxr-xr-x  1 root root 6,3K Sep 11 17:25 README.md
-rwxr-xr-x  1 root root  36K Sep 12 10:00 script.bin
-rwxr-xr-x  1 root root  36K Sep 10 20:13 script.bin.bak
-rwxr-xr-x  1 root root  18K Sep 12 10:00 script.fex
-rwxr-xr-x  1 root root  18K Sep 10 20:13 script.fex.bak
-rwxr-xr-x  1 root root 2,1M Sep 14 20:29 System.map-3.4.112-sun8i
-rwxr-xr-x  1 root root 2,4M Aug  6 17:25 uInitrd
-rwxr-xr-x  1 root root 4,8M Sep 14 20:29 zImage

reboot

root@OrangePi-ONE:~# uname -a
Linux OrangePi-ONE 3.4.112-sun8i #14 SMP PREEMPT Wed Sep 14 20:29:31 CEST 2016 armv7l GNU/Linux
```

Fourdee, 2016-09-19:
https://github.com/Fourdee/DietPi/issues/520

closed 2016-09-19 by Fourdee

k-plan, 2016-09-20:
@Fourdee 

OrangePi One is missing in: [README.md ](https://github.com/Fourdee/DietPi/blob/master/README.md)

Fourdee, 2016-09-20:
@k-plan 

> OrangePi One is missing in: README.md

Sorted, good spot 👍