MichaIng/DietPi #515: DietPi | WiFi connection with static IP address fail

k-plan, 2016-09-17:
> http://dietpi.com/phpbb/viewtopic.php?p=2929#p2929

Fourdee, 2016-09-17:
> https://github.com/Fourdee/DietPi/issues/474#issuecomment-244546570
> https://github.com/Fourdee/DietPi/commit/007b8ca96037b519e02cee5b1b5eb62a1afb85c5

New image should solve this. But i'll test.

k-plan, 2016-09-17:
Ready to go with my RPi B+, RPi Zero or RPi 3 for your new image. 😄 

Please tell which device to use.  👍

Fourdee, 2016-09-17:
@k-plan 

Just did a test on RPi3, DHCP automation is fine, but static still fails to connect, same as reported 0.0.0.0 etc.
Weird.

I'll need to look into it, hoping it might be a kernel/firmware issue lol. 4.4.19

@k-plan 
If you get a chance, would you be able to test RPi 1/2 with USB wifi dongle on v130. Not the automation side, just static IP on a completed installation?

k-plan, 2016-09-17:
> If you get a chance, would you be able to test RPi 1/2 with USB wifi dongle on v130. Not the automation side, just static IP on a completed installation?

Static IP on Wifi dongle by RPI B+ did not work. (switching from DHCP)
Staitic IP on wire connection as well as DHCP wired work. (Both active WiFi=DHCP / Wired=static or DHCP)

But I will start from the scratch .... with v130 without any mod and wired connection.

Fourdee, 2016-09-17:
@k-plan 
No rush, will be tomorrow for me.
Just tried 4.4.21 on RPi 3. Same issue. 

```
NETLINK: Error: Invalid argument
```

Fourdee, 2016-09-17:
@k-plan 

Think i got it.

```
apt-get purge ifmetric
```

Remove ```metric 0``` and ```metric 1``` from ```/etc/network/interfaces```

Now it works.

k-plan, 2016-09-17:
v.130 on RPi B+ wired connection (DHCP) not touch dietpi.txt:

```
Welcome to DietPi-Software

 [Info] Testing connection to http://mirrordirector.raspbian.org/raspbian/
 [Info] Max duration of 20 seconds, please wait...
 [Ok] Connection test | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates
 Please wait...

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100     5  100     5    0     0     26      0 --:--:-- --:--:-- --:--:--    26

 [Ok] No updates required, your DietPi installation is up to date.

 [Info] Current Version : 130
 [Info] Server Version  : 130

 DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: preferipversion (auto)
 Please wait...

 [Ok] preferipversion auto | Completed
 [Info] Testing connection to http://mirrordirector.raspbian.org/raspbian/
 [Info] Max duration of 20 seconds, please wait...
 [Ok] Connection test | Completed

 DietPi
─────────────────────────────────────────────────────
 Mode: Updating user data location
 Please wait...

 [Info] /mnt/dietpi_userdata is already symlinked to target directory.

 [Ok] Completed

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT
 Please wait...

Get:1 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]
Get:2 http://mirrordirector.raspbian.org jessie/main armhf Packages [8,981 kB]
Get:3 http://archive.raspberrypi.org jessie InRelease [13.2 kB]
Get:4 http://archive.raspberrypi.org jessie/main armhf Packages [146 kB]
Get:5 http://archive.raspberrypi.org jessie/ui armhf Packages [8,966 B]
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Get:6 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [37.5 kB]
Get:7 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [70.3 kB]
Get:8 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1,356 B]
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Fetched 9,274 kB in 29s (314 kB/s)
Reading package lists... Done
 [Info] Running apt simulation to check for errors, please wait...
 [Ok] Apt simulation | Completed

 ───────────────────────────────────────
 DietPi     | 19:51 | Sat 17/09/16
 ───────────────────────────────────────
 V130       | RPi B+ (armv6l)
 ───────────────────────────────────────
 IP Address | 172.16.10.89
 ───────────────────────────────────────

 Welcome to DietPi-Software

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT
 Please wait...

Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  raspberrypi-sys-mods
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 6,312 B of archives.
After this operation, 22.5 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian/ jessie/main raspberrypi-sys-mods all 20160916 [6,312 B]
Fetched 6,312 B in 0s (15.9 kB/s)
(Reading database ... 19454 files and directories currently installed.)
Preparing to unpack .../raspberrypi-sys-mods_20160916_all.deb ...
Removing old /etc/sysctl.d/98-rpi.conf...
Adjusting vm.swappiness and vm.min_free_kbytes...
Unpacking raspberrypi-sys-mods (20160916) over (20160426) ...
Setting up raspberrypi-sys-mods (20160916) ...
Removing old sysctl tweaks if present...

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software
 Please wait...

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
 Please wait...

 [Ok] cron

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
 Please wait...

 [Ok] cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Optimize and configure software
 Please wait...

 [Info] Applying DietPi optimizations and configurations for RPi B+ (armv6l)
 [Info] Please wait...

 #               processing | D I E T  P I             #

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
 Please wait...

 [Ok] cron

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: disable
 Please wait...

 [Ok] cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed
 Please wait...

 The system will now reboot.
 This completes the DietPi-Software installation.

root@DietPi:~# processing | D I E T  P I #  processing
```

work!  👍

k-plan, 2016-09-17:
@Fourdee 

> Think i got it.

> apt-get purge ifmetric
Remove metric 0 and metric 1 from /etc/network/interfaces

> Now it works, on my RPi 3.

oh, oh ... but no metric no https://github.com/Fourdee/DietPi/issues/265

Fourdee, 2016-09-17:
@k-plan 

> oh, oh ... but no metric no #265

Yep 😞. I need to re-run the tests again to confirm. But looks like we might need to find another alternative to ```ifmetric```?

Now off to play Descent 2 (was Tron 2.0 last month) for 30 minutes before son goes bed, woop woop 😃

k-plan, 2016-09-17:
@Fourdee 

> ifmetric - An IPv4 route metrics manipulation tool

http://0pointer.de/lennart/projects/ifmetric/ifmetric.8.xml

http://metadata.ftp-master.debian.org/changelogs//main/i/ifmetric/ifmetric_0.3-3_changelog

> But looks like we might need to find another alternative to ifmetric?

Hmm, no. It's time to use the "new" _primary network configuration tools_   😱 

```
root@DietPi:~# ip -V
ip utility, iproute2-ss140804
```

iproute2:  https://wiki.linuxfoundation.org/networking/iproute2

> Most network configuration manuals still refer to **_ifconfig_** and **_route_** as the **_primary network configuration tools_**, but ifconfig is known to behave inadequately in modern network environments. **_They should be deprecated_**, but most distros still include them. Most network configuration systems make use of ifconfig and thus provide a limited feature set. The /etc/net project aims to support most modern network technologies, as it doesn't use ifconfig and allows a system administrator to make use of all iproute2 features, including traffic control.

http://www.policyrouting.org/iproute2.doc.html

Fourdee, 2016-09-18:
@k-plan 

For now, we will remove metric and ifmetric during patch. I'll reopen https://github.com/Fourdee/DietPi/issues/265 so we can "start again" lol 👍

Fourdee, 2016-09-18:
Nearly forgot.

Fix automation needs new image: 🈴

k-plan, 2016-09-18:
@Fourdee 

Testing:

-  with two interfaces with static ip addresses (same broadcast domain)

- `ifmetric` and `netplug` installed

`/etc/network/interfaces`

```
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet static
## metric 0
address 192.168.0.100
netmask 255.255.255.0
network 192.168.0.0
broadcast 192.168.0.255
dns-nameservers 8.8.8.8 8.8.4.4
post-up ip route add default via 192.168.0.1 dev eth0 metric 0
pre-down ip route del default via 192.168.0.1 dev eth0

# Wifi
allow-hotplug wlan0
iface wlan0 inet static
## metric 1
address 192.168.0.200
netmask 255.255.255.0
network 192.168.0.0
broadcast 192.168.0.255
wireless-essid FuzonWifi
wireless-key abcde12345
wireless-mode Managed
wireless-power off
wpa-ssid FuzonWifi
wpa-psk abcde12345
#dns-nameservers 8.8.8.8 8.8.4.4
post-up ip route add default via 192.168.0.1 dev wlan0 metric 1
pre-down ip route del default via 192.168.0.1 dev wlan0
```

have a look at ` ip route show`

test with `ping dietpi.com` and unplug/plug in ethernet cable

k-plan, 2016-09-18:
@Fourdee

Testing:

- with one interfaces with static ip addresses, second with dhcp (same broadcast domain)

- ifmetric and netplug installed

`/etc/network/interfaces`

```
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet static
## metric 0
address 192.168.0.100
netmask 255.255.255.0
network 192.168.0.0
broadcast 192.168.0.255
dns-nameservers 8.8.8.8 8.8.4.4
post-up ip route add default via 192.168.0.1 dev eth0 metric 0
pre-down ip route del default via 192.168.0.1 dev eth0

# Wifi
allow-hotplug wlan0
iface wlan0 inet dhcp
## metric 1
address 192.168.0.200
netmask 255.255.255.0
network 192.168.0.0
broadcast 192.168.0.255
wireless-essid FuzonWifi
wireless-key abcde12345
wireless-mode Managed
wireless-power off
wpa-ssid FuzonWifi
wpa-psk abcde12345
#dns-nameservers 8.8.8.8 8.8.4.4
post-up ip route add default dev wlan0 metric 1
pre-down ip route del default dev wlan0

```
- have a look at ip route show 👍

- test with ping dietpi.com and unplug/plug in ethernet cable 👍

- output of dmesg 👍

k-plan, 2016-09-18:
@Fourdee 

Testing:

- with two interfaces with dhcp (same broadcast domain)  👎 

- testing with iptable2 Policy Routing 👎  
Possible, it's a problem with my skill. 😞  

Yes, revert to v115 practice of either wired or wifi 😢

Fourdee, 2016-09-18:
@k-plan 

v131 image. Static IP WiFi automation should now work correctly: 
http://dietpi.com/downloads/testing/DietPi_RPi-armv6-(Jessie).7z

k-plan, 2016-09-18:
@Fourdee 

> v131 image. Static IP WiFi automation should now work correctly:

Tested on RPi B+ with wifi adapter and non wired connection.

- static wifi ip, Ethernet disable, full manually installation => work 👍 

- static wifi ip, Ethernet disable, automated installation => work 👍 

after reboot it is a DietPi v130

[image]

Hint: are you living now in USA?

[image]

Will you give it to "tohax" for testing? I think it will work.
But i have only tested installation with static ip on wifi, nothing more.

Fourdee, 2016-09-18:
@k-plan 

> Hint: are you living now in USA?

I wish, furthest I'am going is Skegness lol (daytona USA arcades!): http://www.skegness.net/

I'll redo the image with those removed. Not sure how they got there. Good spot 👍 

> Will you give it to "tohax" for testing? I think it will work.

I tested on RPi3 with onboard WiFi and static, worked for me. Was hoping to get v131 out today, but its most likely Monday/Tuesday now.
Pretty confidant we have resolved the issue. I'll get v131 wrapped up 1st, then we can let him know of the new image?

k-plan, 2016-09-18:
@Fourdee 

I wonder, because noting set in `dietpi.txt`

```
#WiFi country code. 2 character value (eg GB US DE JP): https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
wifi_country_code=

```

.. and it is set to `GB`

```
[   12.634275] cfg80211: Regulatory domain changed to country: GB
[   12.634320] cfg80211:  DFS Master region: ETSI
[   12.634334] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   12.634356] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   12.634375] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   12.634394] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   12.634411] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
[   12.634427] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
```

Fourdee, 2016-09-18:
@k-plan 

https://github.com/Fourdee/DietPi/blob/master/dietpi/finalise#L110-L111

Should be 00, i'll change that.

tohax, 2016-09-18:
v131 image did not solve problem for me. Connection test failed again, but now there is a proper ip address in dietpi banner, there was emty field before.

Fourdee, 2016-09-18:
@tohax

> Connection test failed again

Raspbian mirror was offline for me. I had to change to a static mirror.

Try:
```
cat << _EOF_ > /etc/apt/sources.list
deb http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian jessie main contrib non-free rpi
_EOF_
```

tohax, 2016-09-18:
i'll try and report tomorrow. need to sleep.

tohax, 2016-09-19:
it's alive!!! Everything works fine. Thank you.

k-plan, 2016-09-19:
@Fourdee 

> Should be 00, i'll change that.

Hmm, I would say, locales, keyboard, time zone, country code =>> GB 

what's consistent, because Fourdee + DietPi = 🇬🇧 

😄

Fourdee, 2016-09-19:
@k-plan 

> Hmm, I would say, locales, keyboard, time zone, country code =>> GB

Would be nice for me lol. 
But, I can see people "kicking off" with the legality of a GB wifi region in a country that it would "break the law". So, must really set 00 for WiFi.

Fourdee, 2016-09-19:
@k-plan 

> But, I can see people "kicking off" with the legality of a GB wifi region, in a country that its against the law to use. So, must really set 00 for WiFi.

Although, the official Raspbian image has this set to GB by default. I'am game. We'll set to GB by default on all DietPi images.

Fourdee, 2016-09-19:
@tohax 

> it's alive!!! Everything works fine. Thank you.

Excellent thanks for letting us know 👍 

Marking this as closed.

closed 2016-09-19 by Fourdee