MichaIng/DietPi #516: Changing ssh server software

oshank, 2016-09-19:
I have come across an issues of sorts with the removal of dropbear. Currently I am running a Pi 3 with v129 and a constant VPN connection. I was planing on using ssh keys to connect to the Pi but was having issues with setting the public key on the Pi. So, I decided to replace dropbear with OpenSSH using the dietpi-software interface and came across an interesting error.

So I found out prior to tonight that I couldn't use apt-get as the mirror list denies my connection through the vpn and thus not being able to do any updates. When I was denied for trying to download/install the openssh packages the dietpi script continued to remove dropbear all together. And now I am unable to access my pi over ssh.

Just thought I should let someone know that the script is still removing dropbear even though installation failed.

Fourdee, 2016-09-19:
@oshank 

Ah I see, so the Dropbear program was removed, regardless of the failed installation of OpenSSH server?

I'll look into this, prevent it uninstalling on a failed install. Thanks for letting us know 👍

oshank, 2016-09-19:
Yep! That is exactly what happened!

Fourdee, 2016-09-23:
@oshank 

```dietpi-software``` will now exit on a failed apt installation, rather than continue. This will prevent any further changes (including uninstalls) from occurring.

closed 2016-09-23 by Fourdee