MichaIng/DietPi #523: DietPi-Software | UrBackup (err_cannot_create_subdir). Permission denied (code: 13)

Zone22, 2016-09-21:
Upon following [instructions](http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=65#p65) When setting the file path to "/mnt/dietpi_userdata/urbackup" or "/mnt/usb_1/urbackup". 

**Error as detailed in web interface.**
The directory where UrBackup will save backups is inaccessible. Please fix that by modifying this folder in 'Settings' or by giving UrBackup rights to access this directory. (err_cannot_create_subdir). Permission denied (code: 13)

Fourdee, 2016-09-21:
@Zone22 

Great report 👍 

We actually set permissions for the data store directory during installation:
```
chown -R urbackup:urbackup "$FP_DIETPI_USERDATA_DIRECTORY"/urbackup
```

Most likely, we need to apply to the physical location ```/mnt/usb_1``` aswell as the symlink ```/mnt/dietpi_userdata```. I'll run a test to check.

Fourdee, 2016-09-21:
@Zone22 

Tested on a fresh installation, Odroid C2. Permissions are set correctly during installation:
```
root@DietPi:~# ls -lha /mnt/dietpi_userdata/
total 28K
drwxr-xr-x  4 root     root     4.0K Sep 21 16:21 .
drwxr-xr-x 11 root     root     4.0K Sep 21 16:08 ..
drwx------  2 root     root      16K Sep 21 13:31 lost+found
drwxr-xr-x  3 urbackup urbackup 4.0K Sep 21 16:21 urbackup
root@DietPi:~# ls -lha /mnt/usb_1/
total 28K
drwxr-xr-x  4 root     root     4.0K Sep 21 16:21 .
drwxr-xr-x 11 root     root     4.0K Sep 21 16:08 ..
drwx------  2 root     root      16K Sep 21 13:31 lost+found
drwxr-xr-x  3 urbackup urbackup 4.0K Sep 21 16:21 urbackup
root@DietPi:~# ls -lha /mnt/dietpi_userdata
lrwxrwxrwx 1 root root 10 Sep 21 16:08 /mnt/dietpi_userdata -> /mnt/usb_1

```

Which:
- device?
- format type on usb drive?

Zone22, 2016-09-21:
dietpi-drive_manager was used.
Odroid C2
btrfs

Zone22, 2016-09-21:
@Fourdee 
I'm experiencing issues with owncloud as well with file permissions.

Fourdee, 2016-09-22:
@Zone22 

Ok, did the following test on fresh installation:
- Format drive to btrfs (userdata automatically moved during 1st run to ```/mnt/usb_1```)
- Install Owncloud and UrBackup

Permissions set fine for me: 
```
root@DietPi:~# ls -lha /mnt/dietpi_userdata/
total 20K
drwxr-xr-x  1 root     root       62 Sep 22 14:17 .
drwxr-xr-x 11 root     root     4.0K Sep 22 14:02 ..
drwx------  1 root     root        0 Sep 22 14:02 lost+found
drwxrwx---  1 www-data www-data    8 Sep 22 14:17 owncloud_data
drwxr-xr-x  1 urbackup urbackup   36 Sep 22 14:02 urbackup
```

@Zone22 
With owncloud, did you use ```/mnt/dietpi_userdata/owncloud_data``` during the wizard, as per http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p47?

Again urbackup looks fine.

Is there anything else you may have done (eg: move user data from drives etc), that you can remember? Cant seem to replicate this issue.

Zone22, 2016-09-23:
I apologize for this being so irksome.

I've also tested this repeatedly with a fresh install.  I'm at a bit of a loss.

Reading your above post this stuck out to me in bold.
'''Format drive to btrfs (userdata **automatically** moved during **1st run to** /mnt/usb_1)'''

I didn't realize there was an option for the first run experience to format btrfs/GPT  and move user data automatically.

For experimentation sake I've tried the below processes
1. First boot
2. Update Dietpi
3. Run through start up wizard 
4. Install Owncloud and UrBackup
5. dietpi-drive_manager to GPT and btrfs
6. move userdata via Dietpi on hard drive - because dietpi-drive_manager requests you to do so if 
**or**
1. First boot
2. Update Dietpi
3. Run through start up wizard - setting up hard drive default settings
4. Install Owncloud and UrBackup
5.  Move userdata via Dietpi folders off 
6.  Dietpi-drive_manager to GPT and btrfs
7.  Move userdata via Dietpi on external hard drive 

So permissions must be being reset through "Moving userdata via Dietpi"..??
If we still can't replicate the issue on based on what's above I'll create a video capturing my procedures.

Fourdee, 2016-09-23:
@Zone22 

> I didn't realize there was an option for the first run experience to format btrfs/GPT and move user data automatically. 

We have a var in ```dietpi.txt```: ```dietpi_userdata_basedirectory=Auto```https://github.com/Fourdee/DietPi/blob/master/dietpi.txt#L171-L176

In DietPi-Software:
If the var is ```Auto``` during 1st run setup, and, a USB drive is attached and mounted (regardless of filesystem type), DietPi will automatically setup your user data for ```/mnt/usb_1```. If not, it stays on SD. 

> So permissions must be being reset through "Moving userdata via Dietpi"..??

If the drive source is FAT, then there are no permissions to copy over. So, for example, transferring from a FAT to NTFS/EXT4 may cause the issue. Not sure about HFS+.

> If we still can't replicate the issue on based on what's above I'll create a video capturing my procedures.

I'll try both your methods listed, see if I can get it to break 👍

Fourdee, 2016-09-23:
Ok, found it:
- If the drive is FAT or HFS+. Urbackup throws the permission error.
- Cant set permissions on FAT or HFS+ drives.

So we either need to 
- disable FAT/HFS+ formatting and support
- find a way to mount FAT/HFS+ drives that ignores permissions and allows all access.

When moving user data from a FAT/HFS+ drive:
- Set permissions to ```nobody:nogroup``` (as they will all be reset to root:root)

Fourdee, 2016-09-23:
To cover all filesystems:
- grant access to all users on all USB drives in ```/etc/fstab``` by default.

To cover moving of userdata back to SD:
- Attempt to copy existing permissions if the source supports it.
- Add a function into dietpi-software that reapplies all known permissions.

Fourdee, 2016-09-23:
To cover all filesystems:
- USB drives will be mounted with no permissions in ```/etc/fstab``` by default.

To cover moving of user data back to SD:
- After the data transfer is completed, ```dietpi-software setpermissions``` will be run which re-applies permissions for known software.

I've tested with FAT/BTRFS/EXT4 formats and moving userdata across all of them and back to SD. All working fine.

Marking as closed.

@Zone22 
You can test the changes with the testing branch: https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md
Please reopen if required.

Fourdee, 2016-09-23:
EXT4 mounts fail if ```umask=0000``` is set.

Fourdee, 2016-09-24:
EXT4 source, copy existing user/chmod permissions:
- 🈴 ntfs. No permissions copied. 
- 🈺 btrfs. Yes, but requires ```cp -p``` 
- 🈴 vfat.
- 🈺 hfs+.  Yes, but requires ```cp -p```

Fourdee, 2016-09-24:
Reworked moving userdata:
- DietPi will attempt to preserve existing file permissions, if supported by target filesystem.
- At the end of the transfer, DietPi-Software will apply user permissions for known software in the database.

To Do:
- Work out a way of mounting FAT32 and NTFS with no permissions.

closed 2016-09-24 by Fourdee