MichaIng/DietPi #528: Bluetooth support on Pine64

pfeerick, 2016-09-25:
[I'll add the link here to the pine64 forum post once it is available]

When trying to enable on the Bluetooth on the DietPi distro using the same steps that worked for the Debian Jessie distro, I had no luck. You can see the [steps needed here (from my notes)](https://www.evernote.com/shard/s1/sh/cb4b9f07-e182-4abf-8ee3-5db105192626/3a8d6c901cc8acab). 

I have come to believe after examining the `start_bt.sh` script further that that it may not be working as the kernel config flag "CONFIG_SERIAL_8250" is not set, and is apparently needed for this to work. 

Firmware binaries, start_bt.sh, etc were sourced from this repo: https://github.com/lwfinger/rtl8723bs_bt

MichaIng, 2018-03-01:
@pfeerick 
Is the issue and procedure still current?

@Fourdee 
As far as I can see, we do not support bluetooth on Pine64. If there is still no native/repo support for this, it is a bid more to do then for the other devices, but maybe worth trying, setting kernel config flag included, if it (still) works?

Fourdee, 2018-03-02:
@MichaIng 

I'll run a test today.

MichaIng, 2018-04-06:
@Fourdee 
The link provided by @pfeerick includes some other steps after `start_bt.sh`: https://www.evernote.com/shard/s1/sh/cb4b9f07-e182-4abf-8ee3-5db105192626/3a8d6c901cc8acab
- "...it configures the BT module and makes it available for use": `./rtk_hciattach -n -s 115200 /dev/ttyS1 rtk_h5 &` (needs to run in background, hmm serial tty needs to be active, then (?)...)
- "Remove soft block, scan for devices": `rfkill list`, then `rfkill unblock <bt_deviceID>` in case

MichaIng, 2018-08-27:
@Fourdee 
Maybe we can pick this up again, if you find time to test with the steps provided in the link above?
Sadly I cannot test all the WiFi and BT stuff, however about the other WiFi (multiple SSID) topic, I can provide the initial dietpi-config/set_hardware menus/additions, when solution is chosen 😃.

MichaIng, 2019-02-27:
Pine A64 image has been updated and is now based on ARMbian with different (newer) kernel. Perhaps the above works now.

MichaIng, 2020-02-04:
New Buster-based image available for testing: https://dietpi.com/downloads/testing/DietPi_PineA64-ARMv8-Buster.7z
Since there have been vast changes in kernel, system etc. this should or needs to be simply retested, following the linked guide and drivers. If it hangs even after error-free build and verifying that used UART device is not in other use and not blocked and listens to the correct hardware connectors, it should be reported to the drivers repo. I mark this as closed, as it is hence no DietPi-specific issue.

closed 2020-02-04 by MichaIng