MichaIng/DietPi #531: DietPi-First Run Setup | USB drive fails to format.

Zone22, 2016-09-26:
Tested on a fresh installation, Odroid C2. with v132

Previously Formatted Hard Drive:
GTP 
btrfs

 ```
--------------------------------
 D I E T - P I
 External Drive Setup
 --------------------------------
 Drive ready to be formatted to EXT4
```

 WARNING: This will delete all data on the USB drive

 Do you wish to Format this drive? [y/N] y
1+0 records in
1+0 records out
5242880 bytes (5.2 MB) copied, 0.157438 s, 33.3 MB/s
Error: partition length of 7814033408 sectors exceeds the msdos-partition-table-imposed maximum
of 4294967295
umount: /dev/sda1: mountpoint not found
mke2fs 1.42.12 (29-Aug-2014)
The file /dev/sda1 does not exist and no size was specified.
resize2fs 1.42.12 (29-Aug-2014)
open: No such file or directory while opening /dev/sda1

```
Happens during _first run_ when setting up the hard drive.

Fourdee, 2016-09-27:
@Zone22 
Thanks, will take a look 👍

Fourdee, 2016-09-27:
> Error: partition length of 7814033408 sectors exceeds the msdos-partition-table-imposed maximum
of 4294967295

Will set GPT partition format as default to support larger drives.

Zone22, 2016-09-27:
Would it be possible to define the filing system and partition format in dietpi.txt for first boot?
A check would be needed if the filing system and partition format already exists. In that case it would skip formatting.

Fourdee, 2016-09-28:
@Zone22 

Good suggestion 👍  Please create a separate Git ticket for this.

closed 2016-09-28 by Fourdee