MichaIng/DietPi #533: Nextcloud command line installation fails when database root password changed.

sootstone, 2016-09-27:
As described here [http://dietpi.com/phpbb/viewtopic.php?f=12&t=525&p=3045#p3045](url)

A solution might be to have a window after choosing to install 'Nextcloud' that gave you the option to change the default values for installing Nextcloud via the command line if required, if this is possible.

Something along these lines:

```
NEXTCLOUD_DATABASE_USER="root"
NEXTCLOUD_DATABASE_PASS="dietpi"
NEXTCLOUD_ADMIN_USER="admin"
NEXTCLOUD_ADMIN_PASS="dietpi"

sudo -u www-data php occ maintenance:install --no-interaction --database "mysql" --database-name "nextcloud" --database-user $NEXTCLOUD_DATABASE_USER --database-pass $NEXTCLOUD_DATABASE_PASS --admin-user $NEXTCLOUD_ADMIN_USER --admin-pass $NEXTCLOUD_ADMIN_PASS --data-dir "$FP_DIETPI_USERDATA_DIRECTORY/nextcloud_data"
```

Zone22, 2016-09-27:
I think it would be prudent to allow users to put their preference for root passwords either for the system And/or MySQL database per database in dietpi.txt. This would be used during the first run set up to automatically install software along with the desired with the desired route and database passwords.

sootstone, 2016-09-27:
@Zone22

Agree, that sounds good.

Zone22, 2016-09-28:
> Yep, great suggestion, I think we could do a global_password=xxxx in dietpi.txt and apply to all software and root login.
@Fourdee 
Would it be too much trouble to To have a separate password for the root and databases.
root_password=xxxx in dietpi.txt apply root login.
global_database_password=xxxx in dietpi.txt apply to all software.

Fourdee, 2016-09-28:
@Zone22 

> root_password=xxxx in dietpi.txt apply root login.
global_database_password=xxxx in dietpi.txt apply to all software.

Great idea 👍

Fourdee, 2016-09-29:
Notes:
- 🈯 MySQL
- 🈯 MariaDB
- 🈯 PhpMyAdmin
-

Fourdee, 2016-09-29:
@Zone22 @sootstone 

May need to reconsider having 2 vars here. Some software installations use the root login (eg: samba/proftpd), others will use the global password (eg: mysql). 
This will only confuse the user and I can see users changing 1 password, but not the other.
Documentation wise, this is also a nightmare.

I believe 1 global password should be used. A simplified experience for the user. 
That password would be applied to root passwd, and, all software in dietpi-software that has a password set.
We have to accommodate users of all experience, so, I think 1 password is the preferred solution.

Open for discussion.

sootstone, 2016-09-29:
@Fourdee 

I sort of agree, while I would personally like to have two separate passwords for security reasons, this would complicate the user experience. Any passwords the user wished to change, could still be done manually. Would this mean keeping the option to change the root password in dietpi-config?

Just wanted to check since the topic of this issue has changed since I started it (which I'm fine with) , is there still a plan to allow the user to change the 'admin username' of Nextcloud (or any other dietpi-software) before installation? This is less of an issue.

Fourdee, 2016-09-30:
@sootstone 

> Would this mean keeping the option to change the root password in dietpi-config?

Yep. I'll add a notice when changing password, inform the user that software which uses the system ```root``` account, will be affected by a password change.

> allow the user to change the 'admin username' 

With NextCloud, the default username when loading the web interface for the 1st time is ```admin```. The username is already in the text box. So it felt unnecessary to change it from ```admin``` to anything else.

In regards to having a username setting for all software,, some software will use the system ```root``` account for login (eg: proftpd) regardless, others could be changed. But again, this could complicate the users experience.

```NEXTCLOUD_ADMIN_USER="admin"``` I'll add this in, one off special 👍

Zone22, 2016-09-30:
@Fourdee 
First off I think you should do what you think is best for the project. Although here's my two cents. I can see the issue of the complicating things for the end-user. The default password is what it has been thus far 'dietpi'. The average user has been fine with this and it should stay that way. Dietpi.txt is more of an advanced config. Therefore I think it's reasonable to expect a higher level of knowledge from the users that utilize Dietpi.txt. Nevertheless it's not that difficult to set root password through dietpi. Although It's nice to have secure settings on first boot.

sootstone, 2016-09-30:
@Zone22 @Fourdee 

Agree you should do what is best for the project, while I think it's very kind, please don't feel the need to add possibly unwanted features just for my sake. 

As a reference, I've been using DietPi on and off for about 4 months and have installed it probably 15+ times, I only just found out about dietpi.txt through this thread.

Fourdee, 2016-10-02:
@sootstone 

dietpi.txt | Applied during installation
```
#Optional username for nextcloud admin account, default is admin. Applied during installation.
dietpi_nextcloud_username=admin
```

> As a reference, I've been using DietPi on and off for about 4 months and have installed it probably 15+ times, I only just found out about dietpi.txt through this thread.

Yep, its one of those hidden goodies 👍

Fourdee, 2016-10-02:
Completed. New options in ```dietpi.txt```:
- ```AUTO_Global_Password=dietpi``` | sets root pw and all software installed with ```dietpi-software``` that sets a password. Requires v133 image or higher due to root pw being set before 1st update. http://dietpi.com/downloads/testing/
- ```AUTO_Dedicated_Usb_Drive_Format_Filesystem=0```
- ```dietpi_nextcloud_username=admin```

closed 2016-10-02 by Fourdee