MichaIng/DietPi #534: DietPi-Software | Emby fails to install

Fourdee, 2016-09-28:
> 

Caused by mono repo + packages:
```
root@DietPi:~# apt-get install -f
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
10 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up libnunit-core-interfaces2.6.3-cil (2.6.3+dfsg-1) ...
* Installing 1 assembly from libnunit-core-interfaces2.6.3-cil into Mono
Illegal instruction
Use of uninitialized value $_ in scalar chomp at /usr/share/cli-common/runtimes.d/mono line 275.
Use of uninitialized value $fullname in concatenation (.) or string at /usr/share/cli-common/runtimes.d/mono line 225.
Illegal instruction
E: installing Assembly /usr/lib/cli/nunit.core.interfaces-2.6.3/nunit.core.interfaces.dll failed
E: Installation of libnunit-core-interfaces2.6.3-cil with /usr/share/cli-common/runtimes.d/mono failed
dpkg: error processing package libnunit-core-interfaces2.6.3-cil (--configure):
 subprocess installed post-installation script returned error exit status 132
dpkg: dependency problems prevent configuration of libnunit-core2.6.3-cil:
 libnunit-core2.6.3-cil depends on libnunit-core-interfaces2.6.3-cil; however:
  Package libnunit-core-interfaces2.6.3-cil is not configured yet.

dpkg: error processing package libnunit-core2.6.3-cil (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of libnunit-util2.6.3-cil:
 libnunit-util2.6.3-cil depends on libnunit-core-interfaces2.6.3-cil; however:
  Package libnunit-core-interfaces2.6.3-cil is not configured yet.
 libnunit-util2.6.3-cil depends on libnunit-core2.6.3-cil; however:
  Package libnunit-core2.6.3-cil is not configured yet.

dpkg: error processing package libnunit-util2.6.3-cil (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of libnunit-console-runner2.6.3-cil:
 libnunit-console-runner2.6.3-cil depends on libnunit-core-interfaces2.6.3-cil; however:
  Package libnunit-core-interfaces2.6.3-cil is not configured yet.
 libnunit-console-runner2.6.3-cil depends on libnunit-core2.6.3-cil; however:
  Package libnunit-core2.6.3-cil is not configured yet.
 libnunit-console-runner2.6.3-cil depends on libnunit-util2.6.3-cil; however:
  Package libnunit-util2.6.3-cil is not configured yet.

dpkg: error processing package libnunit-console-runner2.6.3-cil (--configure):
 dependency problems - leaving unconfigured
Setting up libnunit-framework2.6.3-cil (2.6.3+dfsg-1) ...
* Installing 1 assembly from libnunit-framework2.6.3-cil into Mono
Illegal instruction
Use of uninitialized value $_ in scalar chomp at /usr/share/cli-common/runtimes.d/mono line 275.
Use of uninitialized value $fullname in concatenation (.) or string at /usr/share/cli-common/runtimes.d/mono line 225.
Illegal instruction
E: installing Assembly /usr/lib/cli/nunit.framework-2.6.3/nunit.framework.dll failed
E: Installation of libnunit-framework2.6.3-cil with /usr/share/cli-common/runtimes.d/mono failed
dpkg: error processing package libnunit-framework2.6.3-cil (--configure):
 subprocess installed post-installation script returned error exit status 132
dpkg: dependency problems prevent configuration of libnunit-mocks2.6.3-cil:
 libnunit-mocks2.6.3-cil depends on libnunit-framework2.6.3-cil; however:
  Package libnunit-framework2.6.3-cil is not configured yet.

dpkg: error processing package libnunit-mocks2.6.3-cil (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of libnunit-cil-dev:
 libnunit-cil-dev depends on libnunit-console-runner2.6.3-cil (= 2.6.3+dfsg-1); however:
  Package libnunit-console-runner2.6.3-cil is not configured yet.
 libnunit-cil-dev depends on libnunit-core2.6.3-cil (= 2.6.3+dfsg-1); however:
  Package libnunit-core2.6.3-cil is not configured yet.
 libnunit-cil-dev depends on libnunit-core-interfaces2.6.3-cil (= 2.6.3+dfsg-1); however:
  Package libnunit-core-interfaces2.6.3-cil is not configured yet.
 libnunit-cil-dev depends on libnunit-framework2.6.3-cil (= 2.6.3+dfsg-1); however:
  Package libnunit-framework2.6.3-cil is not configured yet.
 libnunit-cil-dev depends on libnunit-mocks2.6.3-cil (= 2.6.3+dfsg-1); however:
  Package libnunit-mocks2.6.3-cil is not configured yet.
 libnunit-cil-dev depends on libnunit-util2.6.3-cil (= 2.6.3+dfsg-1); however:
  Package libnunit-util2.6.3-cil is not configured yet.

dpkg: error processing package libnunit-cil-dev (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of libmono-cil-dev:
 libmono-cil-dev depends on libnunit-cil-dev (>= 2.4); however:
  Package libnunit-cil-dev is not configured yet.

dpkg: error processing package libmono-cil-dev (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of mono-devel:
 mono-devel depends on libmono-cil-dev (= 4.6.0.245-0xamarin2); however:
  Package libmono-cil-dev is not configured yet.

dpkg: error processing package mono-devel (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of emby-server:
 emby-server depends on mono-devel (>= 1.0); however:
  Package mono-devel is not configured yet.

dpkg: error processing package emby-server (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libnunit-core-interfaces2.6.3-cil
 libnunit-core2.6.3-cil
 libnunit-util2.6.3-cil
 libnunit-console-runner2.6.3-cil
 libnunit-framework2.6.3-cil
 libnunit-mocks2.6.3-cil
 libnunit-cil-dev
 libmono-cil-dev
 mono-devel
 emby-serverE: Sub-process /usr/bin/dpkg returned an error code (1)
```

Fourdee, 2016-09-28:
Test install on RPi 3 was successful.

Fourdee, 2016-09-29:
Not much we can do about this from our end, will need to wait for package/repo maintainers to provide a fix (or support) for armv6.

Fourdee, 2016-09-30:
Disabled Emby installation option for ARMv6.

Dalto1, 2016-10-03:
Dont know if this can help, but i think i got mono on my pine64

http://dietpi.com/phpbb/viewtopic.php?f=11&t=720

Fourdee, 2016-10-03:
@Dalto1 

Legend 👍 

ARM64 works, but armv6 still fails on beta branch:
```
Setting up binutils (2.25-5) ...
Setting up cli-common (0.9+nmu1) ...
Setting up libmono-system-numerics4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libgdiplus (3.6-1) ...
Setting up libmono-accessibility4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-cecil-private-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-peapi4.0a-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-microsoft-build-framework4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-cairo4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-microsoft-visualc10.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-parallel4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-simd4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-deployment4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-numerics-vectors4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-reflection-context4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-runtime-interopservices-runtimeinformation4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-web-mobile4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-web-regularexpressions4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-workflow-activities4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-workflow-componentmodel4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-workflow-runtime4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmonosgen-2.0-1 (4.6.1.3-0xamarin1) ...
Setting up libmonosgen-2.0-dev (4.6.1.3-0xamarin1) ...
Setting up libmono-2.0-dev (4.6.1.3-0xamarin1) ...
Setting up libmono-system-xml4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-security4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-security4.0-cil (4.6.1.3-0xamarin1) ...
Setting up mono-4.0-gac (4.6.1.3-0xamarin1) ...
Setting up libmono-system-runtime-serialization-formatters-soap4.0-cil (4.6.1.3-0xamarin1) ...
Setting up mono-gac (4.6.1.3-0xamarin1) ...
update-alternatives: using /usr/bin/gacutil to provide /usr/bin/cli-gacutil (global-assembly-cache-tool) in auto mode
Setting up mono-runtime-common (4.6.1.3-0xamarin1) ...
update-binfmts: warning: /usr/share/binfmts/cli: no executable /usr/bin/cli found, but continuing anyway as you request
Setting up mono-runtime-sgen (4.6.1.3-0xamarin1) ...
Setting up mono-runtime (4.6.1.3-0xamarin1) ...
update-alternatives: using /usr/bin/mono to provide /usr/bin/cli (cli) in auto mode
Setting up libmono-system-configuration4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system4.0-cil (4.6.1.3-0xamarin1) ...
Setting up ca-certificates-mono (4.6.1.3-0xamarin1) ...
Setting up libmono-corlib4.5-cil (4.6.1.3-0xamarin1) ...
Illegal instruction
Setting up libmono-posix4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-core4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-csharp4.0c-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-microsoft-csharp4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-data-tds4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-transactions4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-enterpriseservices4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-data4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-drawing4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-io-compression4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-servicemodel-internals0.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-runtime-serialization4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-web-applicationservices4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-identitymodel4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-identitymodel-selectors4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-messaging4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-configuration-install4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-webbrowser4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-i18n4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-i18n-west4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-windows-forms4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-messaging4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-sqlite4.0-cil (4.6.1.3-0xamarin1) ...
[... 173 lines cut ...]
Setting up libmono-system-web-mvc3.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-xml-serialization4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-servicemodel-activation4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-servicemodel-web4.0-cil (4.6.1.3-0xamarin1) ...
Setting up libmono-system-data-services4.0-cil (4.6.1.3-0xamarin1) ...
Processing triggers for libc-bin (2.19-18+deb8u6) ...
Processing triggers for systemd (215-17+deb8u5) ...
Processing triggers for ca-certificates (20141019+deb8u1) ...
Updating certificates in /etc/ssl/certs... 0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d....Updating Mono key store
Illegal instruction
Done
done.
Errors were encountered while processing:
 libnunit-core-interfaces2.6.3-cil
 libnunit-core2.6.3-cil
 libnunit-util2.6.3-cil
 libnunit-console-runner2.6.3-cil
 libnunit-framework2.6.3-cil
 libnunit-mocks2.6.3-cil
 libnunit-cil-dev
 libmono-cil-dev
 mono-devel
 emby-serverE: Sub-process /usr/bin/dpkg returned an error code (1)
```

Dalto1, 2016-10-03:
Yeah, i read to fast LOL. but still helped you for ARM64 :)

MichaIng, 2018-08-27:
@Fourdee 
Isn't this the same as https://github.com/Fourdee/DietPi/issues/1023, thus already resolved? 😃

Fourdee, 2018-08-27:
@MichaIng 

I'll flag for testing, will update ticket once i've been able to try it.

Fourdee, 2018-08-27:
Mono installs fine, Emby fails:

#### Details:
- Date           | Tue 28 Aug 00:40:13 BST 2018
- Bug report     | N/A
- DietPi version | v6.15
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (index=1)
- Kernel version | #1134 Tue Aug 14 16:58:07 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGI: emby-server embymagick
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.
```

Fourdee, 2018-08-27:
Hmm, standalone installs?
https://emby.media/linux-server.html

Fourdee, 2018-08-28:
Reworked installed completed. In regards to ARMv6, not fully supported by Emby:
https://github.com/Fourdee/DietPi/issues/534#issuecomment-416405968

closed 2018-08-28 by Fourdee