MichaIng/DietPi #537: Dietpi.txt | Set filing system and partition format for 1st run.

Zone22, 2016-09-28:
Set filing system and partition format in dietpi.txt for first boot. A check would be needed if the filing system and partition format already exists. In that case it would skip formatting.

Fourdee, 2016-09-30:
@Zone22 

We already have a AUTO_format feature in ```dietpi.txt```:

> #Dedicated USB drive automation
#	To use this feature, create a blank file: /boot/Automation_Format_My_Usb_Drive
#	If the file exists, the USB drive will be formatted to ext4. (WARNING: ALL DATA ON /dev/sda WILL BE DELETED. YOU HAVE BEEN WARNED!).
#   DietPi will configure your software installations to use your dedicated drive for personal data. (eg: owncloud data / bittorrent downloads etc)
#   NB: Plug in 1 USB drive before powering on system. Use 1 USB drive ONLY.

So currently:
- User creates file ```/boot/Automation_Format_My_Usb_Drive``` before 1st boot
- Boots system
- DietPi will wait for attached drive, then format EXT4
- GPT partition type (as of v133)
- Does not check for existing partition, format only.

We could add a feature to this, that sets filesystem type eg ```AUTO_Format_Filesystem=btrfs```, thoughts?

Zone22, 2016-09-30:
I don't understand this step. Why would the user need to create the file?
> User creates file /boot/Automation_Format_My_Usb_Drive before 1st boot

Otherwise it seems great to me.

Fourdee, 2016-09-30:
@Zone22 

> I don't understand this step. Why would the user need to create the file?

To prevent abuse, misuse and accidentally formatting their drive. 
I also would prefer ```AUTO_Format_Drive=1```, but by forcing the user to create a file named ```Automation_Format_My_Usb_Drive before 1st boot```, should prevent accidental formatting.

> Otherwise it seems great to me.

I'll add ```AUTO_Format_Filesystem=``` in.

Fourdee, 2016-09-30:
Completed.

closed 2016-09-30 by Fourdee