MichaIng/DietPi #539: No Local Sound

thechildofroth, 2016-10-01:
Having issues with no local sound from Subsonic/DAC again, can't see any issues with config

Bug report filled: 007bb5db-ab02-4525-9a09-589253418eba-1 

root@subsonic:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0

root@subsonic:~# cat /etc/asound.conf 
pcm.!default {
type hw
card 0
}
ctl.!default {
type hw
card 0
}

Fourdee, 2016-10-02:
@thechildofroth 

Strange, everything looks ok. Did you use dietpi-config to set the soundcard?

Does ```speaker-test -c2``` output any audio or errors?

thechildofroth, 2016-10-03:
Aha, sorry please close / delete / mark as fixed

User error

/redface

Fourdee, 2016-10-06:
@thechildofroth 

No worries. Thanks for letting us know 👍

closed 2016-10-06 by Fourdee